pymeasure/pymeasure #742: How to test adapter.connection calls

bmoneke, 2022-11-14:
Some instruments call the adapter's `connection`property, like #697 which makes ProtocolAdapter fail. Instruments should not be adjusted to make the ProtocolAdapter work.

The problem is, that you are not supposed to use normally the adapter's connection directly, because the `connection` depends on the specific adapter type. If an instrument calls the connection, the ProtocolAdapter fails. Instruments should not be adjusted to make the ProtocolAdapter work.

Two solutions:

- A generic mock connection seems good, because that would care for many use cases. Otherwise we had to add several methods (clear, close, timeout property, get_stb...) individually.
- Or we could add timeout as a property to the `Adapter`, which will call its connection depending on the Adapter type (ProtocolAdapter stores it as a variable). Similarly the "read_stb" and `clear` etc.

bilderbuchi, 2022-11-14:
Option 1 sounds preferable/easier to get right.

bmoneke, 2022-12-06:
Also #778 requires the timeout as a property. I guess the best path is a generic mock, but also a timeout property in the `Adapter`.

Sometimes, there are solutions to use VISA methods or general methods (e.g. setting timeout temporarily to 0). The latter is open for more adapters, while the former requires a VISAAdapter.
Therefore, I prefer the more general approach.

bilderbuchi, 2022-12-06:
> Another advantage of a Adapter.timeout property: That way we can ensure the right timeout setting regardless of the unit of the adapter (see difference VISA and pyserial). If an instrument sets the timeout to 500, it will be half a second, for the VISAAdapter and for the SerialAdapter, because the latter's property transforms it to 0.5 internally.

This is a good application, harmonising that would be useful indeed. Let's use the pyVISA way as a baseline. One thing that concerns me is that this will now make our configuration API more complex, as we need to special-case the `timeout` parameter, and also probably intercept it if people supply it already (but assume the "other" convention), and because it does not land directly in the adapter.connection as we have been saying until now. :-/

Should we use a different name? `timeout_ms` would make the unit clear, allow us to cleanly intercept/deprecate use of `timeout`, and break the correspondence with the respective connection kwarg (pyVISA, serial), so less potential for confusion.

bmoneke, 2022-12-06:
I did not think about the parameters of the adapter, but how the instrument interacts with the present adapter.
However I see the confusion in starting a serial adapter with a timeout of 0.5 and later setting the timeout to 500 for the same effect.

Maybe we just change the serial adapters interpretation of timeout?

closed 2022-12-12 by msmttchr