pymeasure/pymeasure #743: CSV's and file-based datastores are first class entities in Results

physicsninja, 2022-11-14:
Per the suggestion by @bmoneke I am opening an issue here to discuss the re-implementation of Result's I am proposing. 

Issues with current `Results`:
- CSV's are "first class". The `Results` class contains code to parse and format for CSV's only. Any other output type, even if it is file-based, requires some hacking
- File-based datastores are "first class". This restriction is a little less onerous, and exceptions to it uncommon, but one can imagine wanting to directly write to a database or some other data pipeline that  is not a file-based thing

Suggestions:
- Refactor `Results` into something like a `ResultsBase` akin to the `Procedure` base class that requires implementation of five methods. A pair to create and parse a header containing the procedure parameters, a pair to read and write to and from the datastore, and one to create the handlers that logging will use to write the data. The last method is so that when a `Recorder` is instanced, it has the tools to generate the Handlers on the results object, where previously `Recorder` assumed a `FileHandler` and set the formatter  from the results object.
- Implement the suggestion in #136 by @mivade  of lifting the `Logging` object that is used to deal with data up from just a formatter to a `Handler`. This makes it easier to extend to non file-based datastores
- To switch between results storing, the user would need to replace the `ResultsBase` subclass implementation passed in the construction of the experiment i.e. in a gui file something like: 
```   
 def queue(self):
    filename = tempfile.mktemp()
    
    procedure = self.make_procedure()
    results = CSVResults(procedure, filename)
    experiment = self.new_experiment(results)
    
    self.manager.queue(experiment)
```

Current draft:
PR #741 is an extremely rough draft of this proposed refactor. It does not implement any examples beyond file-based datastores, but there is an example that uses json instead of csv. For backwards compatibility `Results` is an alias of `CSVResults`.

bilderbuchi, 2022-11-14:
Drive-by comment (I am not that familiar with thza part of the codebase): maybe switching the messaging transport from logging to zmq would enable further flexibility, as results could be streamed to ~arbitrary sinks? Does that make sense?

physicsninja, 2022-11-14:
@bilderbuchi That is really good suggestion. It wouldn't even require that much more rework. Instead of the recorder instancing `Logging.Handler` -derived things, it could just subscribe to the proper zmq channels. However, we would still need to define the sinks and I image they would end up looking very similar to the suggested results implementation.

Though, by doing it that way we could also have an in-memory store for just the GUI so that the there is no read/write collision to the "permanent" datastore (currently the GUI reloads the data from file when progress is updated, that can lead to collision in some weird cases that I have hacked together. Having an in memory store that gets a duplicate of the data written to disk may be valuable in that regard.

bmoneke, 2022-11-16:
@physicsninja I ask you for some patience. We have a larger PR regarding instrument channels (#718) to finish. Besides, mainly @CasperSchippers works on that part of the code base.

> Having an in memory store that gets a duplicate of the data written to disk may be valuable in that regard.

Just take into account, that the in-memory data might get too large for the available memory.

bilderbuchi, 2022-11-16:
If it's for the GUI, I would expect you mainly need historical data for plotting. Then, you can keep the data in a ring buffer that keeps only the N most recent data points, and avoid memory problems.

CasperSchippers, 2022-11-16:
I think generally what you propose is a good idea to make this a bit more general; I would have to make some time to dig into that part of the code & you proposed idea.

bmoneke, 2022-11-16:
In discussion #676 we talk about a different measurement paradigm (several independent modules communicating).
Maybe we can put this issue and PR with that discussion together, to define some Zmq protocol to exchange data.

Yes, it will require some protocol definition work at the beginning, but it will enable more modularity later on. Are you interested @physicsninja?

physicsninja, 2022-11-17:
I thought about it some more and looked at #676 and breaking things out into modules and the code becoming less monolithic in general and I am torn. I think both the current recorder implementation and the prospect of the more advanced listener implementation suggested is are both valuable.

One of the things that was mentioned a couple of times in the discussion in #676 was the simplicity of pymeasure when it comes to just getting to work. I think having Recorder/Results interfaces that clearly wrangles things is nice in that regard. There is also a performance edge case that I have: 

in workers.py we have this code snippet
```
if self.port is not None and zmq is not None:
            try:
                self.context = zmq.Context()
                log.debug("Worker ZMQ Context: %r" % self.context)
                self.publisher = self.context.socket(zmq.PUB)
                self.publisher.bind('tcp://*:%d' % self.port)
                log.info("Worker connected to tcp://*:%d" % self.port)
                # wait so that the socket will be ready before starting to emit messages
                time.sleep(0.3)
            except Exception:
                log.exception("Couldn't establish ZMQ publisher!")
                self.context = None
                self.publisher = None
```
This means there is a 300 ms penalty to create a ZMQ socket for each instance of a worker. This actually matters a lot for my use case (applying a wf then grabbing a scope trace many many times). The procedure execution time is ~1.5 s so the 300 ms is a substantial time cost over thousands of procedures. Maybe this is more of a me problem and that this is not what pymeasure should be used for.

I guess I would proposed the scope of the PR #741 to have three parts:
- Refactor results to be data store agnostic
- Update the documentation to explain how to create new datastore pipelines
- Allow CSVResults to ingest single rows and dataframes with no extra specification needed (addressing #277 )

Then, as a longer term goal, hash out a protocol and framework for zmq based communication and work on that in a separate PR. This way those users who want to use something xarray or sql or parquet can do so right away and the PR doesn't end up being huge. I am also okay with the fact that a latter transition to zmq may render a lot of the work I am suggesting to do moot. It's more about a low hanging fruit that could be valuable right now (and I already have a lot of it implemented in a fork). 

What are your thoughts @bmoneke @CasperSchippers @bilderbuchi ?

bilderbuchi, 2022-11-18:
> add to the https://github.com/pymeasure/pymeasure/discussions/676 work to spec out the protocol and push that across the finish line.

Ha! that's barely across the *starting line* :D

> * Refactor results to be data store agnostic
> 
> * Update the documentation to explain how to create new datastore pipelines
> 
> * Allow CSVResults to ingest single rows and dataframes with no extra specification needed (addressing

I think the first two parts will be valuable, and we will have to do something like that later anway, so some structural work will be useful and the effort probably _not_ be lost later. We have to take care though to keep the PR size small so it stays reviewable.
I haven't thought about or have an opinion right now about the third point, but it sounds useful, too.

bmoneke, 2022-11-18:
I have not found the time yet to write the answers, but I thought along the same lines as bilderbuchi.

Some parts (data sink etc.) may be useful for that other project as well. Especially if the different parts are divided in different methods/classes.

Regarding the delay: Maybe we can introduce an option to disable zmq usage.

Regarding the Pull request: If there are some lines along which you can divide a Pull request (dividing it in different parts, e.g. one for refactoring, one for the CSVResults with dataframes...) will make the review process easier as you can tackle one part after the other.

I found out the hard way, what a hassle large PRs are...

physicsninja, 2022-11-18:
> Regarding the delay: Maybe we can introduce an option to disable zmq usage.

We kind of did something like that on our fork. We exposed `port` as kwarg on `ManagedWindow`  and then we just set it to `None` so that if statement is dodged. It's maybe a separate issue/PR.

> Regarding the Pull request: If there are some lines along which you can divide a Pull request (dividing it in different parts, e.g. one for refactoring, one for the CSVResults with dataframes...) will make the review process easier as you can tackle one part after the other.

That is an eminently reasonable suggestion. I think narrowing the scope of the first PR to the first two points is reasonable, and I think there is imprimatur enough to keep working on it. I will ping again when it is polished for review

bmoneke, 2022-12-01:
> This means there is a 300 ms penalty to create a ZMQ socket for each instance of a worker. 

@physicsninja I just looked at the code and noticed that `if self.port is not None`, so if you do not specify a port, you do not get that time penalty!
Only if you specify that port, then you get a socket and the time is necessary.