pymeasure/pymeasure #752: Generator for protocol tests

bmoneke, 2022-11-18:
During the Instrument rework, I wrote an initial version of a protocol test generator, see [my branch](https://github.com/pymeasure/pymeasure/compare/master...bmoneke:pymeasure:Test_Generator).

If you communicate with an instrument through this generator, the generator creates a protocol test file according to that communication. Use case is, that you have a working implementation and want to write tests, such that later code changes still produce the same communication.
~~~
# Example usage testing properties of an instrument
g = Generator("test_tc038.py")
g.instantiate(TC038, "COM5", 'hcp')
g.test_property("information")
g.test_property("monitored_value")
g.test_property_setter("setpoint", 20)
g.test_property("setpoint")
~~~

I did not propose it earlier, due to more pressing issues and shortage of labour.

Now I want to bring it back to attention, in order to see, where we can put it in the task queue. If contributors see it as helpful for writing tests for their instruments, I'll move it up on the task list (which will delay other works and PR reviews).

LongnoseRob, 2022-11-19:
Looks interesting,
let me find some time to test it with actual hardware

bilderbuchi, 2022-11-19:
I think this is a useful feature to easily nail down communication into protocol tests. 
Timing-wise, I'd prefer if we first worked down the PR queue a bit before taking up the next (big?) feature PR. I think 50+ open PRs is frustrating for all involved, and would love to get that down as much as feasible (1 page?). This should also improve our future response time, as right now we have to divide our review attention between too many places.

bmoneke, 2022-11-19:
Ok, I improved the example with some comments. Whoever wants to use the generator, may do so with my code already existing.

bmoneke, 2022-11-23:
Thanks for testing it.

The generator was part of an already large PR and therefore postponed. As such it is not yet complete

> for 'test_property_setter` it might be nice to have some logic to use iterables as argument

I already planned to collect all tests of a property together and create a `parametrize` test.
Thanks for suggesting. As a setter might accept an iterable anyway, I can introduce `test_property_setter_values` or such which loops through an iterable (or you can loop through it). However, as you should verify, that the code does what it is supposed to do (e.g. test that setting the voltage to 12.3 V, really does that), it is counterproductive to just test many values without verifying. It might be useful for getting additional values, I concede.

> the test-file is opened but not automatically written if following your example. helped with a g.__del__() at the end

The file is written (at least according to the code), but it is not closed until closure of the generator. Depending on the OS you might not see the changes, because the file is blocked.

> test_method() does not work on methods which return nothing or only return a non-callable

Thanks for pointing out, yes, according to the code, it expects a return value.