pymeasure/pymeasure #755: Better API documentation example for instruments

bmoneke, 2022-11-19:
I just stumbled upon a bad API call, which included all inherited members, such that CI complained.
Therefore I propose to change the line in the documentation from:

"Copy an existing instrument documentation file, which will automatically generate the documentation for the instrument"

to the code, which should be used.

bilderbuchi, 2022-11-19:
I did not understand [that error message](https://github.com/pymeasure/pymeasure/actions/runs/3503905662/jobs/5869253099), though -- a lot of other instruments use the same way of using `:inherited-members:`. Why did just that one fail?

bmoneke, 2022-11-20:
Neither did I understand why it happened then and not earlier.
Nevertheless, I think it is good to put our desired code in the documentation.
Do we want to have the inherited methods as well, or not?

bilderbuchi, 2022-11-21:
> Neither did I understand why it happened then and not earlier.

And also that there remain a lot of instances of what looks like identical usage, but works? I'm puzzled, too :shrug:

bmoneke, 2022-11-22:
> Yes, IMO, at least we should not forbid it. Think of the TEXIO (#698) - that documentation would be pretty empty without inherited members. 😅 Also, in the future if we get instrument classes defined, we will have a much deeper class hierarchy.

That makes sense, yes.