pymeasure/pymeasure #757: Instruments announcements

bmoneke, 2022-11-22:
I have a few instruments in use, which I will publish, once we reduced the number of pull requests and urgent issues. However, in case someone needs these drivers, I'd be happy to open a PR and offer the code. Therefore, I mention these devices here:

- Ophir powermeters: Nova I, Nova II
- Aculight Argos 2400 OPO system
- Laserquantum (Novanta) Finesse laser system
- Thyracont VCR pressure sensor
- IPG Photonics YAR-15K fiber amplifier
- Yokogawa AQ6370 optical spectrum analyzer (very initial state of the driver)

JinchW, 2023-09-22:
Hi, I wonder if it is convenient to share the control program of YOKOGAWA AQ6370 optical spectrum analyzer now? I am a beginner of Pymeasure and want to get some examples about YOKOGAWA optical spectrum analyzer.

bmoneke, 2023-09-23:
I added a link to the corresponding branch in my repository.
I have to warn you, though, it is a very initial version and a year old. It does not use the new Channels functionality, so it is not a good example of the newest pymeasure features. Nevertheless, it is a good starting point (just add channels for the traces).

If you need a working and tested code, you can look at instrumentkit: https://github.com/instrumentkit/InstrumentKit/blob/main/src/instruments/yokogawa/yokogawa6370.py

BenediktBurger, 2024-03-11:
Hey @JinchW , someone else made a good pymeasure driver for the Yokogawa AQ6370. See #1059 .