pymeasure/pymeasure #759: ChannelCreator does not work in parent class

bmoneke, 2022-11-22:
In #758, the ChannelCreator did not work, unless used in the last class defined, because the class dict contains only the items defined in that class, not inherited.

Does "inspect" help here?

msmttchr, 2022-11-22:
Indeed. This was overlooked. I think inspect doesn't help and you need to use a strategy based on MRO (see `_setup_special_names`) as an example.

bmoneke, 2022-11-22:
I had a test (and might still have), which tests inheritance, but not (yet) in this constellation.

It was good to delay the channels for 0.12.

msmttchr, 2022-11-23:
Correction: inspect should work as well.

bilderbuchi, 2022-11-23:
> class dict contains only the items defined in that class, not inherited.

does `dir(ChildClass)` not work? This should also contain inherited attributes afaik.

msmttchr, 2022-11-23:
`dir `give you the list of attribute names(including inherited ones), but it doesn't return the value.

PR #761 (2022-11-28): Fix ChannelCreator resolution broken #759
Fixes #759 

- adds a test to ensure that inheritance (and overriding as expected) works
- outsources the channel creation to its own method
- fixes the problem

Fix description:

1. Look through the MRO (in reverse order) and collect all `ChannelCreator` objects in a dictionary, the key is the variable name of the object. The reverse order ensures, that the `ChannelCreator` assigned to `channels` in a subclass replaces that of a superclass assigned also to `channels`.
2. Create the children according to that collected dictionary.

Not working possibil [... cut]
closed 2022-11-28 by bilderbuchi