pymeasure/pymeasure #774: Instructions in the "Contributing" section of the manual incomplete. No `setup.py` file.

tomverbeure, 2022-12-05:
I'm trying to add support for my instruments, and I'm following the instructions of [Contributing](https://pymeasure.readthedocs.io/en/latest/dev/contribute.html) section.

When I run the following command:

```
pip install -e .```

I get:

```
tom@zen:~/projects/pymeasure$ pip3 install -e .
ERROR: File "setup.py" not found. Directory cannot be installed in editable mode: /home/tom/projects/pymeasure
(A "pyproject.toml" file was found, but editable mode currently requires a setup.py based build.)
```

There is indeed no `setup.py` file at top of tree in the GitHub repo either.

What am I missing?

Thanks!
Tom

bmoneke, 2022-12-05:
Hey @tomverbeure 

as far as I know, this file is not necessary anymore, it is even discouraged. What python version do you have? Are you in the correct directory (i.e. the directory contains the `setup.cfg`, `README.md`...?

I think it might be necessary to have installed `setuptools` and maybe one or more of these packages: `wheel`, `build`.

bilderbuchi, 2022-12-05:
Indeed, try updating your pip and setup tools before trying again

tomverbeure, 2022-12-05:
Hi,

I'm running version 3.8.10 (the default of Ubuntu 20.04). I'm definitely in the correct directory. 

I already had `setuptools` (45.2.0) and `wheel` (0.34.2) nstalled. After installing `build` (0.9.0), the error is reduced to:

```
ERROR: File "setup.py" not found. Directory cannot be installed in editable mode: /home/tom/projects/hp3478a_calibration
```

My current work-around is to just run `pip3 install .` and run that whenever I make a code change. Clunky but it gets me unblocked.

Tom

bmoneke, 2022-12-05:
Maybe you have to update pip itself, as bilderbuchi suggested.

I agree that installing it over and over again is no valid option.

tomverbeure, 2022-12-05:
Pip3 update made no difference either. Installing over and over is really not too bad, it only takes a few seconds.

I'll update this issue when I find some time to update to the latest Python version.

Tom

bilderbuchi, 2022-12-05:
You unfortunately do not give your pip version. The only possible item concerning editable installs I could [hunt down](https://github.com/pypa/pip/blob/main/NEWS.rst#213-2021-10-11) was that pip 21.3 got
 
> Support editable installs for projects that have a pyproject.toml and use a build backend that supports [PEP 660](http://www.python.org/dev/peps/pep-0660)

Possibly your setuptools is too old? I don't know the versioning scheme by heart, but I'm on 65 here. Do you get your python packages from the system packaging (i.e. deb packages via apt), or do you use virtualenvs, conda, or some other environment mechanism?

tomverbeure, 2022-12-06:
Issue has been identified: I had installed the latest pip3 (22.3.1), but my system also had pip3 installed with `apt-get`, and that one was only version 20.x. The latter had priority in my $PATH.

After uninstalling the Ubuntu pip3 package, things work as expected.

Thanks!

bilderbuchi, 2022-12-06:
Glad to hear!