pymeasure/pymeasure #779: Add collection of instrument solutions to documentation

bmoneke, 2022-12-06:
@bilderbuchi proposed in #778 to add a list of solutions for instrument implementations to the documentation:
For different problems there are listed instruments, which solve these problems. 

I think the end of the file might be a good place.

bilderbuchi, 2022-12-07:
A separate file should also be fine -- the `adding_instruments.rst` is slowly growing huge.

To mirror what's in #778, this should be a FAQ-style directory in the documentation where we record weird/special instruments, so that contributors can check their list for implementation inspiration.
I imagine something like
* TC038, x and y implement binary frame-based communication
* aja.DCXS has replies that are all the same lengths
* aja.DCXS has some garbage in the buffer after startup, that needs to be discarded
* XYZ (I forget which) VNA can dynamically configure itself depending on which hardware modules are present
* In the MKS937B, not all the channels have all capabilities
* ...

bilderbuchi, 2023-01-02:
* `tcpowerconversion.CXN` (#800) needs overridden `values` for both the instrument and its channels, and needs to take care to use the correct `ask` method. A class-level override `values=values` can be used as long as we don't have a better solution (#811).

bmoneke, 2023-01-05:
- #802 has different channel names on the front panel (A, B, C) and command ids (1, 2, 3).

bmoneke, 2023-01-09:
- #827 has different Channel property attributes for different subclassed instruments.

PR #853 (2023-02-09): Solutions for instrument implementation challenges added
Closes #779
closed 2023-02-09 by bilderbuchi