pymeasure/pymeasure #783: Timeout property for adapters

bmoneke, 2022-12-07:
In #742 the question was raised to harmonise adapters with a uniform timeout unit as VISA uses milliseconds and pyserial seconds.

As pyVISA is our default, its ms setting should be default as well.

msmttchr, 2022-12-07:
I would also consider whether we want to keep `SerialAdapter` at all. After you managed to switch Prologix to `pyvisa`, I don't see any use of it, except that it requires way less effort to install compared to `pyvisa`.

bmoneke, 2022-12-08:
With pyvisa-py, there should be no reason against using VISA, unless the space and computation requirement differ that much.

I'm not sure, whether it is good to keep the SerialAdapter as a minimum requirements alternative. Maybe a few instruments (calling VISA methods) won't work, but most instruments will work regardless. For this edge case, a warning about the different timeout units might suffice.

LongnoseRob, 2022-12-08:
> With pyvisa-py, there should be no reason against using VISA, unless the space and computation requirement differ that much.
>
> I'm not sure, whether it is good to keep the SerialAdapter as a minimum [...]

How about a rather longterm plan to deprecatate the SerialAdapter and a short term modification to move the the timing topics (it is only timeout?)to the more used ms?

bilderbuchi, 2022-12-08:
It's easier with serial, compared to pyvisa, to do spy/loopback inspection (although we currently don't use that, we could to inspect more deeply than with protocoladapter). It's also a bit more "bare-bones"/less complex to reason about. Installation is also more "immediate" (no need to point people to pyvisa-py).
These are just ancillary points, I'm not saying they are sufficient to keep serialadapter around long-term, I just want to bring them to your attention.