pymeasure/pymeasure #792: Several device tests fail for test_lecroyT3DSO1204_with_device.py

bleykauf, 2022-12-15:
Just ran `test_lecroyT3DSO1204_with_device.py` and encountered several failed tests. Here is the output from the test:

Here is the short summary:

```
=== short test summary info ======================================================================================
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_ch_current_configuration - AssertionError: assert {'attenuation... 'dc 1M', ...} == {'attenuation... 'dc 1M', ...}
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_ch_trigger_level - assert 0.12 == 0.125
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_ch_trigger_level2 - assert 0.12 == 0.125
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_timebase - assert 1e-06 ± 1.0e-07 == 0.0005
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_acquisition_status - AssertionError: assert 'auto' == 'triggered'
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_waveform_points[100] - AttributeError: 'float' object has no attribute 'index'
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_waveform_points[1000] - AttributeError: 'float' object has no attribute 'index'
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_waveform_points[10000] - AttributeError: 'float' object has no attribute 'index'
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_waveform_preamble - AssertionError: assert {'average': N...000000.0, ...} == {'average': N...000000.0, ...}
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[100-C1] - AttributeError: 'LeCroyT3DSO1204' object has no attribute 'single'
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[100-C2] - AttributeError: 'LeCroyT3DSO1204' object has no attribute 'single'
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[100-C3] - AttributeError: 'LeCroyT3DSO1204' object has no attribute 'single'
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[100-C4] - AttributeError: 'LeCroyT3DSO1204' object has no attribute 'single'
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[1000-C1] - AttributeError: 'LeCroyT3DSO1204' object has no attribute 'single'
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[1000-C2] - AttributeError: 'LeCroyT3DSO1204' object has no attribute 'single'
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[1000-C3] - AttributeError: 'LeCroyT3DSO1204' object has no attribute 'single'
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[1000-C4] - AttributeError: 'LeCroyT3DSO1204' object has no attribute 'single'
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[10000-C1] - AttributeError: 'LeCroyT3DSO1204' object has no attribute 'single'
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[10000-C2] - AttributeError: 'LeCroyT3DSO1204' object has no attribute 'single'
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[10000-C3] - AttributeError: 'LeCroyT3DSO1204' object has no attribute 'single'
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[10000-C4] - AttributeError: 'LeCroyT3DSO1204' object has no attribute 'single'
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_single_point - AttributeError: 'LeCroyT3DSO1204' object has no attribute 'single'
================================================================= 22 failed, 74 passed, 5 skipped, 2 warnings in 227.24s (0:03:47) =================================================================
```

And this is the full output:

```
(pymeasure) PS C:\Users\gain\Documents\programming\pymeasure> pytest .\tests\instruments\lecroy\test_lecroyT3DSO1204_with_device.py
======================================================================================= test session starts ========================================================================================
platform win32 -- Python 3.11.0, pytest-7.2.0, pluggy-1.0.0
rootdir: C:\Users\gain\Documents\programming\pymeasure
collected 101 items

tests\instruments\lecroy\test_lecroyT3DSO1204_with_device.py .F................................................FF.F..............F.FFFF.......FFFFFFFFFFFFFsssss..                            [100%]

============================================================================================= FAILURES ============================================================================================= 
________________________________________________________________________ TestLeCroyT3DSO1204.test_ch_current_configuration _________________________________________________________________________ 

self = <test_lecroyT3DSO1204_with_device.TestLeCroyT3DSO1204 object at 0x000001B3ECA3BC90>
autoscaled_scope = <pymeasure.instruments.lecroy.lecroyT3DSO1204.LeCroyT3DSO1204 object at 0x000001B3E121A450>

    def test_ch_current_configuration(self, autoscaled_scope):
        autoscaled_scope.ch_1.offset = 0
        autoscaled_scope.ch_1.trigger_level = 0
        autoscaled_scope.ch_1.trigger_level2 = 0
        expected = {"channel": 1,
                    "attenuation": 1.,
                    "bandwidth_limit": False,
                    "coupling": "dc 1M",
                    "offset": 0.,
                    "skew_factor": 0.,
                    "display": True,
                    "unit": "V",
                    "volts_div": 0.05,
                    "inverted": False,
                    "trigger_coupling": "dc",
                    "trigger_level": 0.,
                    "trigger_level2": 0.,
                    "trigger_slope": "positive"
                    }
        actual = autoscaled_scope.ch(1).current_configuration
>       assert actual == expected
E       AssertionError: assert {'attenuation... 'dc 1M', ...} == {'attenuation... 'dc 1M', ...}
[... 575 lines cut ...]
    def __getattribute__(self, name):
        """ Prevent read access to variables with special names used to
        support dynamic property behaviour."""
        if name in ('_special_names', '__dict__'):
            return super().__getattribute__(name)
        if hasattr(self, '_special_names'):
            if name in self._special_names:
                raise AttributeError(
                    f"{name} is a reserved variable name and it cannot be read")
>       return super().__getattribute__(name)
E       AttributeError: 'LeCroyT3DSO1204' object has no attribute 'single'

pymeasure\instruments\common_base.py:215: AttributeError
========================================================================================= warnings summary ========================================================================================= 
pymeasure\adapters\telnet.py:25
  C:\Users\gain\Documents\programming\pymeasure\pymeasure\adapters\telnet.py:25: DeprecationWarning: 'telnetlib' is deprecated and slated for removal in Python 3.13
    import telnetlib

tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_image_default_arguments
  C:\Users\gain\Documents\programming\pymeasure\pymeasure\adapters\adapter.py:227: DeprecationWarning: The binary mode of fromstring is deprecated, as it behaves surprisingly on unicode inputs. Use frombuffer instead
    return np.fromstring(data, dtype=dtype, **kwargs)

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
==================================================================================
```

bilderbuchi, 2022-12-17:
@LastStarDust can you reproduce this with your instrument?

bleykauf, 2022-12-18:
The `single` method was replaced by `arm_acquisition` in [edd643b](https://github.com/pymeasure/pymeasure/pull/697/commits/edd643bad5ec466d73d5196f5595731f7d431de3). Simply replacing the method names resulted in some divide-by-zero error but did not look into it further, yet.

bleykauf, 2022-12-18:
For some reason, I cannot checkout this commit, `fatal: reference is not a tree: edd643bad5ec466d73d5196f5595731f7d431de3`. Look like all changes are in a single commit, fb73c5b6094461c08d65933df8cf5db46d4ee1d0. Original branches are deleted/not reachable for me. Don't really understand since the commit history seems to be there and shows up on Github.

bilderbuchi, 2022-12-18:
Yes, I had to rebate and merge that PR separately, see #770.

bleykauf, 2022-12-18:
Was aware of #770 but did not see anything about squashing there. Interesting that the info about the individual commits is still present on github even though the source branches have been deleted or made private. Seems like it should be possible to checkout the state of the repo at at one of these commits given that info still being there but I am not familiar enough with git to see how...

bleykauf, 2022-12-18:
It is possible to checkout the PR with the [Github CLI](https://cli.github.com/) with `gh pr checkout 697`.

bilderbuchi, 2022-12-18:
Huh, I misremembered about squashing, only rebased. What do are you looking for with git? I can help sort it out.

LastStarDust, 2022-12-19:
@bilderbuchi @bleykauf 
Hello. Sorry for the late reply but I am temporarily working remotely from Europe.

Since the last time I run the with_device test, I modified the source code so I guess something got out of sync.
Anyway, in another project of mine, I am controlling the oscilloscope with the latest version of the code without any software-related issues, so I am pretty confident that the code is mostly fine.

I won't have access to the hardware until the end of January. Would it be OK if I debug the tests then?

LastStarDust, 2022-12-19:
One important thing that I am silently assuming, is that when running with_device tests, CH1 should be connected to the test output of the oscilloscope.

bleykauf, 2022-12-19:
@bilderbuchi: I just wanted to look into the commit before edd643bad5ec466d73d5196f5595731f7d431de3 (when `single` was renamed) whether the tests pass there and then figure out the problem from there. I was (and actually still am to some degree) confused of how Github handles #770/#697: due to the squashing, I cannot checkout edd643bad5ec466d73d5196f5595731f7d431de3, since only https://github.com/pymeasure/pymeasure/commit/fb73c5b6094461c08d65933df8cf5db46d4ee1d0 is part of the history. Usually, I would simply add the LastStarDust's repo as a remote and checkout his branch `/dev/lecroy-t3dso1204` but both source branches (https://github.com/pymeasure/pymeasure/tree/pr-697 and https://github.com/LastStarDust/pymeasure/tree/dev/lecroy-t3dso1204) have since been deleted. However, Github still displays the complete history, so I guess they are internally creating a branch that tracks the source branch(?). Long story short, I could checkout the branch from PR with `gh pr checkout 697` and everything works now as expected.

@LastStarDust: With test output, you mean Pass/Fail or Trigger Out (page 12 on https://cdn.teledynelecroy.com/files/manuals/t3dso1000_usermanual.pdf)? (I don't have the device in front of me right now but can test it this week).

LastStarDust, 2022-12-19:
@bleykauf By the test output I mean the 1kHz square wave terminals. Number 21 on page 11 of the user manual you linked.

bleykauf, 2022-12-19:
Thanks, most of the errors are now gone at least with commit https://github.com/pymeasure/pymeasure/pull/770/commits/67a1f007142209d1c7daaac3ddd378b4f3031f21 (the last commit before `single` method was renamed to `arm_acquisition`. The output is posted below.

I started [branch](https://github.com/bleykauf/pymeasure/tree/dev/fix-lecroy-t3dso1204-tests) that add a description how to set up the instrument and fixes the method name to `arm_acquisition`. This branches off of [this branch](https://github.com/bleykauf/pymeasure/tree/dev/instrument-tests) but can soon be rebased from master once #789 is merged. However, here I get 16 errors, see output below.

I most probably won't investigate this much further but can always run tests on my device if it helps.

Out with 67a1f007142209d1c7daaac3ddd378b4f3031f21:

```
======================================================================= short test summary info ======================================================================== 
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[100-C2] - BufferError: read bytes (18) != requested bytes (118)
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[100-C4] - BufferError: read bytes (18) != requested bytes (118)
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[1000-C1] - BufferError: read bytes (18) != requested bytes (1018)
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[1000-C3] - BufferError: read bytes (18) != requested bytes (1018)
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[10000-C1] - BufferError: read bytes (18) != requested bytes 
(10018)
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[10000-C3] - BufferError: read bytes (18) != requested bytes 
(10018)
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_single_point - ValueError: cannot call `vectorize` on size 0 inputs unless `otypes` is set
=================================================== 7 failed, 89 passed, 5 skipped, 14 warnings in 229.41s (0:03:49) =================================================== 
```

Output with [partially fixed branch](https://github.com/bleykauf/pymeasure/tree/dev/fix-lecroy-t3dso1204-tests):

```
======================================================================= short test summary info ======================================================================= 
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_waveform_points[100] - AttributeError: 'float' object has no attribute 'index'
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_waveform_points[1000] - AttributeError: 'float' object has no attribute 'index'
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_waveform_points[10000] - AttributeError: 'float' object has no attribute 
'index'
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[100-C1] - ZeroDivisionError: float division by zero        
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[100-C2] - ZeroDivisionError: float division by zero        
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[100-C3] - ZeroDivisionError: float division by zero        
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[100-C4] - ZeroDivisionError: float division by zero        
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[1000-C1] - ZeroDivisionError: float division by zero       
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[1000-C2] - ZeroDivisionError: float division by zero       
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[1000-C3] - ZeroDivisionError: float division by zero       
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[1000-C4] - ZeroDivisionError: float division by zero       
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[10000-C1] - ZeroDivisionError: float division by zero      
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[10000-C2] - ZeroDivisionError: float division by zero      
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[10000-C3] - ZeroDivisionError: float division by zero      
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_data[10000-C4] - ZeroDivisionError: float division by zero      
FAILED tests/instruments/lecroy/test_lecroyT3DSO1204_with_device.py::TestLeCroyT3DSO1204::test_download_single_point - BufferError: read bytes (18) != requested bytes (19)
========================================== 16 failed, 80 passed, 12 skipped, 699 deselected, 2 warnings in 231.08s (0:03:51) ========================================== 
```

bmoneke, 2022-12-19:
@bleykauf and @LastStarDust I want to thank you for your investigation.

LastStarDust, 2022-12-19:
@bleykauf Thank you for the investigation. 
The error `BufferError: read bytes (18) != requested bytes (118)` means that you are receiving empty data for some reason. 18 bytes is the length of the package header and footer without any data.
I will try to reproduce this problem in January. I might contact you then.

LastStarDust, 2022-12-19:
My (wild) guess is that the `ZeroDivisionError: float division by zero` is happening because the code is trying to process empty data. I am probably missing a sanity check to check if the data is non-empty.

bilderbuchi, 2022-12-20:
Thanks for the detailed investigation! 
Sure, take your time, there's no hurry with the investigation, Christmas is approaching fast, after all.

LastStarDust, 2023-01-27:
@bilderbuchi @bmoneke @bleykauf I created a new pull request with the bugfixes: https://github.com/pymeasure/pymeasure/pull/841

closed 2023-01-30 by bilderbuchi