pymeasure/pymeasure #804: SCPIbaseMixin class instead of includeSCPI

bmoneke, 2022-12-23:
Ideas regarding instruments with common controls include Mixin classes which contain controls.

The first element are the base SCPI commands (right now included in `Instrument`)

I propose the following transition path:

1. Create a Mixin class and write the proper documentation (right now, the includeSCPI is not in the documentation!). I propose `SCPIbaseMixin` as class name, as it contains the SCPI base commands.
2. Deprecate `includeSCPI`
3. Add the SCPIbaseMixin to all instruments with `includeSCPI=True`, add `SCPIbaseUnknownMixin` to all instruments without an explicit definition of `includeSCPI`. The latter class is a subclass of the former one and raises a deprecation warning, such that we get information, whether the instrument is SCPI conform (changing to the SCPIbaseMixin) or not (removing the Mixin).

During step 3, we can tackle #658 as well, as both require touching the init of all instruments.

msmttchr, 2022-12-23:
I think this is a very good proposal.
I would consider simply removing `includeSCPI` (and associated code) rather than deprecating it.
I am not sure also how useful is `SCPIbaseUnknownMixin` since the default has always been `includeSCPI=True` and there was no recommendation to explicitly set it to `True`.

Finally, I am not clear how the Mixin class would be defined. Is it derived from `object` or other class?
Are we going to use control/measurement/setting properties?

If you have time, could you please sketch some implementation in the thread ?

bmoneke, 2022-12-23:
> I am not sure also how useful is SCPIbaseUnknownMixin since the default has always been includeSCPI=True and there was no recommendation to explicitly set it to True.

There are two reasons, why an instrument does not have `includeSCPI` defined: Either the implementor wanted to use SCPI and left the default, or the implementor did not know, that it exists and therefore could not set it to True.
As we do not know, which one is True, I want to have this UnknownMixin (only for our current codebase, not for future additions) in order to get feedback of users and to sort the codebase.

> Finally, I am not clear how the Mixin class would be defined. Is it derived from object or other class?

The Mixin just contains properties and methods (both suppose that an Instrument as "mainclass" exists), so it does not need a superclass:

~~~python
class SCPIbaseMixin:

    @property
    def id(self):
        """ Requests and returns the identification of the instrument. """
        return self.ask("*IDN?").strip()

    def clear(self):
        """ Clears the instrument status byte
        """
        self.write("*CLS")

    some_control = Instrument.control("W %s", "R", """Some control could be implemented as well.""")

class SomeInstrumentWithSCPI(SCPIbaseMixin, Instrument):
    """Some example instrument which contains the SCPI base commands."""
~~~

> Are we going to use control/measurement/setting properties?

The pymeasure properties can be used as well, as the actual instrument will have also the init and other methods of `Instrument`.

msmttchr, 2022-12-23:
> There are two reasons, why an instrument does not have `includeSCPI` defined: Either the implementor wanted to use SCPI and left the default, or the implementor did not know, that it exists and [...]

Thanks for the clarification, I understand your willingness to be cautious and improve codebase quality.
Still, I would think that this will not justify the effort of adding an extra class. In any case, I understand your motivations, and, in some way, I share them.

> The Mixin just contains properties and methods (both suppose that an Instrument as "mainclass" exists), so it does not need a superclass:
>
> ```python
> class SCPIbaseMixin:
>
>     @property
>     def [...]

Very clear and nice, thanks. I would think that, once this is done, we can start addressing also #416.

bmoneke, 2022-12-23:
The additional class is quite simple, as we just add raising a warning:
~~~python
class SCPIbaseUnknownMixin(SCPIbaseMixin):
    def __init__(self, *args, **kwargs):
        warn("Please provide feedback, whether this instrument supports SCPI commands or not.")
        super().__init__(*args, **kwargs)
~~~

bilderbuchi, 2022-12-23:
The proposed approach sounds good. 
I agree that we should use a deprecation approach. 
We also have to take care to catch user-implemented instruments and transition those to the new usage. If they used just the default way, i.e. without an explicit `includeSCPI` argument, this will probably need some logic in `Instrument` to find instruments that had no `includeSCPI` arg and neither of the new classes as a base class. 
It feels kinda "ugly" to _need_ a `SCPIbaseUnknownMixin` to declare the _absence_ of something (I'd rather go without that), so I'm open to other proposals, I haven't thought deeply through this.

bilderbuchi, 2022-12-23:
Another thing we need to take care of: proper usage of `super().__init()` to ensure cooperative multiple inheritance. 
Do we need to prescribe the order of base classes, i.e., is `MyInstr(Instrument, SCPIbaseMixin)` and MyInstr(SCPIbaseMixin, Instrument)` fully equivalent (would be great if it were!)?
Do our "magic" features (channels, dynamic properties) still work as expected with this?

I remember some concerns along those lines in #416, and I think the SCPI stuff is an opportunity with simple constraints to investigate these things -- if it turns out it's not an issue, we could also use mixins for a trait-like enabling of features (like `PSUMixin`) when solving #416.

bmoneke, 2022-12-23:
> I agree that we should use a deprecation approach.

I'd keep the current code working (though `includeSCPI=True` will raise a deprecation warning).

> It feels kinda "ugly" to need a SCPIbaseUnknownMixin to declare the absence of something

This SCPIbaseUnknownMixin does declare SCPI base commands, but also states, that it is unknown, whether the commands work or not. Those instruments, which do not have SCPI commands, won't have any Mixin class.

> Do we (can we, even?) need to check user-defined instruments for this?

I don't see a simple way, how to change the default value of `includeSCPI` (removing the SCPI commands basically inverts the default value) without breaking user-defined instruments, which do not declare it explicitly.

> Do we need to prescribe the order of base classes

Afaik, the order of classes is important, once you have name clashes. Therefore, the `Instrument` should be the last in the list of classes (such that the particularities override the general implementation).
As long as `Instrument.__init__` is called, dynamic properties should work.

CasperSchippers, 2022-12-24:
I think this is indeed a great idea, way better than the `includeSCPI` keyword argument.

> > I agree that we should use a deprecation approach.
> 
> I'd keep the current code working (though `includeSCPI=True` will raise a deprecation warning).

I agree that a depreciation warning is the best option, especially since users might also have private/custom instrument implementation that are not part of the library and can therefore not by updated to the new mixin architecture in a PR.

> This SCPIbaseUnknownMixin does declare SCPI base commands, but also states, that it is unknown, whether the commands work or not. Those instruments, which do not have SCPI commands, won't have any [...]

I agree that it is generally good to make this distinction somehow.
Just to coin another option: we might be able to use the [__init_subclass__](https://docs.python.org/3/reference/datamodel.html#object.__init_subclass__) (something I just learned exists a few days ago), allowing you to pass keyword arguments upon inheriting. Then you could do something like this:
```
class InstrumentImplementation(SCPImixin, Instrument, SCPI_unknown=True):
```

bilderbuchi, 2022-12-24:
> I'd keep the current code working (though `includeSCPI=True` will raise a deprecation warning).

:+1:

> > It feels kinda "ugly" to need a SCPIbaseUnknownMixin to declare the absence of something
> 
> This SCPIbaseUnknownMixin does declare SCPI base commands, but also states, that it is unknown, whether the commands work or not. Those instruments, which do not have SCPI commands, won't have any Mixin class.

ah of course, that makes sense. 

> > Do we (can we, even?) need to check user-defined instruments for this?
> 
> I don't see a simple way, how to change the default value of `includeSCPI` (removing the SCPI commands basically inverts the default value) without breaking user-defined instruments, which do not declare it explicitly.

Agreed, which is why I raised this. Maybe we'll find a way. Maybe `__init_subclass__` is it.

> > Do we need to prescribe the order of base classes
> 
> Afaik, the order of classes is important, once you have name clashes. Therefore, the `Instrument` should be the last in the list of classes (such that the particularities override the general implementation).

I think we can make sure that we have no name clashes in the core classes/mixins/traits supplied by pymeasure. Then user-derived instruments should work regardless of base class order in most cases.

PR #905 (2024-03-01): SCPI base class
This PR adds a SCPI base class to supplant `includeSCPI=True`.
- Closes #804
- Closes #587 
- See #825 

With this transition I propose to remove SCPI attribute as well (at the end).

ToDos
- [x] Implement a base SCPI instrument
- [x] Ensure correct working
- [x] All instruments in the codebase with (implicit or explicit) `includeSCPI=True` should inherit one of the mixins.
- [x] All instruments with a custom `check_errors` should transition to use `fetch_next_error`, if applicable. Done in #1024 

Transition path:
1. step
- `includeSCPI=True` (whether explicit or implicit as t [... cut]
closed 2024-03-01 by BenediktBurger