pymeasure/pymeasure #807: Migrating Prologix / AR488 from 0.10. to 0.11. VI_ERROR_RSRC_NFOUND

pignoniG, 2022-12-28:
Hi, I'm having an issue after updating to 0.11, I understand that the implementation of the PrologixAdapter has changed, previously I was creating an adapter with this code:

`adapter = PrologixAdapter(port='/dev/cu.usbserial-AC008NYI',baudrate=115200,serial_timeout=1)`
now this doesn't work anymore and as far as I understand from the tutorial it should be updated to:
`adapter = PrologixAdapter('/dev/cu.usbserial-AC008NYI')`

This doesn't work and cannot find a solution, this is the error that I get:
`---------------------------------------------------------------------------
VisaIOError                               Traceback (most recent call last)
/var/folders/s_/k_qsdmpn09ng7xgzq1lrkjbm0000gn/T/ipykernel_13296/2630458825.py in <module>
      6 import math
      7 import time
----> 8 sourcemeter = Keithley2000("GPIB::16")
      9 generator = VISAAdapter("GPIB::4")

~/opt/anaconda3/lib/python3.9/site-packages/pymeasure/instruments/keithley/keithley2000.py in __init__(self, adapter, **kwargs)
    438 
    439     def __init__(self, adapter, **kwargs):
--> 440         super().__init__(
    441             adapter, "Keithley 2000 Multimeter", **kwargs
    442         )

~/opt/anaconda3/lib/python3.9/site-packages/pymeasure/instruments/instrument.py in __init__(self, adapter, name, includeSCPI, **kwargs)
    149         try:
    150             if isinstance(adapter, (int, str)):
--> 151                 adapter = VISAAdapter(adapter, **kwargs)
    152         except ImportError:
    153             raise Exception("Invalid Adapter provided for Instrument since "

~/opt/anaconda3/lib/python3.9/site-packages/pymeasure/adapters/visa.py in __init__(self, resource_name, visa_library, preprocess_reply, **kwargs)
     96                 del kwargs[key]
     97 
---> 98         self.connection = self.manager.open_resource(
     99             resource_name,
    100             **kwargs

~/opt/anaconda3/lib/python3.9/site-packages/pyvisa/highlevel.py in open_resource(self, resource_name, access_mode, open_timeout, resource_pyclass, **kwargs)
   3282                     )
   3283 
-> 3284             res.open(access_mode, open_timeout)
   3285 
   3286             for key, value in kwargs.items():

~/opt/anaconda3/lib/python3.9/site-packages/pyvisa/resources/resource.py in open(self, access_mode, open_timeout)
    280             constants.StatusCode.success_device_not_present
    281         ):
--> 282             self.session, status = self._resource_manager.open_bare_resource(
    283                 self._resource_name, access_mode, open_timeout
    284             )

~/opt/anaconda3/lib/python3.9/site-packages/pyvisa/highlevel.py in open_bare_resource(self, resource_name, access_mode, open_timeout)
   3207 
   3208         """
-> 3209         return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
   3210 
   3211     def open_resource(

~/opt/anaconda3/lib/python3.9/site-packages/pyvisa/ctwrapper/functions.py in open(library, session, resource_name, access_mode, open_timeout)
   1849     # [ViSession, ViRsrc, ViAccessMode, ViUInt32, ViPSession]
   1850     # ViRsrc converts from (str, unicode, bytes) to bytes
-> 1851     ret = library.viOpen(
   1852         session, resource_name, access_mode, open_timeout, byref(out_session)
   1853     )

~/opt/anaconda3/lib/python3.9/site-packages/pyvisa/ctwrapper/highlevel.py in _return_handler(self, ret_value, func, arguments)
    224                 session = None
    225 
--> 226         return self.handle_return_value(session, ret_value)  # type: ignore
    227 
    228     def list_resources(

~/opt/anaconda3/lib/python3.9/site-packages/pyvisa/highlevel.py in handle_return_value(self, session, status_code)
    249 
    250         if rv < 0:
--> 251             raise errors.VisaIOError(rv)
    252 
    253         if rv in self.issue_warning_on:

VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.`

Thanks for the help!

msmttchr, 2022-12-28:
In 0.11 Prologix is now inherited from pyvisa, your original code should work with following comments:
- `baudrate` should be renamed `baud_rate`
- instrument address should be conform to pyvisa addressing names

I think you are facing the second problem at the moment. Are you able to list pyvisa resources in your system?

pignoniG, 2022-12-28:
Hi, if I run pymeasure.instruments.list_resources()
this is what I get:
`0 : ASRL1::INSTR : Not known`
it might be the first time I actually try to use a VISA resource since before I was using the previous implementation
Should a com port appear there or do I need to add it manually? 
Thanks

msmttchr, 2022-12-28:
I think you could try using address "ASRL1::INSTR" and baud_rate=115200 and this should work, if ASLR1 is the address of your instrument.

pignoniG, 2022-12-28:
adapter = PrologixAdapter("ASRL1::INSTR",baud_rate=115200, timeout=1000)
worked
but then gave me this error 
`---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
/var/folders/s_/k_qsdmpn09ng7xgzq1lrkjbm0000gn/T/ipykernel_14345/225292594.py in <module>
     11 #adapter = PrologixAdapter('ASRL/dev/tty.usbserial-AC008NYI', 4)
     12 adapter = PrologixAdapter("ASRL1::INSTR",baud_rate=115200, timeout=1000)
---> 13 sourcemeter = Keithley2000(adapter.gpib(16))
     14 generator = Instrument(adapter.gpib(4),"PH5190")
     15 time.sleep(1)

~/opt/anaconda3/lib/python3.9/site-packages/pymeasure/adapters/prologix.py in gpib(self, address, **kwargs)
    189         :returns: PrologixAdapter for specific GPIB address
    190         """
--> 191         return PrologixAdapter(self, address, **kwargs)
    192 
    193     def _check_for_srq(self):

~/opt/anaconda3/lib/python3.9/site-packages/pymeasure/adapters/prologix.py in __init__(self, resource_name, address, rw_delay, serial_timeout, preprocess_reply, **kwargs)
     82             warn("Use 'timeout' in ms instead", FutureWarning)
     83             kwargs['timeout'] = serial_timeout
---> 84         super().__init__(resource_name,
     85                          asrl={
     86                              'timeout': 500,

~/opt/anaconda3/lib/python3.9/site-packages/pymeasure/adapters/visa.py in __init__(self, resource_name, visa_library, preprocess_reply, query_delay, log, **kwargs)
     98             self.resource_name = getattr(resource_name, "resource_name", None)
     99             self.connection = resource_name.connection
--> 100             self.query_delay = resource_name.query_delay
    101             return
    102         elif isinstance(resource_name, int):

AttributeError: 'PrologixAdapter' object has no attribute 'query_delay'`

if I add
`self.query_delay = rw_delay `
in  def __init__() in prologix.py 
all seems to work

pignoniG, 2022-12-28:
could this be that as this is not a original prologix adapter (it's an arduino with https://github.com/Twilight-Logic/AR488 ) it gets recognised differently by visa (as a generic serial device)?

msmttchr, 2022-12-28:
I think, this is linked to a problem in 0.11 described in #760. This issue is now fixed in the current code base, but it's obviously present in release 0.11. In #760, you should find also a workaround that avoids to change the pymeasure code.
Additionally, you find some alternative way of specifying address according to pyvisa needs, e.g. something like ASLR/dev/...
Finally, I don't think the problems you are facing are linked to the fact that your adapter is not a genuine Prologix, in all cases if you spot any relevant difference, you could try to customize the adapter for Arduino and possibly submit a PR in the future.

pignoniG, 2022-12-28:
Thanks for the help, I will try the next update but now it seems to be working fine.

bilderbuchi, 2022-12-31:
I just released version 0.11.1, please check if this fixes this issue for you.

pignoniG, 2022-12-31:
Hi, yes it works without editing prologix.py, but I think the documentations is out of date, the example for a prologix adapter still says:
```
from pymeasure.adapters import PrologixAdapter
adapter = PrologixAdapter('/dev/ttyUSB0')
```
Otherwise, thanks for the help!

bmoneke, 2023-01-01:
@pignoniG thanks for pointing out outdated documentation. I did not find that line in a quick search, could you please tell me, where you found that documentation?

pignoniG, 2023-01-01:
Hi @bmoneke that particular part was in [https://pymeasure.readthedocs.io/en/latest/tutorial/connecting.html#using-adapters](url) But I think also [https://pymeasure.readthedocs.io/en/latest/api/adapters.html#prologix-adapter](url) should be updated, the parameters are still for the previous version, if I'm not mistaken.

bmoneke, 2023-01-01:
Thanks @pignoniG, that helps a lot. I see, we have to touch the tutorial and best the docstring as well.

PR #813 (2023-01-03): Prologix adapter documentation adjusted to recent changes
The Prologix adapter documentation (and tutorial) is adjusted to the recent change of the base adapter (serial to visa).

Closes #807
closed 2023-01-03 by bmoneke