pymeasure/pymeasure #815: Discourage the use of direct setting of interface specific properties in Instruments

dkriegner, 2023-01-04:
Directly setting interface specific communication parameters as suggested in the documentation in the [Single Interface](https://pymeasure.readthedocs.io/en/latest/dev/adding_instruments.html#single-interface) section:

e.g.:
```
def __init__(self, adapter, name="Extreme 5000", baud_rate=2400, **kwargs):
    super().__init__(
        adapter,
        name,
        baud_rate=baud_rate,
        **kwargs
    )
```

leads to the disadvantage that this device can only be used with this particular connection. Even if a device might offer only a specific hardware interface many transparent converters exist (e.g. RS232 to Ethernet) which can make the device visible at the control computer under a different interface. 

With the `VisaAdapter` one would only need to change the resource name and a connection would be possible if the syntax proposed in the [multiple interfaces section](https://pymeasure.readthedocs.io/en/latest/dev/adding_instruments.html#multiple-interfaces) of the documentation is used.

For the example above, this would mean:

```
def __init__(self, adapter, name="Extreme 5000", baud_rate=2400, **kwargs):
    super().__init__(
        adapter,
        name,
        asrl={'baud_rate': baud_rate},
        **kwargs
    )
```

I therefore would propose to discourage using the "single interface" way and thereby allow the user to use hardware adapters more easily.

In particular for serial device communication I could anyhow only find the [LakeShore421 device](https://github.com/pymeasure/pymeasure/blob/9013d6f6d2c5666063b30d6b0bbe8d34deae859d/pymeasure/instruments/lakeshore/lakeshore421.py#L63) which uses the "single interface"-way

Other solutions:
@bmoneke pointed out in #800 that a user can always specify an Adapter and therefore also the single interface implementation would allow to be used with communication Adapters. While this is true, it requires the user to reapply all other default parameters like `read/write_termination, timeout, ...` which typically are unchanged if a transparent converter is used.

LongnoseRob, 2023-01-04:
Comment from the side-line:
`pyvisa` manages certain properties on resource-level such as `read\write_termination` these should not be globally overwritten. --> further detail [here](https://pyvisa.readthedocs.io/en/latest/introduction/communication.html#getting-the-instrument-configuration-right)

bilderbuchi, 2023-01-08:
As I'm unfamiliar with these ethernet-serial converters: As I understood it, they are addressed via a `TCPIP...` (in any case, _not_ `ASRL...`) VISA resource string. However, if they "transparently" convert, where do they take the baud rate to choose from? 
IIUC, your proposed way of doing `aasrl={'baud_rate': baud_rate}` would mean that converter does not get that baud rate information at all.

dkriegner, 2023-01-09:
Maybe "transparent" is not a good word here. 

We use varies types of such converters in our lab to converge all the communication to Ethernet. The different types work all in the same way: 
1) One needs to configure the serial interface parameters: baudrate, databits, stopbits, ... (typically via a web interface)
2) After that, they are addressed via `"TCPIP.."` VISA resource strings without the need to perform any particular settings of the communication. 

With transparent I referred to the fact that with this setup above every byte sent to the TCP/IP port is sent further on the serial interface.

So in fact when using such converters it's sort of indented to ignore the `asrl` dict.

bilderbuchi, 2023-01-09:
Hmmm, I see. So, probably a [Moxa](https://www.moxa.com/en/products/industrial-edge-connectivity/serial-device-servers/general-device-servers) NPort or something like that?
With your converter, when using `baud_rate=baud_rate`, you get an error from pyvisa that for TCPIP connections this is an invalid argument, correct?

bmoneke, 2023-01-09:
One point to consider for consistency:
If we put the baud rate into that dict. Why not also the read / write termination?
The term chars can (and do) differ for different connection types to the same instrument.

For your implementation you need "globally" defined term chars and "local" (that dict) baud rate, right?

Another solution: Could we add an option to ignore excessive kwargs in the Visa Adapter?

bilderbuchi, 2023-01-09:
> If we put the baud rate into that dict. Why not also the read / write termination?
> The term chars can (and do) differ for different connection types to the same instrument.

That's a good point/question.

> Another solution: Could we add an option to ignore excessive kwargs in the Visa Adapter?

We could feed the kwargs to pyvisa and pop-off-nonexistinent-kwargs, log and retry if the instantiation fails, until it works. We'll also probably have scary log messages from pyvisa. I'd prefer if we find a "cleaner" way to deal with this than swallowing kwargs.

bilderbuchi, 2023-01-09:
> While this is true, it requires the user to reapply all other default parameters like `read/write_termination, timeout, ...` which typically are unchanged if a transparent converter is used.

> * One needs to configure the serial interface parameters: baudrate, databits, stopbits, ... (typically via a web interface)
> 
> * After that, they are addressed via `"TCPIP.."` VISA resource strings without the need to perform any particular settings of the communication.

Wait, I'm confused. You say that the (serial) communication parameters are set via web interface (on the translator). But then you also say that _all other default parameters_ have to be reapplied -- why? you also say that you don't need any particular settings?

bmoneke, 2023-01-09:
The cleanest solution is to create your adapter instance manually and hand that to the instrument.
For that you have to know the instrument details, though and give them manually (at least some) to that adapter.

Through this discussion, the manual adapter creation seems for me the best way for these converters. (I know, that it is not the most convenient way)

bmoneke, 2023-01-09:
As all the connection information is in the init, it should not be difficult to create adapter instances for your devices.

msmttchr, 2023-01-09:
> As all the connection information is in the init, it should not be difficult to create adapter instances for your devices.

I fully agree with this approach which is on line with PyMeasure custom adapter use cases.

bmoneke, 2023-01-10:
I have an idea, how we could facilitate the use of an external adapter (like Prologix or in this case):

Instruments have their connection settings in a dictionary at class level.
If started with a resource name, the instrument takes this dict for the visa Adapter.
Any user may use this dictionary easily for instantiating their own adapter (ignoring some parts of that dict, if necessary).

Would that solution be fine with you @dkriegner?

dkriegner, 2023-01-10:
I try to catch up with the discussion:

> Hmmm, I see. So, probably a [Moxa](https://www.moxa.com/en/products/industrial-edge-connectivity/serial-device-servers/general-device-servers) NPort or something like that?

I do not know this particular one, but it looks comparable to what we are using.

> With your converter, when using baud_rate=baud_rate, you get an error from pyvisa that for TCPIP connections this is an invalid argument, correct? Can you show such an error?

[image]
sorry for the screenshot.

> Wait, I'm confused. You say that the (serial) communication parameters are set via web interface (on the translator). But then you also say that all other default parameters have to be reapplied (when using a separate Adapter instance) -- why? you also say that you don't need any particular settings of the communication?

With the other default parameters I mean the terminators. Since the Serial/Ethernet translator passes through everything 1:1 these are the same for this instrument independent of the used hardware connection. I agree that there are cases where depending on the hardware connection these are different, but this seems already solved since one can specify `read/write_termination` in the `asrl` (and equivalent) dictionaries.
A test for this is already in place:
https://github.com/pymeasure/pymeasure/blob/9013d6f6d2c5666063b30d6b0bbe8d34deae859d/tests/instruments/test_connection_configuration.py#L61-L62

The solution proposed by @bmoneke sounds promising to me. 
The only thing I am wondering is if you do not replace one dictionary which you seem to dislike by another one?
Code might get even more complicated with the proposed solution as it is with when using the MultiProtocol Adapter approach.

If you prefer to take no action (or even disallow code as in PR #800) then it just makes the user code duplicating the terminators in case such an adapter is used. I can live with that, but would find the already existing multiprotocol way the best solution.

dkriegner, 2023-01-10:
One more note: IMHO this Serial/Ethernet translators can't be compared with the Prologix Adapter. The Prologix adapter has many commands it processes on its own via the same interface used for the device communication.

bmoneke, 2023-01-10:
I mentioned the PrologixAdapter, as it has to be created manually (and requires the information).

My problem with your solution is, that it is unclear, why some parameters should be defined for all connection types (term chars) and others only for specific connection types (baud_rate). What do we do with the timeout? Is it global or specific? The ethernet connection adds lag and therefore might require a larger timeout.

With my solution, it is clear, that you define globally, what all (or the only) connection type require and specifically, what only some connection types require.

dkriegner, 2023-01-10:
Ok, I agree.  Likely many Instruments need to adapt this new behavior but this can likely be done step by step.

Regarding timeout: Indeed this is a valid concern, but it can't be solved in pymeasure. The added lag strongly depends on which kind of translator is used. Cheap ones we had added >100ms, while the better ones are almost negligible (few ms). But changing timeout does not require to define one's own Adapter, so I think it is a minor issue. -> So I would put it to the global dictionary except a specific device requires it otherwise.

bilderbuchi, 2023-01-10:
> With my solution, it is clear, that you define globally, what all (or the only) connection type require and specifically, what only some connection types require.

However, isn't this already possible _right now_, if you pass 
```python
...
asrl={'baud_rate': baud_rate}, # only for asrl
read_termination='\r\n',  # global
...
```
arguments? 
I'm not sure I see what the additional indirection through a class-level dict (that gets overridden by regular kwargs if needed) brings in functionality? :confused:

bmoneke, 2023-01-10:
> However, isn't this already possible right now, if you pass

Yes. My solution (in comparison to dkriegners) keeps that mechanics as is, but makes it more simple to create an adapter manually (because you have all the information available in one point).
In fact, with that, you could use a SerialAdapter easily if you liked.

Therefore we do not need to change, how we define the connection.

I hope that made it clearer.

dkriegner, 2023-01-11:
I must say the more I think about it, the more I like the proposed solution. It defines a central place to define common Adapter arguments while still maintaining that they can be overwritten as needed.

The proposed solution basically replaces (most of these) `setdefault `statements, which are now in many devices:
```
def __init__(self, adapter, name="Extreme 5000", **kwargs):
    kwargs.setdefault('timeout', 120000)
    kwargs.setdefault('read_termination', '\n')
    super().__init__(
        adapter,
        name,
        **kwargs
    )
```

And at the same time makes it easier to use different Adapters (which now requires studying the code of the Instrument).

bilderbuchi, 2023-01-11:
> I hope that made it clearer.

It did not clear up the question of how it improves the situation 
> that it is unclear, why some parameters should be defined for all connection types (term chars) and others only for specific connection types (baud_rate).

bmoneke, 2023-01-11:
The original request of this issue was, to put some parameters globally and others specifically, due to the requirements of a specific converter.

I propose to follow the documentation: define globally, if possible, and specifically if needed (due to different connection possibilities).

bilderbuchi, 2023-01-11:
So, if we go with an `Instrument.adapter_kwargs` dict as #828 proposes, how does that resolve the original problem of having general kwargs in single-interface instruments that trip up tcpip translator devices?

If the instrument now has `SomeInstr.adapter_kwargs = {'baud_rate': 2400}` instead of having a default value in a signature, this still fails with a `TCPIP...` resource string. 
So, you have to check the contents of that dict, and strip out the baud_rate arg before passing it to your adapter ctor. You save the effort of looking in the signature, or the init code for arguments.

However, _all other users_ now lose the ability to recognize from the signature (and thus also from Python's help facilities) that there is a baud_rate setting with a default, that might be adapted by them.
If you want to preserve that info, now you have to _duplicate_ that default setting in both the signature _and_ your class-level kwarg, to preserve the advantage when creating a separate adapter (that needs a certain subset of those args).

Honestly, from my point of view I'm not sure if the proposed remedy does not have more of a negative impact than thought. 

I'm slowly thinking that the way with less impact might be to, as the issue titles suggests, discourage giving protocol-specific connection configuration _without_ the protocol selector, i.e. guide people to write `asrl={'baud_rate': baud_rate}` instead of `baud_rate=baud_rate`. That will take some education on which args are fully generic (termination, e.g.), but it would not add indirection, duplicate information, and would solve (afaict) the original problem of creating a `TCPIP` connection with a translator device inbetween.

bilderbuchi, 2023-01-11:
> I propose to follow the documentation: define globally, if possible, and specifically if needed (due to different connection possibilities).

Ah, I thought you proposed to switch to the class-level adapter_kwargs way?

dkriegner, 2023-01-12:
> So, you have to check the contents of that dict, and strip out the baud_rate arg before passing it to your adapter ctor. You save the effort of looking in the signature, or the init code for arguments.

I believe there is still a misunderstanding here. As far as I understand #828, `baud_rate` is interface specific and therefore never should go into the class-level `adapter_kwargs` (at least not as `baud_rate: 2400`). It therefore would also not be lost from the call signature.
What should not be in the call signature are the parameters like: "read_termination", "write_termination", "timeout". I think there is little danger of duplication if done right, but identifying what needs to go where could be challenging.

It's true that a user with a translator has to look into the code/manual in any case to identify the settings for the translator.
But with #828 one should be able to assume that:
```
adapter = PrologixAdapter("resource_name", address=5, **MyInstrument.adapter_kwargs)
inst = MyInstrument(adapter)
```
works, and in your code you could still write 

```
def __init__(self, adapter, name="My Instrument", baud_rate=2400, **kwargs):
    super().__init__(
        adapter,
        name,
        baud_rate=baud_rate,
        **kwargs
    )
```
avoiding the "asrl" dict.

That being said: I am of course not against my original proposal. It would certainly be a much more minor change.

dkriegner, 2023-01-12:
Ok, if you want really everything in the `adapter_kwargs` I agree with @bilderbuchi that duplication is unavoidable and clearly a negative thing. I thought that `adapter_kwargs` will only contain what you need for all adapters.

bilderbuchi, 2023-01-15:
My yardstick for what should go into the signature: Parameters that the end user can be expected to want to tweak. Having those in the signature directly informs the user about which "knobs" are available to turn.

Unfortunately, due to the large variation in instrument (hardware) implementations we are interfacing with, what exactly can/should be changed varies from instrument to instrument...
* A primary candidate would be adjusting a `timeout`, as people have faster/slower or more or less capable serial adapters/PC serial peripherals, and want reliable and as fast as possible communication
* Similar reasoning for `baud_rate`. Some instrument hardware can be configured with what they can deal with (so you'd adjust accordingly on the pymeasure side). However, some others might have it hard-coded, then it does _not_ make sense to put this in the signature, as changing it would render comms nonfunctional.
* Similar case for `read_termination`/`write_termination`. Some instrument hardware can configure it (by used transport method, even), for some it's hard-coded. For some the termination chars differ by interface, for some it's the same across them all.

Add to that the fact that we want implementations/interfaces to be as clear as possible (to reduce confusion/bug potential/duplication), and as flexible as possible (to accomodate use cases of varying "exoticness"), you can imagine that there's a certain tension, and resulting challenge in identifying a "good" design approach/implementation guidelines.

dkriegner, 2023-01-15:
I understand (and agree with) your reasoning!

So I think it boils down to what I originally proposed: Keep the core code as is with all its flexibility.
If you want to help supporting user-defined `Adapters` it would likely be good to discourage what is suggested in the single interface part of the documentation. This would not change anything regarding the possible call signatures of the `Instrument`. It merely would mean to sometimes have one dictionary more than absolutely needed inside the `Instrument.__init__`.

bmoneke, 2023-01-15:
User defined adapters ignore the connection settings in the instrument, therefore it does not matter, whether it is defined as single connection or multi connection.

dkriegner, 2023-01-15:
> User defined adapters ignore the connection settings in the instrument, therefore it does not matter, whether it is defined as single connection or multi connection.

Of course you are right. Given that most of the time one will use "just" a different Visa resource name the single connection definition basically disables this option.

bmoneke, 2023-01-15:
I don't understand you.

If you have
~~~
adapter = SomeAdapter(...)
Inst = MyInst(adapter...)
~~~
The adapter does not get anything defined in "MyInst.__init__".

dkriegner, 2023-01-15:
Yes, but I think what most user want to do is just:

`Inst = MyInst("<DIFFERENT_VISA_RESOURCE>")`

In this way I do not need to repeat the other settings of the communication which are in my version done set in the `__init__`

bmoneke, 2023-01-15:
The problem for you is, that baud_rate is no valid tcp parameter, right?

Baud rate is a parameter for the ASRL connection type only, isn't it?
Maybe all parameters which are invalid for some connection type should be defined specifically (documentation)?
That means baud rate has always to be defined in ASRL dict, while therm chars and timeout are used by all connection types.
You can always override a parameter (with values for your converter), but you cannot delete a parameter which from inst.init.

dkriegner, 2023-01-15:
Exactly!

bmoneke, 2023-01-15:
Another idea : the visa Adapter removes any kwarg, which is set to None.
That way you can override the baud rate.

This saves us from touching many instruments.

We can adjust the documentation as well to improve future code.

bilderbuchi, 2023-01-16:
> Maybe all parameters which are invalid for some connection type should be defined specifically (documentation)?
> That means baud rate has always to be defined in ASRL dict, while therm chars and timeout are used by all connection types.

Yes, I think recommending this would be viable and not too onerous. It's clearer (as `baud_rate` remains "scoped" to the `asrl` resourcetype). 
So far we did not do this because for single-interface instruments we did not see using another resourcetype, so having to use `asrl` if there's no other option anyway feels (felt) redundant.

However, with this
> Another idea : the visa Adapter removes any kwarg, which is set to None.
That way you can override the baud rate.

I think we have a winner, it's a great solution! The change is minimal, should not affect any other code that I can think of, and importantly @dkriegner can use this with any instruments that don't scope `baud_rate` to the `asrl` resource type.

The question remains, which of these three do we implement:
a) remove any kwargs with value `None` in the processing for Adapter instantiation, and add a Note about this mechanism in the docs
b) Change the docs to recommend putting resourcetype-specific kwargs like `baud_rate` _always_ into a respective dict
c) Change all existing instruments in the codebase?

IMO 
* definitely a)
* b) if nobody disagrees (it does make implementations a bit more complicated)
* c) I don't know (leaning towards yes if we do b), to be consistent with the docs

bmoneke, 2023-01-16:
I'm for a and b.
We can do all the instruments together with the Scpi mixin(#804) and name kwargs (#658).

dkriegner, 2023-01-17:
This sounds like a good solution.

Just a side note: All devices which use interface specific settings (even if the are fixed in the device) will have to expose them on their public interface (of `__init__`). So for example a device with fixed `baud_rate` should **not** be implemented like this:

```
def __init__(self, adapter, name="My Instrument", **kwargs):
    super().__init__(
        adapter,
        name,
        baud_rate=2400,
        **kwargs
    )
```

because one would not be able to set `baud_rate` to None. Or am I mistaken?

Not sure if such cases are present in the current code base, but for a device with fixed settings I guess this would have been done like this until now.

bmoneke, 2023-01-17:
> Or am I mistaken?

Oh yes. If `super().__init__(baud_rate=123, **kwargs)`, we cannot do `Instrument(baud_rate=None)`, as that would give twice that kwarg to the super().__init__.

Consequently, no connection detail should be in the super init call, such that you may override every setting.

N.B: If it is `super.__init__(asrl={'baud_rate': 123}, **kwargs)`, then `Instrument(baud_rate=None)` will override the fixed one. (and here we are at the beginning again...)

bilderbuchi, 2023-01-17:
> will have to expose them on their public interface (of `__init__`).

..not necessarily in the signature, via setdefault should be sufficient (and preferable if the arg value should not be changed, to not clutter the signature)

bmoneke, 2023-01-17:
> But then we don't need the override in the first place, because it does not interfere with tcpip devices, right?

Yes, that is what I meant, that we are at the beginning (of the thread).

However, now we have found out (through a lot of discussion and ideas), which parameters should be where: General ones can be globally defined (setdefault) and connection specifics in the appropriate dicts.

bilderbuchi, 2023-01-18:
> connection specifics in the appropriate dicts.

..which should also be defined via setdefault if they are "adjustable", so that they can be overridden.

bmoneke, 2023-01-18:
If they are in the dicts, they can be overriden (individually): The VISAAdapter uses `setdefault` to inject the dict entries to the kwargs. Therefore, if a (VISAAdapter) kwarg is already 'baud_rate', the one defined in the 'asrl' dictionary is going to be ignored.

bilderbuchi, 2023-01-18:
Ah, you are right, I forgot about that part 😅

bmoneke, 2023-01-25:
TODO:
Adjust the documentation, that general parameters can be globally defined (via kwargs.setdefault) and connection specifics in the appropriate dicts (e.g. `asrl={'baud_rate': 12345}`.