pymeasure/pymeasure #816: Toptica iBeamSmart issue

waveman68, 2023-01-04:
While some basic functionality is implemented, there is an issue with writing. The laser sends a prompt, to indicate that it is ready for writing as a handshake:

```
CMD>
```

Currently, read is implemented to wait for just `[OK]`.

I would be happy to make a PR, but first need to better understand the pymeasure philosophy. I see a few options where to put this:

- in a State (e.g., ready for write) + 
- in read
- in write

Does anybody have some guidance?

Thanks
Sam

dkriegner, 2023-01-04:
I am not the right one to comment on the pymeasure philosophy, but I have originally implemented this device (have no access to it anymore).

I do not understand the problem. Yes the device wants to send a prompt but the TopticaAdapter turns this feature off in its init method (line 52):

https://github.com/pymeasure/pymeasure/blob/20fe65307432ce4156ea115a0a9c43d5c77431fa/pymeasure/instruments/toptica/adapters.py#L43-L56

So there should be no need to read the prompt when the device is opened using the TopticaAdapter.

bmoneke, 2023-01-04:
Good to see people who know the device and who have it. Would you @waveman68 and @dkriegner be willing to support me moving the logic from the TopticaAdapter to the instrument itself?
That would enable us to write tests for that device.

Thanks @dkriegner for helping out.

dkriegner, 2023-01-04:
As I said I have no hardware access anymore. But I think it should be straight forward to move the logic into the Instrument since the core logic improved since the time this Instrument was made. The reason the Adapter was made was (quote from #352)

> So this device's protocol has some similarities with the Attocube/anc300 device merged in PR https://github.com/pymeasure/pymeasure/pull/327. Since I received some comments this could be implemented via the check_errors functionality I attempted this at first without actually making a dedicated adapter. It worked, but IMHO was not very elegant. It felt especially awkward that I had to reimplement Instrument.values because otherwise my read/write/ask methods would have been bypassed. Also every property had to use check_set/get_errrors=True. [The check_errors implementation can be seen here](https://github.com/dkriegner/pymeasure/blob/ibeam/pymeasure/instruments/toptica/ibeamsmart.py).

Unfortunately it seems I deleted the branch which contained this other implementation.
But `values` now for sure does not need reimplementation and checking of the acknowlegdement message can IMHO also remain in `read`.

bmoneke, 2023-01-04:
I opened a PR #819 with my (already some months ago started) change to the VISAAdapter. If you want, @waveman68 you can support me or start from there.

@dkriegner your commits are still in the PR #352 , so nothing is lost.

dkriegner, 2023-01-13:
I now had remote access to the IBeamSmart and can confirm that some changes since I submitted this device broke the Instrument. 

(The problem in the current master is that something changed in the `super().write()` used in the `__init__` which causes the reported issue)

But #819 fixes the issue for me! I will contribute some changes in #819

bmoneke, 2023-01-13:
Good to know, that we are on the right track.

PR #819 (2023-01-19): Remove Toptica adapter
Remove the Toptica Adapter and make the ibeamSmart depending on the VISAAdapter alone.
Takes hints from original PR #352 and the new issue #816.

The tests require empty strings, which are added to the ProtocolAdapter in #818 

I added a few tests from the examples @dkriegner gave in #352.

Fixes #816