pymeasure/pymeasure #821: Critical Bug:  Use of %f instead of %e for values truncates values

maor1993, 2023-01-05:
Hey, I've been working with this library for a while using the DSOX1102G I noticed that once I use a value smaller than 1e-6, the scope does not receive it correctly and uses the smallest value possible.

This issue occurs as the python default %f max decimal points is 6:
[image]

The more concise solution would be to use %e which is always supported by SCPI devices:
[image]

Looking at the repo there are about 19 devices use %f and not %e

This is uncaught by testing as it was never tested for.

bilderbuchi, 2023-01-05:
Thanks for raising this! I agree that %g is probably the more sensible default choice.

For all instruments where you want to move away from %f, you'll have to check (with a device if available, or in the manual) if the resulting numbers format is accepted by the device.

maor1993, 2023-01-05:
Thanks for the quick reply.
the list is as follows:
[image]

I think I might have a moment to check the programming manuals, will update.

maor1993, 2023-01-05:
Okay done:
| device         	| link                                                                                                                  	| supports exp? 	| Notes                         	|
|----------------	|-----------------------------------------------------------------------------------------------------------------------	|---------------	|-------------------------------	|
| AWG401x        	| ?                                                                                                                     	| ?             	| Can't find programming manual 	|
| agillent33220A 	| http://ecelabs.njit.edu/student_resources/33220_user_guide.pdf                                                        	| YES           	|                               	|
| agilent33500   	| https://www.testequipmentdepot.com/usedequipment/pdf/33500b-33600a-series_manual.pdf                                  	| YES           	|                               	|
| agilent33521A  	| https://www.testequipmentdepot.com/usedequipment/pdf/33500b-33600a-series_manual.pdf                                  	| YES           	|                               	|
| anritsuMS9710C 	| https://dl.cdn-anritsu.com/en-au/test-measurement/files/Manuals/Operation-Manual/MS9710C_W1580AE_remote_opm_e_4_0.pdf 	| YES           	|                               	|
| danfysik8500   	| https://www.danfysik.dk/media/1182/software-manual-sys8500.pdf                                                        	| NO?           	|                               	|
| keithley24xx   	| https://download.tek.com/manual/2400S-900-01_K-Sep2011_User.pdf                                                       	| YES           	|                               	|
| keithley6517b  	| https://download.tek.com/manual/6517B-901-01D_Feb_2016.pdf                                                            	| YES*          	| some fields do and some don't 	|
| keysightDSOXxx 	| https://www.keysight.com/il/en/assets/9018-07554/programming-guides/9018-07554.pdf                                    	| YES           	|                               	|
| keysightN7776C 	| https://www.keysight.com/il/en/assets/9018-70058/programming-guides/9018-70058.pdf                                    	| YES           	|                               	|
| ips120_10      	| https://muquip.triumf.ca/sites/default/files//instruction_manuals/mnl_spec_bNMR_oxford_powerSupply_IPS-120-10.PDF     	| NO            	| Uses decimal point only       	|
| itc503         	| http://lmu.web.psi.ch/docu/manuals/bulk_manuals/OxfordInstruments/ITC503.pdf                                          	| NO            	| Uses decimal point only       	|
| cnt91          	| https://www.manualslib.com/manual/1375130/Pendulum-Cnt-90.html                                                        	| YES           	|                               	|
| sr570          	| https://www.thinksrs.com/downloads/pdfs/manuals/SR570m.pdf                                                            	| NO            	|                               	|
| sr830          	| https://www.thinksrs.com/downloads/pdfs/manuals/SR830m.pdf                                                            	| NO            	|                               	|
| sr860          	| https://www.thinksrs.com/downloads/pdfs/manuals/SR860m.pdf                                                            	| NO            	|                               	|
| ibeamsmart     	| ?                                                                                                                     	| ?             	| Can't find programming manual 	|

LongnoseRob, 2023-01-05:
@maor1993 AGW401x - is probably [this product](https://www.activetechnologies.it/products/signal-generators/arbitrary-waveform-generators/arb-rider-awg-4000/).
But I could not find a public available programming manual on manufacturers website

bmoneke, 2023-01-05:
Let's put a hint in #779 regarding small numbers.

bmoneke, 2023-01-05:
Thanks for looking up the manuals.

Would you mind opening a PR with the changes, @maor1993 ?

You do not have to change all instruments possible.
I'd guess, that it does not occur that often, that you want to set a value to below 1e-5. And if it occurs, we can still fix that device (with a person being able to test the change).

It is more important, that devices continue to work (down to 1e-5) than to break at any input. If you're in doubt, whether it will work for a given instrument, leave as is.

closed 2026-08-17 by BenediktBurger