pymeasure/pymeasure #823: File structure for instrument traits / default instruments

bmoneke, 2023-01-08:
As we plan to introduce traits / default implementations for instrument classes (#416, spectrum analyzer for example #580), we have to think, where to place these files.

I propose to put these files in a folder "instruments/traits".

We could call them "genérica" as well, but as some of them might be just parts (for example Scpi base instrument), traits fits better.

bilderbuchi, 2023-01-08:
Depending on their number, a module `instruments/traits.py`and/or `instruments/standard_types.py` (e.g.) might be sufficient.
So far any sub_packages_ below `instruments` are all instrument manufacturers, a `traits` subpackage would go against that.

bmoneke, 2023-01-08:
Good point. "Genérica" or "pymeasure" would be a manufacturer and would be better than to put a generic spectrum analyzer in the root path (where only the "building blocks" (instrument, channel...) live

bilderbuchi, 2023-01-08:
I'd put a generic spectrum analyzer into `instruments/generic_types.py`? And aren't the generic types "building blocks" (for the manufacturer-specific instruments)?

Side note: I'm sure it's not intended as such here, but please no diacritics in package/module names. :-)

bmoneke, 2023-01-08:
Auto correction at its best...

bmoneke, 2023-01-09:
> I'd put a generic spectrum analyzer into `instruments/generic_types.py`? And aren't the generic types "building blocks" (for the manufacturer-specific instruments)?

Mhmm. They are builiding blocks, but could also be used standalone (if a device implements the SCPI set).