pymeasure/pymeasure #824: Summary table of content for "adding instruments"

bmoneke, 2023-01-08:
As "adding instruments" documentation grows (even more with upcoming traits / generica), I propose to start the file with a short summary linking to the main parts (properties, channels, traits).

bilderbuchi, 2023-01-08:
I'd rather rip the "main parts" apart into multiple documents, to be honest. (the good thing is, Sphinx is smart about headings, in every file the heading levels are relative, so there is no need to re-flow/adapt all section depths.
Maybe generate a sub-table of contents if that's possible?

bmoneke, 2023-01-08:
My point is, that users should know the main features of pymeasure instruments in the beginning and not have to read everything regarding adding instruments first.

I thought about something along the lines of (obviously more refined than that): "Pymeasure instruments use byte based communication (or ASCII chars) via different adapters. For ease of use, we have property creators and Channel implementations defined. 
Commonly used properties (for example for a spectrum analyzer) are available as a trait. "

Whether it is one ore more files, does not matter for that purpose.

(although I agree, that splitting the file might be good).

bilderbuchi, 2023-01-08:
Ah I see. I agree.

bmoneke, 2023-01-09:
Done in #826

closed 2023-01-09 by bmoneke