pymeasure/pymeasure #834: Pyvisa-py `flush` raises NotImplementedError

bmoneke, 2023-01-18:
We could (and should) add the error handling in the `VISAAdapter.flush_read_buffer` method, not here, until pyvisa-py is updated. Please feel free to open an issue (or directly a PR).

The instrument will work flawlessly without this fix. I'll resolve.

_Originally posted by @bmoneke in https://github.com/pymeasure/pymeasure/pull/819#discussion_r1073511548_

bilderbuchi, 2023-01-18:
Related upstream PR: https://github.com/pyvisa/pyvisa-py/pull/208#issue-494956830

dkriegner, 2023-01-18:
New issue raised upstream: https://github.com/pyvisa/pyvisa-py/issues/348

dkriegner, 2023-03-18:
small update: pyvisa-py for TCP sockets now has a flush method: https://github.com/pyvisa/pyvisa-py/pull/350
I am afraid for other resource types this won't be so easy.