pymeasure/pymeasure #837: Instrument.read_bytes(-1) does not behave as advertised

dkriegner, 2023-01-20:
The docstring of read_bytes says:

https://github.com/pymeasure/pymeasure/blob/c04f21157d96e5c73e18b025b4b82c77ff2431e7/pymeasure/instruments/instrument.py#L145-L149

Upon testing (see below) it, however, turns out that it only returns the data until a termination character is contained. (The termination character is not stripped). If no termination character is received, `read_bytes(-1)` raises a `pyvisa.errors.VisaIOError`.

I tested this with pyvisa (pyvisa-py) with the code below:

```
from pymeasure.instruments.mksinst import MKS937B
g = MKS937B("TCPIP::10.0.131.254::10001::SOCKET")  # termination_char here is ";"
g.write("SN?")
g.write("SN?")

g.read_bytes(-1) # returns b'@253ACK022869254 ;'
g.read_bytes(-1) # returns b'FF@253ACK022869254 ;'
g.read_bytes(-1) # raises the error below (but the remaining "FF" which is still in the read buffer can not be obtained.
```

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\AJAlab\Documents\GitHub\pymeasure\pymeasure\instruments\instrument.py", line 153, in read_bytes
    return self.adapter.read_bytes(count, **kwargs)
  File "C:\Users\AJAlab\Documents\GitHub\pymeasure\pymeasure\adapters\adapter.py", line 121, in read_bytes
    read = self._read_bytes(count, **kwargs)
  File "C:\Users\AJAlab\Documents\GitHub\pymeasure\pymeasure\adapters\visa.py", line 161, in _read_bytes
    return self.connection.read_raw(**kwargs)
  File "C:\Users\AJAlab\AppData\Roaming\Python\Python310\site-packages\pyvisa\resources\messagebased.py", line 405, in read_raw
    return bytes(self._read_raw(size))
  File "C:\Users\AJAlab\AppData\Roaming\Python\Python310\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\AJAlab\AppData\Roaming\Python\Python310\site-packages\pyvisa_py\highlevel.py", line 519, in read
    return data, self.handle_return_value(session, status_code)
  File "C:\Users\AJAlab\AppData\Roaming\Python\Python310\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```

dkriegner, 2023-01-20:
If the code is adapted to behave like the docstring suggests then `VisaAdapter.flush_read_buffer` could be simplified. see https://github.com/pymeasure/pymeasure/pull/836#issuecomment-1398295698

bilderbuchi, 2023-01-20:
Thanks for spotting this inconsistency!
I don't have a strong opinion on which way is the better one. However, I think we should also be consistent among our _Adapter_ types (to make interchanging them easier), and `SerialAdapter._read_bytes(-1)` also reads up to a termination character (if one is defined). So, maybe the VISAAdapter (and all others) should do the same thing?

bmoneke, 2023-01-21:
I think it is best, if `(-1)` reads the whole buffer. I could set the read_termination temporarily to nothing, such that `read_raw` returns the whole buffer.

> SerialAdapter._read_bytes(-1) also reads up to a termination character (if one is defined). So, maybe the VISAAdapter (and all others) should do the same thing?

In SerialAdapter I mimicked our default, the VISAAdapter, as I tried to make them consistent. I want to keep them consistent per default.

Question: What do you think about an option to read until term_char or the whole buffer (similar to pyvisas read_bytes, which can break at a term_char; I'd use their parameter name)?

bilderbuchi, 2023-01-21:
Hm, why not. Is it probable that users will need both behaviours? We should avoid that people need to implement their own logic for "read bytes until termchar" -- is that something that is needed?

Another question: have we inadvertently changed behaviour of read_bytes (in SerialAdapter and VISAAdapter) compared to earlier (before the Adapter refactor)?

What would you put as the default value? So that the behaviour is unchanged?

bmoneke, 2023-01-21:
> have we changed behavior? 

The serial adapter did not have term chars. If you do not define them, the behavior is the same as before.

The visa Adapter did just "read bytes". So here reading "-1" bytes is a new feature as well.

Reading up to a term_char is programmatically more elegant in both adapters:
- visa can read up to a term char natively with "read raw". Reading the whole buffer (with a term char defined) is more complicated. 
- pyserial can read until a char as well. If you want to read everything, you read as many times as possible and concatenate all the results. Each of the reads has its own timeout timer and you read until timeout. All together it can take a lot longer than the defined timeout value. 

However, as reading bytes might be about binary data and is often without term chars, it seems more natural to read the whole buffer.

> Is it probable that users will need both behaviours? 

I do not think they will need to read until term char often, as you normally read either ASCII (until term char) or a number of bytes. 

Together I think we can return the whole buffer at - 1.

bilderbuchi, 2023-01-22:
Thanks for elaborating! I agree.

PR #839 (2023-01-28): Adapters' read_bytes reworked
Closes #837 

As discussed: Default is to **not** break at a term char, but the option remains. Changes are implemented in all adapters, which have `read_bytes`: VISA, Fake, Serial, VXI11.

Tests are added accordingly.

The first commit contains the changes.
The second commit contains some cosmetic docstring changes.
closed 2023-01-28 by bmoneke