pymeasure/pymeasure #855: Do not allow kwargs in property creators

bmoneke, 2023-02-10:
We noticed, that removing kwargs from `setting` caught some errors. ( #843)

We could deprecated the usage of "kwargs" in control and measurement, adding a "v_kwargs" parameter, which is given to values as kwargs.

Additionally, all parameters of "values" have to be added to the signature as well. (spyder already composes the signature, including called methods, too) 

That way we keep flexibility (you can hand on arbitrary arguments to values and then to ask), but we catch many errors in normal cases.

What do you think?

bilderbuchi, 2023-02-12:
> but we catch many errors in normal cases.

Which errors are you thinking about, here? Doesn't `values` already barf when it receives unknown/incompatible arguments? Remember that in #843 the kwargs were silently discarded, which afaik cannot happen anymore now.

bmoneke, 2023-02-12:
Yes, values barks, but only if you call it (in a test or real live). 
The property creators, however, are called at import time, so at least with "test all instruments",  they bark, if a parameter is wrong. 

A wrong spelled parameter name (for example 'validatores'), will not be noticed (I think) until a test is written, as 'control' thinks, it is a values kwarg.

"setting" notices such a mistake, however.

bmoneke, 2023-02-12:
I thought about misspelled or not applicable parameters.

bmoneke, 2023-02-13:
For example, importing this Instrument will raise an error with the `setting`, not with the `measurement`.

~~~python
from pymeasure.instruments import Instrument

class Test(Instrument):

    test_get = Instrument.measurement("cmd", "doc", wrong_parameter=7)

    test_set = Instrument.setting("cmd", "doc", wrong_parameter=7)
~~~

bilderbuchi, 2023-02-15:
OK, I see the attraction, good points! There are some things I'd rather change in #856, I'll offer feedback there.

PR #856 (2023-02-21): Remove control kwargs
Closes #855 

This PR deprecates kwargs in the property creators in order to catch wrong used arguments, for example validators in a `measurement`.

The raised warning helped to remove some superfluous code.

This warning (and later TypeError) will reduce review effort.
closed 2023-02-21 by bmoneke