pymeasure/pymeasure #857: Add support for the Teledyne LeCroy HDO6xxx oscilloscope

RobertoRoos, 2023-02-20:
I would like support for the [LeCroy HDO6054B oscilloscope](https://teledynelecroy.com/oscilloscope/oscilloscopemodel.aspx?modelid=11553). I already have an old private package that controls it, so I hope to be able to port this myself.

Note that that the product is called "Teledyne Lecroy", and I see `pymeasure` has both instrument folders for Teledyne and Lecroy, so I'm not sure where to put it. Any suggestions?

And I hope I can reuse some of the logic from https://pymeasure.readthedocs.io/en/latest/api/instruments/lecroy/lecroyT3DSO1204.html.

I expect that the HDO6xxx family can be controlled the same way, but I am not sure and I cannot test them all. How should I name and describe the device in `pymeasure`?

RobertoRoos, 2023-02-20:
Actually, the interface for the T3DSO1204 might be identical. I cannot find a different programming interface for the HDO6054B, and my existing code uses mostly the same commands (but the long spelling instead of the short one).

bilderbuchi, 2023-02-20:
Welcome!
> Note that that the product is called "Teledyne Lecroy", and I see pymeasure has both instrument folders for Teledyne and Lecroy, so I'm not sure where to put it. Any suggestions?

This happens regularly, with instrument manufacturers going through mergers/renames/acquisitions etc. We normally go with what it says on the  nameplate, figuring that is what users will look for first. So, if your device front says "Teledyne Lecroy", you can add another manufacturer tree `teledynelecroy` -- I think this should now cover all bases :sweat_smile:.
 
It's very good to reuse logic, as this avoids code duplication -- please see the docs here: https://pymeasure.readthedocs.io/en/latest/dev/adding_instruments/properties.html#instruments-with-similar-features
First, you could check if you can operate your device with the T3DSO1204 class as expected -- this could confirm that is can deal with the short spelling too, that would be ideal. 
If not, you can try the other way round and compare with the T3DSO1204, if that can deal with the long spelling. (Those long/short names_should_ be interchangeable, but not all devices actually work like that).

> I expect that the HDO6xxx family can be controlled the same way, but I am not sure and I cannot test them all. How should I name and describe the device in pymeasure?

You can check in the respective manuals to see what the differences are -- if the main differences are parameter ranges or some missing functionality, it can be easy to split out at least a base class for common functionality `HDO6xxx` or `HDO6000`, and derive/extend your specific model class `HDO6054B` from that. See [here](https://pymeasure.readthedocs.io/en/latest/dev/adding_instruments/properties.html#instrument-family-with-different-parameter-values). 
If you end up mostly testing the "main" family class code with your instrument, too, that would be enough for us. 

If you have further questions, we're here to assist. :-)

bmoneke, 2023-02-21:
Regarding the name:
If it says Teledyne Lecroy, I'd put it into Teledyne, because people tend to start to compare at the beginning.
Nevertheless, both manufacturer's page says, that there is also the other page to look for instruments, so either way is good.

RobertoRoos, 2023-02-27:
I've done some work, and the interface is about 95% similar to the existing T3DSO1204. I will look into making an abstract base class for LeCroy oscilloscopes.

And Teledyn emailed back that this manual is appropriate for the HDO6000: https://cdn.teledynelecroy.com/files/manuals/maui-remote-control-automation_27jul22.pdf

PR #868 (2023-10-09): Added instrument Teledyne HDO6xxx
Fixes #857

Adding support the Teledyne oscilloscope. I am not done yet, but I would like to hear if this approach is acceptable: I am currently extending the `LeCroyT3DSO1204` class, instead of for example creating a mutual base class.

According to Teledyne, the leading manual for Visa interfacing is: https://cdn.teledynelecroy.com/files/manuals/maui-remote-control-automation_27jul22.pdf

There are some discrepancies between the `LeCroyT3DSO1204` device and my `TeledyneHDO6xxx`, which I can fix by overriding some method/properties. But I was also checking th [... cut]
closed 2023-10-09 by bmoneke