pymeasure/pymeasure #858: Add support for the Velleman K8090 relay board

RobertoRoos, 2023-02-21:
I would like to add support for the Velleman K8090, an 8-channel relay board that communicates using serial.

Like my other issue, I already have old code that I would like to port. This issue is mostly to indicate my work-in-progress.

Please let me know if this device is not popular enough and there is no interest to merge it, which I could understand.

bmoneke, 2023-02-21:
Even if it were a `control`, on setting a value, you only send and do not read (unless you use `check_set_errors` and redefine `check_errors` to read).

You should find a solution, which does not leave a filled buffer, for example:
- you could read always and catch the (possible, but expected) timeout error, if no change happens (maybe with a shorter timeout)
- you could write your command and clear the read buffer after a short time, such that the buffer is empty afterwards.

RobertoRoos, 2023-02-22:
Thanks for the suggestion. I'd made a draft in the meantime with the opposite approach: let the device fill the buffer on the `switch`, and when reading again, read the entire buffer but only use the last bit we're expecting.

This prevents additional blocking in the `switch` action. Do you think this is acceptable?

bmoneke, 2023-02-22:
What a stupid design idea, that a command may have a response or not, such that you never know, whether you have to read for a response or not...

> I'd made a draft in the meantime with the opposite approach: let the device fill the buffer on the switch, and when reading again, read the entire buffer but only use the last bit we're expecting.

Yes, it is acceptable.

A variation of your proposal:
After setting `switch`, you always send another command. Then you have an answer for sure (from the second command) and you can determine whether you received one or two answers.
With this option, you do have to make this distinction only with the `switch` command and not in all read actions, **and** you are able to know, whether the switch command changed anything (some answer) or not.

PR #859 (2023-03-06): Added Velleman K8090 relay device
Fixes #858
closed 2023-03-06 by bmoneke