pymeasure/pymeasure #862: `TypeError` on SerialAdapter `read_bytes(-1)`

RobertoRoos, 2023-02-27:
When I use `self.read_bytes(-1)` from my instrument using a `SerialAdapter`, I get the error:

> C:\Program Files\Python310\lib\ctypes\__init__.py:66: TypeError

With the traceback:

```
..\..\..\pymeasure\instruments\velleman\velleman_k8090.py:247: in read
    response = self.read_bytes(-1)
..\..\..\pymeasure\instruments\instrument.py:153: in read_bytes
    return self.adapter.read_bytes(count, **kwargs)
..\..\..\pymeasure\adapters\adapter.py:123: in read_bytes
    read = self._read_bytes(count, break_on_termchar, **kwargs)
..\..\..\pymeasure\adapters\serial.py:108: in _read_bytes
    return self.connection.read(1e99 if count == -1 else count, **kwargs)
..\..\..\venv\lib\site-packages\serial\serialwin32.py:278: in read
    buf = ctypes.create_string_buffer(n)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

init = 1e+99, size = None

    def create_string_buffer(init, size=None):
        """create_string_buffer(aBytes) -> character array
        create_string_buffer(anInteger) -> character array
        create_string_buffer(aBytes, anInteger) -> character array
        """
        if isinstance(init, bytes):
            if size is None:
                size = len(init)+1
            _sys.audit("ctypes.create_string_buffer", init, size)
            buftype = c_char * size
            buf = buftype()
            buf.value = init
            return buf
        elif isinstance(init, int):
            _sys.audit("ctypes.create_string_buffer", None, init)
            buftype = c_char * init
            buf = buftype()
            return buf
>       raise TypeError(init)
E       TypeError: 1e+99

C:\Program Files\Python310\lib\ctypes\__init__.py:66: TypeError

```

It looks like reading 1e99 bytes no (longer?) works.

bmoneke, 2023-02-27:
Oh.
Creating a that large buffer is beyond many computers...

Thanks for raising that issue.

So we have to reduce the size of the bytes to read.
Maybe we can read a certain number, let's say 1000, and read afterwards all bytes which are in the buffer (via in_waiting).
In total it should be only one timeout time (as we know how many bytes are to read the second time), but we wait at least some time for a device response.

RobertoRoos, 2023-02-27:
Sounds like a good suggestion.

Won't `in_waiting` always be 0 though, when a timeout occurs?
And if there are 1001 bytes in the buffer, we would immediately read the 1000, followed by the additional 1 byte, but never end up waiting a timeout duration.

I was thinking of reading in chunks instead:

```
            data = bytes()
            buffer_empty = False
            chunk_size = 256
            while not buffer_empty:
                chunk = self.connection.read(chunk_size, **kwargs)
                data += chunk
                buffer_empty = len(chunk) < chunk_size  # If fewer bytes got returned, we had a timeout
            return data
```

I think this results in only a single timeout.

bmoneke, 2023-02-27:
Very good ideas!

I like that `in_waiting + 1000`.
For really odd cases we could assert, that `in_waiting == 0` afterwards (here reduced to 100, which should be enough:

~~~python
result = read(in_waiting + 100)
if in_waiting:
    result += read(in_waiting)
return result
~~~

RobertoRoos, 2023-02-28:
Eh, my nitpicking might start to annoy soon, but the second clause won't do much. In case data is still actively coming in right after `read(in_waiting + 100)`, then the current buffer queue size `in_waiting` does not say enough. I.e., after `result += read(in_waiting)` there could still be more.

Let's go for just the static read or some kind of `while`-structure. Preference?

bmoneke, 2023-02-28:
> Eh, my nitpicking might start to annoy soon

We're trying to find a good solution. Better to talk about a good solution beforehand than to have failing code.

> Let's go for just the static read or some kind of while-structure. Preferenc

We had such a structure with `return b"\n".join(self.connection.readlines(**kwargs))`. I tried to get rid of it (reasoning in the PR comment: https://github.com/pymeasure/pymeasure/pull/839#issuecomment-1403373579 ) as it reads several times (until newline) with several possible timeouts.

After this long discussion, it seems to be the best idea...

PR #866 (2023-03-06): Changed behavior of `read_bytes(-1)` for `SerialAdapter`
Fixes #862 

There are a couple of ways to handle the read-all operation. I don't think it matters too much, as long as the read timeout does not elapse more than once.
closed 2023-03-06 by bilderbuchi