pymeasure/pymeasure #863: `VISAAdapter` still terminating on default term character in `read_bytes(-1)`

RobertoRoos, 2023-02-27:
Pretty odd and specific issue, not sure if this belong here or on PyVisa.

When I try to read the complete buffer in a serial connection using the `VISAAdapter`, it still breaks on the byte corresponding to `\n`:

```
    def __init__(self, adapter, name="Velleman K8090", timeout=1000, **kwargs):
        super().__init__(
            adapter,
            name=name,
            asrl={"baud_rate": 19200},
            write_termination="",
            read_termination=chr(self.BYTE_ETX),
            timeout=timeout,
            **kwargs,
        )
      
    # ...

    def read(self):
        response = self.read_bytes(-1)

        # `response` will end with "\n", even though there are more bytes in the buffer!
```

bmoneke, 2023-02-27:
> read_bytes() currently has a bit of an odd structure:

What is odd with that structure?

> It looks like that when self.connection.read_termination is equal to None (in the last clause of the if-statement), a default temchar is used instead of none at all.

Oh, that should not be, at least if I understand VISA correctly:
~~~python
self.set_visa_attribute(constants.ResourceAttribute.termchar, ord(self.LF))
            self.set_visa_attribute(
                constants.ResourceAttribute.termchar_enabled, constants.VI_FALSE
            )
~~~

RobertoRoos, 2023-02-27:
Using the application com0com I have this minimal example:

```
#
# Use a loopback to create two linked serial devices: COM3 and COM4
#

import pytest

from serial import Serial
from pymeasure.adapters import VISAAdapter

@pytest.fixture()
def adapter():
    return VISAAdapter(
        "ASRL3::INSTR",
        visa_library="@py",
        asrl={"baud_rate": 19200},
        timeout=3000,
        read_termination=chr(0x0F),
        write_termination="",
    )

@pytest.fixture()
def loopback():
    return Serial("COM4", baudrate=19200)

def test_regular(adapter, loopback):
    """Regular read with fixed number of bytes."""
    loopback.write(
        bytearray.fromhex("0471FF")
    )
    result = adapter.read_bytes(3)

    assert len(result) == 3

def test_all(adapter, loopback):
    """Regular read with -1 number of bytes."""
    loopback.write(
        bytearray.fromhex("0471FF")
    )
    result = adapter.read_bytes(-1)

    assert len(result) == 3

def test_newline(adapter, loopback):
    """Regular read with -1 number of bytes."""
    loopback.write(
        bytearray.fromhex("04710AFFFFFFFFFF")
    )
    result = adapter.read_bytes(-1)

    assert len(result) == 8
```

There are two issues: in `test_all` there is a timeout instead of a full read and in `test_newline` the newline is used as terminating character.

RobertoRoos, 2023-02-27:
> Oh, that should not be, at least if I understand VISA correctly:

Yeah, I was stepping through the code and `termchar_enabled` really does get disabled. I can't see where this goes wrong.

bmoneke, 2023-02-27:
I think I have the culprit:
https://github.com/pyvisa/pyvisa-py/blob/b410a6e20a8593ceb515a880a22b2e0aa3c8a4e1/pyvisa_py/serial.py#L122-L168

It reads the asrl_end (which seems to be set with the term_char in the `read_termination` I quoted above) and therefore it stops reading, as it is the "end_checker" method.
Right now, `suppress_end_check` is always "False (second to last parameter in the `_read` call)".

So, either I misunderstand pyvisa/VISA, or pyvisa-py needs an update.

> Using the application com0com I have this minimal example:

Thanks for the code.
It is easier to understand the "string formatting" for bytes. For example `b"8123vbla\n"` is a bytes object with the appropriate ASCII chars (last one is a newline). If you have to define a bytes value (e.g. byte 0), you can use b"\x00".

bmoneke, 2023-02-27:
Could you check the behaviour with another backend? (NI for example?)

bmoneke, 2023-02-27:
I opened a request for desired behaviour.

bmoneke, 2023-02-27:
pyvisa's maintainer responded with a suggestion to look at the VISA attributes in pyvisa-py and NI and see if there is a difference.

Probably pyvisa needs a fix.

bmoneke, 2023-02-28:
As there seems to be no upstream solution (different definition of `read_raw` than expected), I created a PR solving the question manually.

PR #867 (2023-03-11): VISAAdapter fix read_bytes(-1)
Fixes #863 

Goal: test expected `read_bytes(-1)` behaviour and make it work.

- First commit is a test (which should be added anyways).
- Second commit is a fail proof (though not very elegant) solution to read the whole buffer.

Drawback of the solution: It takes at least one timeout period.
closed 2023-03-11 by bilderbuchi