pymeasure/pymeasure #872: The `preprocess_reply` argument in `Instrument.__init__()` is no longer used in a `Instrument.values()` call

mcdo0486, 2023-03-08:
Previous to the PR https://github.com/pymeasure/pymeasure/pull/718 rework , `Instrument.values()`  would return `Adapter.values()`.  After the rework `Instrument.values()` is implemented in `CommonBase.values()` and no longer calls `Adapter.values()`.

Previously as part of an Instrument's initialization the argument `preprocess_reply` argument would be passed to the `Adapter` which in turn would be used as `self.preprocess_reply`  in the `Adapter.values()` function:
https://github.com/pymeasure/pymeasure/blob/8bf3c295fa2f977145ea42d633da8f16a7cc4151/pymeasure/adapters/adapter.py#L184

Now `CommonBase` doesn't ingest the `preprocess_reply` argument and so `CommonBase.values()` doesn't create or assign `self.preprocess_reply`:
https://github.com/pymeasure/pymeasure/blob/8bf3c295fa2f977145ea42d633da8f16a7cc4151/pymeasure/instruments/common_base.py#L312

That means when an Instrument driver passes along a `preprocess_reply` argument as part of Instrument initialization, that argument has no effect when `Instrument.values()` is called.

Two instrument drivers in the code base use that functionality:

https://github.com/pymeasure/pymeasure/blob/8bf3c295fa2f977145ea42d633da8f16a7cc4151/pymeasure/instruments/fluke/fluke7341.py#L67

https://github.com/pymeasure/pymeasure/blob/8bf3c295fa2f977145ea42d633da8f16a7cc4151/pymeasure/instruments/signalrecovery/dsp7265.py#L72

bmoneke, 2023-03-09:
Thanks for raising the issue.

bmoneke, 2023-03-09:
@bilderbuchi shall we add a (deprecated) global `preprocess_reply` to `CommonBase` for backward compatibility for instruments not in pymeasure?

Right now, if an external instrument hands `preprocess_reply=something` to the Instrument's constructor, a deprecation warning is raised in the VISAAdapter (if they do not provide their own one), but the Adapter's `values` method is never called, so this parameter is useless.

The best transition process would be to add the global preprocess to the `CommonBase` (with a deprecation warning) and then remove it again, whenever we cut those old ties.

bmoneke, 2023-03-09:
TODOs for this issue:

- [ ] General solution for instruments not in pymeasure?
- [ ] fix dsp7265, see #873 
- [ ] fix fluke instrument

bilderbuchi, 2023-03-11:
> @bilderbuchi shall we add a (deprecated) global `preprocess_reply` to `CommonBase` for backward compatibility for instruments not in pymeasure?

Yes, this sounds like the cleanest approach, thanks.

PR #876 (2023-03-27): Add (deprecated) global preprocess reply
As this PR fixes a regression bug, the deprecation version is the original one (0.11), not the next version.

The `Channel` class does not allow to use the global `preprocess_reply`, as Channels were added after the deprecation.

Closes #872