pymeasure/pymeasure #879: Allow property setter to read

bmoneke, 2023-03-16:
Some instruments respond to every command, they even acknowledge setting a value.
Right now, the solution is to set `check_set_errors=True` and to read in `check_errors`, which seems more a hack than a solution.

Some solutions:

1. Previously (in some discussion) I proposed to add a further parameter to `control`, which switches between `write` and `ask` in the setter.
Advantage is, that you can choose it for every property.
Disadvantages are, that we have another parameter and that you have to add it to every single property.
2. `control`'s setter calls `set_property` (name to be determined). This method will just call `write` in the default implementation. An instrument might override this method in order to read afterwards and interpret the read response (which might be an acknowledgment or an error).
Advantages: it is declared only once; no additional parameter for `control`; you can raise an appropriate error "setting property xyz failed" (some instrument stores the last written command in order to be able to raise such an error!).
Disadvantage: You cannot do it for a subset of properties. This should be no problem, as mostly an instrument responds to all setters or to no one.

I like the second solution and would like to implement it. The first solution and current hack are given as alternative solutions.

What are your ideas?

bilderbuchi, 2023-03-26:
> instruments respond to every command, they even acknowledge setting a value.

> the read response (which might be an acknowledgment or an error).

I'm not sure I'm sold on this, these things sound like error checking to me, which sounds like it fits the use case of `check_errors` perfectly, and not hackish? :shrug: 
What am I missing?

> raise an appropriate error "setting property xyz failed"

Are you sure about this? That `set_property` method will only know the `command` it got passed, and not which _property_ caused this, no? And again, raising an error sounds like a job for `check_errors`?

bmoneke, 2023-03-27:
As far as I understand, `check_errors` **requests** (i.e. writes something) a list of all errors.
These instruments might have a special "give me all errors" command to read the current error state.

In this case the errors are part of the response.

Obviously we could ignore the errors in the response, but that does not seem to be good, either.

bilderbuchi, 2023-03-27:
The default one, yes -- I thought that such instruments need a specific implementation. I did not read with enough attention I think---I agree that a separate `read` in `check_errors` feels hackish.
 
Ideally, I'd like to find a way to extend/use the existing `check_set_errors`/`check_errors` mechanism: the purpose is similar (checking for errors/consuming acknowledge messages), and it's an established API, and we don't have to introduce additional moving parts.

If we can't find a way to do that, I prefer your approach #2. 
We could check for the presence of a callable with a given name (`write_ack`? `acknowledged_write`? something else), and if available use that instead of `write`. That way we could avoid having yet _another_  class method for a probably quite obscure/seldom-used feature -- our call stack is getting pretty deep and complicated, with all the optional modifications, mutations and checks. 
OTOH, discovery of that feature gets more obscure (but maybe that's not so bad?).

bmoneke, 2023-03-27:
> Ideally, I'd like to find a way to extend/use the existing check_set_errors/check_errors mechanism: the purpose is similar (checking for errors/consuming acknowledge messages), and it's an established API, and we don't have to introduce additional moving parts.

A suggestion: `check_xy_errors=True` will make sure that different methods will be called (`check_getter_errors`, `check_setter_errors`). In the default implementation, both will call `check_errors`
In the case of such an instrument, you woul redefine `check_setter_errors` as appropriate.

bilderbuchi, 2023-03-29:
Yeah, this sounds like a sensible approach. 

By the way, do we have concrete instrument models that would make use of this feature?

bmoneke, 2023-04-03:
Two instruments from #757 (one already as PR: #885 ) would profit from this change as well.

seb5g, 2023-04-18:
I actually stumbled on this issue myself trying to use the Ametek DSP 7270 instrument. The one currently written in PyMeasure does'nt work well for setting parameters because the protocol needs the NULL character to be read (and checked it has been replied) whenever a Control is set. The easiest way to make it work seemlessly was to replace write by ask in the fset. I'm new to PyMeasure so not sure to know what solutions would be best but I think it should be done somehow.

If I understand correctly the solution using check_errors is to reimplement it in a derivative class and just do a read?

By the way, I would be glad to do a pull request on the Ametek instrument as I found other issues, what is the policy about this (and the testing of the modified features?)

bmoneke, 2023-04-18:
> A suggestion: check_xy_errors=True will make sure that different methods will be called (check_getter_errors, check_setter_errors). In the default implementation, both will call check_errors
In the case of such an instrument, you woul redefine check_setter_errors as appropriate.

this is under work in a PR: #883

bmoneke, 2023-04-18:
> I actually stumbled on this issue myself trying to use the Ametek DSP 7270 instrument. The one currently written in PyMeasure does'nt work well for setting parameters because the protocol needs the [...]

Right now the official solution (see hcp TC038 as an example) suggests to set `check_set_errors=True` for the property and implement the necessary reading and error checking in `check_errors`. Quote from the [FAQ](https://pymeasure.readthedocs.io/en/latest/dev/adding_instruments/solutions.html#communication-protocol-issues):
> The instrument answers every message, even a setting command. You can set the setting’s check_set_errors = True parameter and redefine check_errors() to read an answer, see [hcp.TC038D](https://pymeasure.readthedocs.io/en/latest/api/instruments/hcp/tc038d.html#pymeasure.instruments.hcp.TC038D)

In #883 we are creating a specific `check_set_errors`, which would replace `check_errors`.

> If I understand correctly the solution using check_errors is to reimplement it in a derivative class and just do a read?

Exactly.

> By the way, I would be glad to do a pull request on the Ametek instrument as I found other issues, what is the policy about this (and the testing of the modified features?)

Create a PR as early as possible (maybe marking it as a draft, if it is really early stage). If you expect some discussions, because it touches core points of pymeasure, you might open an issue first (discussions in PRs do not work that well).

For tests, we made it possible to test that an implementation creates / interprets the right commands, see [protocol tests](https://pymeasure.readthedocs.io/en/latest/dev/adding_instruments/tests.html#protocol-tests).

Whenever you have questions, feel free to ask.

PR #883 (2023-05-17): Split check errors
The property creators call `check_get_errors()`, `check_set_errors()`, which default to `check_errors()`.
Closes #879 

Tests expanded accordingly.
Instruments make usage of that new feature.
In this circumstance: Attocube moved to the new Channels infrastructure (but backwards compatible).

Question: How is the default relationship between these three error checking functions and instrument and Channel?
- Should the Channel functions relate to the instrument's one or should all relate to `check_errors` and that in turn to the instrument's `check_errors` (right now implement [... cut]
closed 2023-05-17 by bmoneke