pymeasure/pymeasure #881: SR510 does not work

Bamgm14, 2023-03-28:
Tried using for SR510 but commands time out or do even execute. I'll give examples later, I made my own code after the the fact and it works fine. Just wanted to note. I plan to see if I can fix it (when I get the time)

bmoneke, 2023-03-28:
Thanks for raising the issue.
Please provide feedback, that we may fix it. We gladly support you in fixing it, if you want to.

Bamgm14, 2023-04-04:
Would be appriciated.
Basically its a timeout error.
`AttributeError: 'SR510' object has no attribute 'freq'
>>> a.frequency
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Bamgm14\AppData\Roaming\Python\Python310\site-packages\pymeasure\instruments\instrument.py", line 211, in __getattribute__
    return super().__getattribute__(name)
  File "C:\Users\Bamgm14\AppData\Roaming\Python\Python310\site-packages\pymeasure\instruments\instrument.py", line 430, in fget
    vals = self.values(command_process(get_command), **kwargs)
  File "C:\Users\Bamgm14\AppData\Roaming\Python\Python310\site-packages\pymeasure\instruments\instrument.py", line 317, in values
    results = str(self.ask(command)).strip()
  File "C:\Users\Bamgm14\AppData\Roaming\Python\Python310\site-packages\pymeasure\instruments\instrument.py", line 303, in ask
    return self.read()
  File "C:\Users\Bamgm14\AppData\Roaming\Python\Python310\site-packages\pymeasure\instruments\instrument.py", line 263, in read
    return self.adapter.read(**kwargs)
  File "C:\Users\Bamgm14\AppData\Roaming\Python\Python310\site-packages\pymeasure\adapters\adapter.py", line 107, in read
    read = self._read(**kwargs)
  File "C:\Users\Bamgm14\AppData\Roaming\Python\Python310\site-packages\pymeasure\adapters\visa.py", line 150, in _read
    return self.connection.read(**kwargs)
  File "C:\Users\Bamgm14\AppData\Roaming\Python\Python310\site-packages\pyvisa\resources\messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "C:\Users\Bamgm14\AppData\Roaming\Python\Python310\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\Bamgm14\AppData\Roaming\Python\Python310\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\Bamgm14\AppData\Roaming\Python\Python310\site-packages\pyvisa\ctwrapper\highlevel.py", line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\Bamgm14\AppData\Roaming\Python\Python310\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.`

bmoneke, 2023-04-04:
Did you try to increase the timeout value?
Do you give the correct VISA resource string to the device? (compare it with your own code)

bmoneke, 2023-04-04:
What version of pymeasure are you using (from pypi or from github)?
What pyvisa version do you have?
What VISA backend are you using?

Could you post the code you put together and the code you used to generate above error message?

If you send a command (setting some parameter), can you verify on the front panel (if it has any), that the command is processed correctly, even though the response does not work as expected?

How is the device connected? If via serial, is the baud_rate correct?

Bamgm14, 2023-04-13:
v0.11.1 
v1.7.x - v1.8.x (Not sure which subversion)
VISA MAX NI
The code was just a live query (a = SR510(...))
The Baud_Rate is the default rate (9600) for the apparatus tho I am not sure if that's the case with pymeasure so that is something I didn't take into consideration. 
(Sorry for the late reply, was busy)

bilderbuchi, 2023-04-16:
> From MAX NI test panel, I can request data from the device.

That's interesting, and good that we have a working solution. Now we find out what the difference is. 
Can you tell what messages the test panel sends and receives? Is there a logging or debug function to observe that?
For pymeasure, can you do 
```python
import logging
logging.basicConfig(level=logging.DEBUG)
# try to connect with instrument
``` 
before trying to talk to the instrument?
This way, we should get a bit more information, specifically what our Adapter writes, and then pyvisa.