pymeasure/pymeasure #884: ManagedDockWindow tutorial code in documentation results in an AttributeError

anupamGak, 2023-03-31:
The example code for ManagedDockWindow in the documentation [(link)](https://pymeasure.readthedocs.io/en/latest/tutorial/graphical.html#using-the-manageddockwindow) results in the following error message when I run it:

```
Traceback (most recent call last):
  File "E:\Programming Projects\Keithley DMM6500\example.py", line 68, in <module>
    window = MainWindow()
  File "E:\Programming Projects\Keithley DMM6500\example.py", line 47, in __init__
    super().__init__(
  File "C:\Users\Anupam\AppData\Local\Programs\Python\Python310\lib\site-packages\pymeasure\display\windows\managed_dock_window.py", line 88, in __init__
    self.browser_widget.browser.measured_quantities.update(measure_quantities)
AttributeError: 'list' object has no attribute 'update'
[Finished in 1.1s]
```

The issue seems to be buried too deep in layers of imports for me to easily track it down.

Any help is appreciated. Thanks!

bmoneke, 2023-03-31:
Thank you for raising the issue.
Maybe @bilderbuchi or @CasperSchippers may be able to help you, as they are more seasoned with the GUI.

I looked at the code and it seems to work: `measured_quantities` is defined as a `set()`, which has an `update` function. I was not able to find a redefinition of that variable.

A few more questions for the context:
- Which pymeasure version do you use? Pypi package or the current master?
- Which python version do you use?

anupamGak, 2023-03-31:
My Python is v3.10.5 with pymeasure v0.11.1 installed from Pypi.

I uninstalled pymeasure and reinstalled it from the current master, and doing so has solved this issue. My pymeasure version is currently '0.11.2.dev443+ge71107f'.

Thanks again!

closed 2023-03-31 by anupamGak

bmoneke, 2023-04-03:
Good to hear, that the current master fixes this bug.