pymeasure/pymeasure #891: Help installing cloned pymeasure repo with pip

jwbrooks0, 2023-04-15:
I'm working through the instructions [here](https://pymeasure.readthedocs.io/en/latest/dev/contribute.html) and I get to the instructions that say:
`cd /path/for/code/pymeasure`
`pip install -e .`

When I run `pip install`, it complains that there is no `setup.py` file.  I checked, and there is in fact no `setup.py` file anywhere in the repo.  Am I missing something??  

As an alternate method, I tried to add `path/for/code/pymeasure` to my PYTHONPATH (the usual way I "install" repositories), but I get the following error when I try to import.  
`XXXXX/GitHub/pymeasure/pymeasure/__init__.py:49: UserWarning: Could not find pymeasure version, it does not seem to be installed. Either install it (editable or full) or install setuptools_scm
  warnings.warn('Could not find pymeasure version, it does not seem to be installed. `

I'm by no means a python expert so maybe I'm missing something.  

I'm using GitHub Desktop, virtualenv, python 3.8.10.,  Ubuntu 20.04.6 LTS

jwbrooks0, 2023-04-15:
After more digging, [this website](https://towardsdatascience.com/setuptools-python-571e7d5500f2#:~:text=you%20can%20now%20create%20a%20dummy) recommended creating a _dummy_ setup.py file.  I did so, and the install seems to have worked.  

Is this the way it is supposed to work?  Should the contributing website be updated?  And why is it setup so I can't add the pymeasure directory to my PYTHONPATH?

bilderbuchi, 2023-04-16:
> When I run pip install, it complains that there is no setup.py file. I checked, and there is in fact no setup.py file anywhere in the repo. Am I missing something??

Yes, your pip is probably too old - installation without the legacy setup.py has been added quite some time ago (I don't know the exact version). I cannot reproduce this with pip 23.1, ` pip install -e .` completes as expected.

>  Either install it (editable or full) or install setuptools_scm 

Have you tried installing `setuptools_scm` as the message advises?