pymeasure/pymeasure #894: VisaAdapter.close should likely not close the pyvisa.ResourceManager

dkriegner, 2023-04-17:
I am attempting to write code in which an `Instrument` connection can be closed (the instrument potentially be powered off) and later reconnected to the still running code. For this purpose, I call `Instrument.adapter.close()` when I detect that the device is gone.
This, however, causes also other pyvisa connections in the same python session to fail since the `ResourceManager` is invalidated by `VisaAdaper.close` (line 141):

https://github.com/pymeasure/pymeasure/blob/fb08f797247fd9e63103cd394bbbcaaa436c6dfc/pymeasure/adapters/visa.py#L131-L143

Of course one could say this is a pyvisa bug since the `ResourceManager` in the different `VisaAdapter` instances should be independent (see upstream bug: pyvisa/pyvisa#742), but is this behavior in pymeasure intended? Since [`__del__` ](https://github.com/pymeasure/pymeasure/blob/fb08f797247fd9e63103cd394bbbcaaa436c6dfc/pymeasure/adapters/adapter.py#L65) of `Adapter` calls `close()` this means that no Instrument should be garbage collected as long as other instruments are still needed. This seems wrong to me.

For my purpose it seems sufficient to close the device connection which allows later to reopen it and keep other device connections open.

bmoneke, 2023-04-17:
The goal was, to clear the buffer, as we experienced strange behaviour, see [#839 ](https://github.com/pymeasure/pymeasure/pull/839#discussion_r1087137595).

bilderbuchi, 2023-04-17:
The relevant upstream issue is https://github.com/pyvisa/pyvisa-sim/issues/68 although [this comment](https://github.com/pyvisa/pyvisa-sim/issues/68#issuecomment-1404191801) suggests that it (or another issue) should still be investigated.

dkriegner, 2023-04-17:
Do I understand it correctly that the `manager.close()` in `Adapter.close()` was added to work around a bug in pyvisa/pyvisa-sim? (This might have deserved a comment in the code?!)

bilderbuchi, 2023-04-17:
Yes. And yes, you're probably right.
We might even be able to make this conditional so the RM closing only happens when we are using the `@sim` library (if the RM object exposes that info).

dkriegner, 2023-04-17:
Yes making this conditional to the `@sim` library would make a lot of sense. `ResourceManager.visalib.library_path` should give some hint on the used library.

I also do not understand the `AttributeError` which is caught in [Line 142](https://github.com/pymeasure/pymeasure/blob/fb08f797247fd9e63103cd394bbbcaaa436c6dfc/pymeasure/adapters/visa.py#L142). This likely can also only be triggered by the test (@sim) code?

bmoneke, 2023-04-17:
> I also do not understand the AttributeError which is caught in [Line 142](https://github.com/pymeasure/pymeasure/blob/fb08f797247fd9e63103cd394bbbcaaa436c6dfc/pymeasure/adapters/visa.py#L142). This likely can also only be triggered by the test (@SiM) code?

You can use the same connection (not only resource manager) for several instruments, e.g. because you use a single GPIB or RS485 line for several instruments. Each instrument has its own adapter. The first adapter closes the resource manager and all the adapter's don't have a resource manager anymore, so their `close` would raise the `AttributeError`.

dkriegner, 2023-04-17:
This is obviously a side issue, but how could I trigger the `AttributeError`? I do not see how closing the resource manager would remove the attribute from the adapter.

When I remove the `try/except` and directly call `adapter.close()` from different Instruments with the same `Adapter` object it seems to be no problem:

The following code runs without error:
```
import pymeasure.adapters
import pymeasure.instruments
adapter = pymeasure.adapters.VISAAdapter("TCPIP::localhost::50004::SOCKET")
i1 = pymeasure.instruments.Instrument(adapter, "1")
i2 = pymeasure.instruments.Instrument(adapter, "2")
i1.adapter.close()
del i1
# i2.write("test")  # this of course would fail
i2.adapter.close()
# i2.adapter.manager.close()  # this also can be called repeatedly. 
```

bmoneke, 2023-04-17:
In your example, you use the same adapter instance for both instruments.

The problem arises with several instances using the same connection. The most prominent example is the PrologixAdapter's `gpib` method. This method returns a new Adapter instance, but using the same physical connection:
~~~python
adapter1 = PrologixAdapter("some Resource string")
adapter2 = adapter1.gpib(5)

# alternatively:
adapter1 = VISAAdapter("some Resource string")
adapter2 = VISAAdapter(adapter1)  # new adapter using the same connection as adapter1
~~~

dkriegner, 2023-04-27:
Thanks for all the input. Also using your examples the problem actually is not inside the close routine, but it can be only triggered when `close()` is called from `__del__()`. I will propose changes to address this issue in a PR.

bilderbuchi, 2023-06-22:
No, you're right.

closed 2023-06-22 by bilderbuchi