pymeasure/pymeasure #913: Attocube Anc300 uses properties instead of method

bmoneke, 2023-05-16:
The following two properties should be methods instead, I think
~~~python
    stepu = Instrument.setting(
        "stepu %d",
        """Set the steps upwards for N steps. Mode must be 'stp' and N must be
        positive.""",
        validator=strict_range, values=[0, inf], check_set_errors=True)

    stepd = Instrument.setting(
        "stepd %d",
        """Set the steps downwards for N steps. Mode must be 'stp' and N must be
        positive.""",
        validator=strict_range, values=[0, inf], check_set_errors=True)
~~~

This requires a deprecation etc., therefore is not done in #883

bilderbuchi, 2023-05-16:
I agree, from what the docstring says, this could probably be a method `step(nr_steps)` or so. One interface/method, hides the detail that the commands are different, which the user does not care about.

bmoneke, 2023-05-17:
In fact, there is such a method "steps" which uses these properties...

I guess the initial author implemented each command as a property.

bilderbuchi, 2023-05-17:
Oh, might even be that I reviewed that 👼 😁 
So, close this?

dkriegner, 2023-06-27:
I am the original author. I think this was my first Instrument and it might be I got too overenthusiastic with the properties

Just removing these properties however seems not like the best option. The `move()` method is not a real replacement of these two properties since move does a bit more. So to keep the flexibility for the user move would need to gain an option to be non-blocking. I can propose something if thats welcome.

bmoneke, 2023-06-27:
Improvements are always welcome.
If you deem another name better, I'm open for suggestions. Maybe `move_raw` or something.

bilderbuchi, 2023-07-12:
Seems to be closed by #938

closed 2023-07-12 by bilderbuchi