pymeasure/pymeasure #950: Find USB device from hardware information

bmoneke, 2023-07-19:
The COM port of virtual Com ports via USB may change from time to time.
I propose a simple helper method, which retrieves the COM port of a USB virtual port according to hardware information (vendor id, product id, serial number).
This method allows to specify the hardware information and then to connect always to the correct device, independent of the COM port number assigned at that time.

Rough idea:

~~~python
from serial.tools import list_ports

def findPort(vid: int, pid: int, sn: str) -> str:
    """Find the port name ('COM5') of the first device with the given USB information.

    :param int vid: Vendor ID.
    :param int pid: Product ID.
    :param str sn: Serial number.
    :return: Port as a string as `device` attribute returned from `list_ports.comports`.
    """
    for port in sorted(list_ports.comports()):
        if port.vid == vid and port.pid == pid and port.serial_number == str(sn):
            return port.device
    raise AttributeError("No device found for the given data.")
~~~

bilderbuchi, 2023-07-22:
You'd use this to find the vid/pid/sn _once_, and then use that combination when instantiating instruments? I.e. `instr = TC038(findPort('a', 'b', 'c'))`? 
Mind that pyvisa-py will still not accept (I think) `COM5` as a resource name, so you'd have to find the full resource name, or manually create one from that. _Or_ you make this a bit "smarter" and use `list_resources` output to find the correct full one, and return _that_. 

Sounds like a good idea, could complement `resources.list_resources` nicely. Maybe name it `find_port`, feels like this would fit better into the naming scheme.

bmoneke, 2023-07-22:
> You'd use this to find the vid/pid/sn once, and then use that combination when instantiating instruments? I.e. instr = TC038(findPort('a', 'b', 'c'))?

exactly.

> Mind that pyvisa-py will still not accept (I think) COM5 as a resource name

For a pyvisa-py device, I use `findPort().replace("COM", "ASRL")`.

> Mind that pyvisa-py will still not accept (I think) COM5 as a resource name, so you'd have to find the full resource name, or manually create one from that. Or you make this a bit "smarter" and use list_resources output to find the correct full one, and return that.

good idea

> Sounds like a good idea, could complement resources.list_resources nicely. Maybe name it find_port, feels like this would fit better into the naming scheme.

I agree. I copied the code (legacy naming scheme, inspired by Qt) from my current working setup to show the idea.
A pymeasure implementetion will be reworked.

PR #982 (2023-10-25): Add find_serial_port to find a serial port by providing USB information
Closes #950 

I did not add "::INSTR" at the end of the resource string, which is optional.

Reasoning for adding it to instruments.__init__: There is potential to use it often, therefore import should be shorter.
Additionally, the module `resources` is already imported and the additional dependency `serial` was already loaded by pyvisa (if pyvisa-py is used), therefore, it should not increase import time.
closed 2023-10-25 by bmoneke