pymeasure/pymeasure #954: Gpib - USB prologix problem with read

lucort, 2023-07-23:
Hi,
I have a problem when using Prologix adapter (GPIB-USB). Communication is fine but my trouble is using read type function.
It seems that the character at the end is "\x00" (Null char) but the read stops before. Consequently in the buffer "\x00" is still here and, at the next query, it is sent to the reply. An example with some details will be more comprehensible:
untitled0.py:
```python
#!/usr/bin/env python3
# -*- coding: utf-8 -*-

from pymeasure.instruments.keithley import Keithley2400
from pymeasure.adapters import PrologixAdapter

# Connect and configure the instruments
adapterS = PrologixAdapter('ASRL/dev/ttyUSB4::INSTR', address=25)

sourcemeter = Keithley2400(adapterS)
```

Then at the prompt:
```python
runfile('/Scripts/untitled0.py', wdir='/Scripts')

sourcemeter.id
Out[2]: 'KEITHLEY INSTRUMENTS INC.,MODEL 2400,1304207,C31   May 28 2010 10:34:39/A02  /R/J'

sourcemeter.id
Out[3]: '\x00KEITHLEY INSTRUMENTS INC.,MODEL 2400,1304207,C31   May 28 2010 10:34:39/A02  /R/J'

sourcemeter.ask("*IDN?")
Out[4]: '\x00KEITHLEY INSTRUMENTS INC.,MODEL 2400,1304207,C31   May 28 2010 10:34:39/A02  /R/J\n'

adapterS.ask("*IDN?")
/home/opsaw/anaconda3/lib/python3.10/site-packages/pymeasure/adapters/prologix.py:214: FutureWarning: `Adapter.ask` is deprecated, call `Instrument.ask` instead.
  warn("`Adapter.ask` is deprecated, call `Instrument.ask` instead.", FutureWarning)
Out[5]: '\x00KEITHLEY INSTRUMENTS INC.,MODEL 2400,1304207,C31   May 28 2010 10:34:39/A02  /R/J\n'

adapterS.read_bytes(1)
Out[6]: b'\x00'

sourcemeter.id
Out[7]: 'KEITHLEY INSTRUMENTS INC.,MODEL 2400,1304207,C31   May 28 2010 10:34:39/A02  /R/J'
```
As you can see the first query is fine but after at each query it returns "\x00" at the beginning. This is problematic in a general context.
I have the same thing for HEWLETT-PACKARD 34420A Meter.
I tried to play with eos or eoi but no way.
Any idea ?
Using pymeasure 0.11.1 (pretty sure), pyvisa = '1.13.0' and Ubuntu 22.04.2 LTS
Thanks
Hope I post to the good place.

bilderbuchi, 2023-07-23:
@msmttchr do you know what's going wrong here?

msmttchr, 2023-07-24:
I think it's possible that prologix is configured with `eot_enable=1` and `eot_char=0` (see user manual for details). Also, I know that you can save configuration permanently in nvm, so settings will be preserved even after power cycle.
Please try querying your adapter for the above parameters, to see if my guess is right.

lucort, 2023-07-24:
Thanks a lot.
In the prologix documentation (https://prologix.biz/downloads/PrologixGpibUsbManual-6.0.pdf) I found what you are talking about. I got a look but as my prologix device worked in previous pymeasure release I didn't attach more attention.
Anyway, I put eot_enable = 0 (disable): Do not append character when EOI detected (in Prologix manual)
What I did:
```python
voltmeter.id
Out[18]: '\x00HEWLETT-PACKARD,34420A,0,9.0-5.0-2.0'

adapterV.write("++eot_enable")

adapterV.read()
Out[20]: '\x001\r\n'

adapterV.write("++eot_char")

adapterV.read()
Out[22]: '0\r\n'

adapterV.write("++eot_enable 0")

adapterV.write("++eot_enable")

adapterV.read()
Out[26]: '0\r\n'

adapterV.write("++eot_char")

adapterV.read()
Out[28]: '0\r\n'

voltmeter.id
Out[29]: 'HEWLETT-PACKARD,34420A,0,9.0-5.0-2.0'

voltmeter.id
Out[30]: 'HEWLETT-PACKARD,34420A,0,9.0-5.0-2.0'
```
Test:
1)restart python kernel and run again my initial script (see above): it works
2)disconnect USB prologix cable and run script: it works

Once again, thank you for your help.

bmoneke, 2023-07-24:
So, your problem is solved?

msmttchr, 2023-07-24:
Ok, thanks for the confirmation. Possibly we need to add the disable of eot
in the initialization code, to make sure it's set in the expected mode.

Previous versions of pymeasure were probably hiding this problem.