pymeasure/pymeasure #955: reset() for DSP7265 not working

CHennighausen, 2023-07-30:
The 'reset()' method for the DSP7265 does not appear to be working. When running it, I get the error message: "NotImplementedError: Non SCPI instruments require implementation in subclasses"

LongnoseRob, 2023-07-30:
Yes, on SCPI type instruments there is a reset-command ("*RST").
On non-SCPI complient instruments like the SRS DSP7265, this `reset()` method needs to be implmented by the user.
Glasncing across the code in `instruments/srs` I see no `reset()` method being defined.

You could check it the SRS documentation states a specific reset command or try if a `clear()` works as well for what you wanted to do.

neuschs, 2023-10-18:
Can you try
```write("ADF1")```

From the documentation it is called Auto default and will reset everything to the default except the communication settings.