pymeasure/pymeasure #956: Add Symbol to the graph plot

jbuencuerpo, 2023-08-17:
Hi, 
I have been testing the ManageWindow class and following the tutorial to create an interface for measuring at the lab.
I am trying just to do a small customization that is to add a symbol to each  point in the graph. But, I haven't found in the docs an easy way to do it. Is there an easy way of doing it, or it requires changing the widget for plotting by creating your own widget?
Thank you

msmttchr, 2023-08-18:
Please find attached a customization of  `examples/Basic/gui.py`.

Ugly, but working in my PC. I am not sure whether something simpler can be done.
[gui_with_symbols.zip](https://github.com/pymeasure/pymeasure/files/12379730/gui_with_symbols.zip)

jbuencuerpo, 2023-08-18:
Hi,

Thanks! it works, I will play with it and maybe add a simple widget to select the symbol.

If I can suggest a future improvement it would be nice to select the symbol as the same way is done for the colors, it helps when measuring many curves.

Thanks

lucort, 2023-08-25:
Hi,
I tried your example gui_with_symbols.zip. It works but when I try to implement it using ManagedDockWindow I'm lost with widgetlist. Maybe there is a way to do that but I don't know which one.
Thanks

msmttchr, 2023-08-25:
> 

Untested possible ideas below.

In `ManagedWindowBase`, you should modify line:
    `def new_experiment(self, results, curve=None)`:

adding a `kwargs` parameter

You need to pass this `kwargs` to `new_curve` method.

In `PlotWidget`, change method `new_curve` as follow:
```
   def new_curve(self, results, color=pg.intColor(0), **kwargs):
        if 'pen' not in kwargs:
            kwargs['pen'] = pg.mkPen(color=color, width=self.linewidth)
        if 'antialias' not in kwargs:
            kwargs['antialias'] = False
        If not 'symbolBrush' not in kwargs:
           kwargs['symbolBrush'] = pg.mkBrush(color)
        curve = ResultsCurve(results,
                             wdg=self,
                             x=self.plot_frame.x_axis,
                             y=self.plot_frame.y_axis,
                             **kwargs,
                             )
        return curve

```

In this way, you should be able to pass the argument `symbol='o'` to `new_experiment` call in `basic/GUI.py` example and get what the symbol in the curve.

lucort, 2023-08-27:
Hi,
I followed your idea , it almost works but with some troubles.
So I did the following 

    def new_curve(self, results, color=pg.intColor(0), **kwargs):
        if 'pen' not in kwargs:
            kwargs['pen'] = pg.mkPen(color=color, width=self.linewidth)
        if 'antialias' not in kwargs:
            kwargs['antialias'] = False
        if 'symbol' not in kwargs:
            kwargs['symbol'] = '+'
            kwargs['symbolBrush'] = pg.mkBrush(color)
            kwargs['symbolSize'] = 25
        curve = ResultsCurve(results,
                             wdg=self,
                             x=self.plot_frame.x_axis,
                             y=self.plot_frame.y_axis,
                             **kwargs,
                             )
        # curve.setSymbol(None)
        # curve.setSymbolBrush(None)        
        return curve

and I run 

experiment = self.new_experiment(results)

in my iv_Keithley_Agilent_seq_dock.py code. So default now is line and symbol.

By clicking on 'Queue' button, curves are plotted with 'standard' lines and symbols BUT when I start a new experiment (launching the queue) the color is updated for line BUT NOT for symbol. The symbol color stays in red color.

Any ideas ?
  Thank you

msmttchr, 2023-08-28:
Not sure, why this is happening, sorry :(.
On the other hand, I have tried to implement the proposed idea on the attached patch file.

Could you please have a look?
[patch.zip](https://github.com/pymeasure/pymeasure/files/12454325/patch.zip)

lucort, 2023-08-28:
With gui.py example it works fine.
But when using class MainWindow(ManagedDockWindow) in the guy.py , I have colored lines but only red symbols for all curves. 
(of course importing: "from pymeasure.display.windows.managed_dock_window import ManagedDockWindow"
)

msmttchr, 2023-08-28:
Ok, try changing line https://github.com/pymeasure/pymeasure/blob/e5045467a2550d9a6e08c20fb837df971df17d67/pymeasure/display/widgets/dock_widget.py#L161
as follow:
` curves.append(self.plot_frames[i].new_curve(results, color=color, **kwargs)) `
I am not sure whether this is a bug.

lucort, 2023-08-28:
Fine !
In ManagedDockWindow, I focused on ManagedWindowBase.
I didn't know that DockWidget class has also a new_curve definition.
Probably looks like a little bug and would need to be revised in the future.
Thank you very much

lucort, 2023-08-28:
In another way, if we want to change line width using ManagedDockWindow (for illustration purpose in GUI.py example):

class MainWindow(ManagedDockWindow):

    def __init__(self):
        super().__init__(
            procedure_class=TestProcedure,
            inputs=['iterations', 'delay', 'seed'],
            displays=['iterations', 'delay', 'seed'],
            x_axis='Iteration',
            y_axis='Random Number', linewidth=6
        )
        self.setWindowTitle('GUI Example')

    def queue(self):
        filename = tempfile.mktemp()

        procedure = self.make_procedure()
        results = Results(procedure, filename)
        experiment = self.new_experiment(results, symbol='o')

        self.manager.queue(experiment)

Note linewidth=6.

I needed to change in managed_dock_window.py (line 82):

        self.dock_widget = DockWidget("Dock Tab", procedure_class, self.x_axis_labels,
                                      self.y_axis_labels)

in

        self.dock_widget = DockWidget("Dock Tab", procedure_class, self.x_axis_labels,
                                      self.y_axis_labels, linewidth=linewidth)

I don't know if it is the best way. 
Just to notice to do a 'similar change' once we change in __init__ of the class (for the width) and another time it is in 'def queue' (for adding symbols):

 experiment = self.new_experiment(results, symbol='o')

msmttchr, 2023-09-03:
> In another way, if we want to change line width using ManagedDockWindow (for illustration purpose in GUI.py example):
>
> class MainWindow(ManagedDockWindow):
>
> ```
> def __init__(self): [...]

@lucort, I think your points are valid:
 - `ManagedDockWindow` diesel not obey to linewidth parameter (this is a bug)
 - Customization to plot behavior can be achieved in two ways, this is design point of discussion, I believe.
 
I will try to submit the pr for the bugs and leave the design question open for the time being.