pymeasure/pymeasure #957: Commit 9bd6e08 breaks `download_image` method of Teledyne LeCroy T3DSO1204 oscilloscope

LastStarDust, 2023-08-20:
Hello!
I am the contributor for the following instruments:

- Teledyne LeCroy T3DSO1204 oscilloscope
- Texio PSW family power supply

A colleague of mine noticed that with the latest version of pymeasure v0.12.0, he cannot download the images from T3DSO1204 oscilloscope.

I confirmed that the [`download_image` method](https://github.com/pymeasure/pymeasure/blob/e5045467a2550d9a6e08c20fb837df971df17d67/pymeasure/instruments/lecroy/lecroyT3DSO1204.py#L1001) is never returning, i.e. it is blocking indefinitely.

By using `git bisect` I found out that [this commit](https://github.com/pymeasure/pymeasure/commit/9bd6e082f76d7f8f792e11c3835a7fb732e79381) breaks the `download_image` method.

The commit message is `VISAAdapter's read_bytes(-1) reads until timeout.` and it makes sense because this issue is `timeout` related.

bilderbuchi, 2023-09-02:
ping @bmoneke this came from #867. do you know how to best fix this?

bmoneke, 2023-09-05:
I'll have a look.

bmoneke, 2023-09-05:
I looked at the code and the difference is, how we define `read_bytes(-1)`.
Previously we used `read_raw` to read until some (pyvisa defined) end, for example up to newline character if no term char was given.

I suspect two possibilities: Either the timeout is too long (and never encountered) as we read now until a timeout occurs, or the previous implementation expected implicitly a termchar as due to a pyvisa bug (https://github.com/pyvisa/pyvisa/issues/728) a newline char was assumed as termchar.
You could try to set a different timeout (a short time) and check, whether that stops the reading at some point. Or even more basic, try to read, whatever the osci sends (read_bytes(1) in a loop until timeout).

btw @LastStarDust, why did you hardcode the adapter timeout:
~~~ python
if self.adapter.connection is not None:
    self.adapter.connection.timeout = 3000
~~~

instead of giving a normal kwarg init?