pymeasure/pymeasure #958: GUI: Popup menu as Input parameters

lucort, 2023-08-24:
Hello,
as input parameter is it possible to use a popup menu with a list of possible values ? For an instrument, we want to select some settings which are only available for certain values. For example, in a (nano)-voltmeter we can select the NPLC parameter with 0.1, 0.2, 1, 2, 10, 20, 100 and 200. 
I try to find some examples in issue #358 and #360 but I haven't been able to go any further.
I'm working using "examples/Current-Voltage Measurements/iv_keithley.py".

I have also a question concerning linked (dependent) parameters and the way to update them in the datafile and display when launching an experiment but I will ask more in details in a separate issue.
Thanks,
Luc

CasperSchippers, 2023-08-26:
Hello,

Do you mean something like this?
[image]

If so, you can achieve this with
``` python
NPLC = ListParameter('NPLC', choices=[0.1, 0.2, 1, 2, 10, 20, 100, 200], default=1)
```

Hope this helps; if not let me know!

lucort, 2023-08-27:
Hi,
It works well ! Sorry I should have found it by myself by looking in the documentation or in the code...
We need also to import ListParameter of course.
Thanks a lot

msmttchr, 2023-08-29:
> Hi, It works well ! Sorry I should have found it by myself by looking in the documentation or in the code... We need also to import ListParameter of course. Thanks a lot

@lucort, if you think the solution is satisfying, could You please close the issue?

closed 2023-08-29 by lucort