pymeasure/pymeasure #965: Connection Issue with ANC300 device

ileu, 2023-09-15:
Hello,

I have a problem using the ANC300 Controller. The communication over telnet works fine over the console. But when I try to connect with it in a python script I get the following error:

```
Traceback (most recent call last):
  File "C:\Users\ONG_C54_01\AppData\Roaming\JetBrains\PyCharmCE2022.2\scratches\scratch.py", line 3, in <module>
    controller = ANC300Controller(adapter="TCPIP::192.168.1.2::7230::SOCKET", axisnames=["RZ", "RY", "RX", "LZ", "LY", "LX"], passwd="123456", query_delay=2)
  File "C:\Users\ONG_C54_01\Documents\CryoGUI\venv\lib\site-packages\pymeasure\instruments\attocube\anc300.py", line 282, in __init__
    super().read()
  File "C:\Users\ONG_C54_01\Documents\CryoGUI\venv\lib\site-packages\pymeasure\instruments\instrument.py", line 152, in read
    return self.adapter.read(**kwargs)
  File "C:\Users\ONG_C54_01\Documents\CryoGUI\venv\lib\site-packages\pymeasure\adapters\adapter.py", line 108, in read
    read = self._read(**kwargs)
  File "C:\Users\ONG_C54_01\Documents\CryoGUI\venv\lib\site-packages\pymeasure\adapters\visa.py", line 172, in _read
    return self.connection.read(**kwargs)
  File "C:\Users\ONG_C54_01\Documents\CryoGUI\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 0: ordinal not in range(128)
```

and the corresponding script I run is:

```
from pymeasure.instruments.attocube import ANC300Controller

controller = ANC300Controller(adapter="TCPIP::192.168.1.2::7230::SOCKET", axisnames=["RZ", "RY", "RX", "LZ", "LY", "LX"], passwd="123456", query_delay=2)

print(controller.status)
```

It seems as ``self.adapter.flush_read_buffer()`` doesn't actually flush the buffer and then the adapter tries to read the non-ascii telnet control characters and crashes.

Any help is greatly appreciated.

bmoneke, 2023-09-15:
What version of pymeasure do you use? I think such an issue got already addressed, but I do not know, whether it is already released.

ileu, 2023-09-15:
I use version 0.12.0

bmoneke, 2023-09-19:
Sorry for the late response.

I checked the code, funnily enough, the line numbers of your error message do not match the code of the tag 0.12.0. Nevertheless, you use the latest code (regarding the init).

you could try to increase the query_delay, as it might be, that the answer of the device arrives after the read buffer has been flushed. `query_delay` determines the time between writing and reading. This is also the delay between writing and flushing the buffer.

If you want, we could count the number of bytes of the initial conversation and read that many bytes. That would ensure, that we have all the initial conversation (with possibly non ASCII chars).

Could you try, whether the flush method run manually flushes the buffer?

ileu, 2023-09-19:
Yeah no worries. The line numbers probably don't match since I already tried some things to see why it isn't working ;).

I figured out some stuff in the mean time:

- I also tried with increased `query_delay`, I set it to 2 seconds and the same thing happened.
- I added after the `fself.adapterlush_read_buffer()' a print call with `super().read_bytes(-1)`  and got the following:
```
b'\xff\xfd\x01\xff\xfd!\xff\xfb\x01\xff\xfb\x03attocube ANC300 controller version 1.2.2-3 2019-01-25 11:02\r\nANC150 compatibillity console\r\n\r\nAuthorization code: '
```
  and then I could connect and control the controller.
- Next, thing I tried is to change the argument of the `self.connection.flush` inside of the `flush_read_buffer` method of the `VISAAdaper` from `discard_read_buffer` to `discard_receive_buffer` . After doing that the connection worked even without the `super().read_bytes` call

So I got it to work, but I dont know how it will influence other devices that use the `VISAAdapter`.

bmoneke, 2023-09-19:
Thanks for that information. Now I think I understand the problem:
Previously, only ASRL did have pyvisa's flush read buffer. For that reason, we implemented it manually (with read_bytes).
In https://github.com/pyvisa/pyvisa-py/pull/350 , TCP (which is used here), received the flush method.
It appears, that for TCP flushing the read_buffer is not enough.

I think, if someone calls `flush_read_buffer`, they intend to also flush the receive buffer. Therefore, we should add `discard_receive_buffer` in the `flush_read_buffer` method, additionally to `discard_read_buffer`.

Would you mind to implement it?

bmoneke, 2023-09-19:
As an alternative (or addition), we could add implement a better filtering scheme in the ANC300's init (here information from the manual would be useful: is it always the same number of bytes for all devices etc.)

ileu, 2023-09-19:
The ANC300 uses the Telnet protocol, so I think there should be a standard, but I need to look it up.

I can make a pull request later today.

ileu, 2023-09-22:
I created a separate pull request just for the `flush_read_buffer`.

As for the ANC300, I could not find anything in the manual. But I think it send always the same amount of control characters.

The only helpful thing I found is:
```
A command on the standard console is composed of normal ASCII characters 
and needs to be ended by a carriage return and a line feed character (DOS 
line endings: CR = ASCII code 13; LF = ASCII code 10). When a command has 
been executed, the answer should be “OK “, with optional lines before containing data.
In case of an error, the electronics returns “ERROR” together 
with an error message if applicable. The typical timeout for these commands 
is about 30 ms.
```

I will look into it more over the next week if the controller works correctly and keep you updated.