pymeasure/pymeasure #969: Struggles with USB-Serial Adapter / toptica_ibeam.py

waveman68, 2023-09-22:
We have considerable issues using the `toptica_ibeam.py` instrument with the USB-serial adapter that Toptica provided. We managed to get it working more or less but faced a number of issues when upgrading to 0.12 (we already submitted a fix that is already in 0.13).

The core issue is one of timing or buffer management. We regularly get responses from previous commands. This error leads us to believe that it may be a buffer issue:

```python
laser = IBeamSmart(adapter="ASRL/dev/ttyUSB0::INSTR", baud_rate=115200)
```

```sh
temperature_diode gave unexpected response OFF  # this should be a float

# setting the power causes this:
Setting self.laser.power caused error 'SerialInstrument' object has no attribute 'flush_read_buffer'
```

Adjusting the `query_delay` per documentation (BTW: this causes a deprecation warning) does not help – 200 ms should be more than enough. I guess we are supposed to override `wait_for`?

```sh
.../pymeasure/adapters/visa.py:90: FutureWarning: Parameter `query_delay` is deprecated. Implement in Instrument's `wait_for` instead.
```

At some point there was an `adapter.py` for  `toptica_ibeam.py` but this was removed. Is there some way of restoring the  'flush_read_buffer'? I would be happy to do a PR if someone would point me in the right direction.

bmoneke, 2023-09-23:
I found something:
[                self.adapter.connection.flush_read_buffer()
](https://github.com/pymeasure/pymeasure/blob/1b15505b61fc58a845083c77874a68a53e15437e/pymeasure/instruments/toptica/ibeamsmart.py#L167)
https://github.com/pymeasure/pymeasure/blob/1b15505b61fc58a845083c77874a68a53e15437e/pymeasure/instruments/toptica/ibeamsmart.py#L188
These lines cannot work, as the `pyvisa.SerialInstrument` does not have a `flush_read_buffer` method.
Instead, the lines should be `self.adapter.flush_read_buffer`. Could you try that, please?

Some informational comments:

> Adjusting the query_delay per documentation (BTW: this causes a deprecation warning) does not help – 200 ms should be more than enough. I guess we are supposed to override wait_for?

Yes, you're supposed to override `wait_for`.

> At some point there was an adapter.py for toptica_ibeam.py but this was removed. 

Exactly, there was an adapter for the ibeam, but we moved the logic from that adapter into the ibeam_smart itself in order to be able to use doctests and to harmonize adapter usage in the codebase.

bmoneke, 2023-09-23:
Those two lines are definitively wrong and have to be changed as indicated (would you mind doing that as a PR?).
I wonder, whether that is the issue.

BenediktBurger, 2024-03-14:
I created a PR #1065 which fixes the code bug I found above.
Would you mind testing, whether this fixes your problem @waveman68 ?

waveman68, 2024-04-30:
Yes, I would but I'm currently in home office after an operation.

BenediktBurger, 2024-04-30:
I wish you a good recovery.

BenediktBurger, 2024-05-22:
By the way, PR #1065 has been merged, therefore the fix is in the master and might be shortly on Pypi, depending on #1104.