pymeasure/pymeasure #971: Adding another properties value to a set command

neuschs, 2023-10-03:
Hi,

I stumbled over a bug in my own implementation for the HP856Xx spectrum analyzers:
You **must** append the unit as string to the value you are writing otherwise a error is created. 
This unit is accesible. This only matters for the amplitude. For time or frequency this value can be const 's' or 'Hz'.

Now to my question:
The Instrument.control / Instrument.setting 's set_process for example has no access to 'self' to access the parameter self.amplitude_units. 
Now I could implement my own properties where I would have access to self, but that would make no sense to me.

Another option would be to simply not allow any other amplitude unit than e.g. 'dBm'.

What would be your implementation guidance ?

bmoneke, 2023-10-04:
The most simple solution I see, is to add a small string formatting to the `write` method:
~~~python
def write(self, command):
    if "{amplitude_units}" in command:
        command = command.format(amplitude_units=self.amplitude_units)
~~~

Does this happen in a Channel or in the main instrument?
In a Channel, you could do it in the `insert_id` method, which does such a replacement for the channel ID.

neuschs, 2023-10-04:
Thanks, that sounds like a damn good idea !
I will try that.

closed 2023-10-09 by neuschs