pymeasure/pymeasure #972: Pymeasure for older python version

cphatak, 2023-10-04:
hi,

I am looking to install pymeasure in a miniconda environment which is currently limited to the following versions of python and numpy.
python = 3.7.2150.1013
numpy = 1.18.2

Is there an older version of pymeasure that can work in this environment?

Thank you,

cheers,
-cd

bilderbuchi, 2023-10-04:
We support Python versions until they reached their end-of-life date. For Python 3.7 this happened a [couple of months ago](https://endoflife.date/python).
You should be able to install a pymeasure version, probably 0.12.0 as we only dropped python 3.7 in 0.13.
What did you try, and what happened? Using `python=3.7 numpy=1.18 pymeasure` I can create an environment without problems, so maybe something else in your existing environment conflicts.

If conda didn't give good information to troublehshoot this, you can try [mamba ](https://github.com/mamba-org/mamba)(it's also often faster).

cphatak, 2023-10-04:
I was able to install pymeasure 0.12 and it seems to work. I haven't explicitly tried connecting with instruments but I don't get error importing it in. With latest version of pymeasure, below is the information of my environment, packages installed, and error I get when I install pymeasure, if it is helpful.

Conda Info:
  
   active environment : CD_ENV
    active env location : C:\ProgramData\Miniconda3\envs\CD_ENV
            shell level : 1
       user config file : C:\Users\cphatak\.condarc
 populated config files : C:\Users\cphatak\.condarc
          conda version : 4.8.3
    conda-build version : not installed
         python version : 3.7.1.final.0
       virtual packages :
       base environment : C:\ProgramData\Miniconda3  (read only)
           channel URLs : https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : C:\ProgramData\Miniconda3\pkgs
                          C:\Users\cphatak\.conda\pkgs
                          C:\Users\cphatak\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\cphatak\.conda\envs
                          C:\ProgramData\Miniconda3\envs
                          C:\Users\cphatak\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/4.8.3 requests/2.21.0 CPython/3.7.1 Windows/10 Windows/10.0.19041
          administrator : False
             netrc file : None
           offline mode : False

Conda list - 

(CD_ENV) C:\Users\cphatak>conda list
# packages in environment at C:\ProgramData\Miniconda3\envs\CD_ENV:
#
# Name                    Version                   Build  Channel
ca-certificates           2020.1.1                      0    <unknown>
certifi                   2020.4.5.1               py37_0    <unknown>
charset-normalizer        3.3.0                    pypi_0    pypi
cycler                    0.11.0                   pypi_0    pypi
fonttools                 4.38.0                   pypi_0    pypi
h5py                      3.8.0                    pypi_0    pypi
idna                      3.4                      pypi_0    pypi
importlib-metadata        6.7.0                    pypi_0    pypi
kiwisolver                1.4.5                    pypi_0    pypi
matplotlib                3.5.3                    pypi_0    pypi
numpy                     1.18.2                   pypi_0    pypi
openssl                   1.1.1f               he774522_0    <unknown>
packaging                 23.2                     pypi_0    pypi
pandas                    1.3.5                    pypi_0    pypi
pillow                    9.5.0                    pypi_0    pypi
pint                      0.18                     pypi_0    pypi
pip                       20.0.2                   py37_1    <unknown>
psutil                    5.9.5                    pypi_0    pypi
pyparsing                 3.1.1                    pypi_0    pypi
pyqtgraph                 0.12.4                   pypi_0    pypi
pyserial                  3.5                      pypi_0    pypi
python                    3.7.2               h8c8aaf0_10    <unknown>
python-dateutil           2.8.2                    pypi_0    pypi
pytz                      2023.3.post1             pypi_0    pypi
pyvisa                    1.12.0                   pypi_0    pypi
pyvisa-py                 0.6.3                    pypi_0    pypi
requests                  2.31.0                   pypi_0    pypi
setuptools                68.0.0                   pypi_0    pypi
six                       1.16.0                   pypi_0    pypi
sqlite                    3.31.1               he774522_0    <unknown>
[... 23 lines cut ...]
Requirement already satisfied: typing-extensions>=3.6.4; python_version < "3.8" in c:\programdata\miniconda3\envs\cd_env\lib\site-packages (from importlib-metadata; python_version < "3.8"->pymeasure) (4.7.1)
Requirement already satisfied: zipp>=0.5 in c:\programdata\miniconda3\envs\cd_env\lib\site-packages (from importlib-metadata; python_version < "3.8"->pymeasure) (3.15.0)
Requirement already satisfied: packaging in c:\programdata\miniconda3\envs\cd_env\lib\site-packages (from pint->pymeasure) (23.2)
Requirement already satisfied: python-dateutil>=2.7.3 in c:\programdata\miniconda3\envs\cd_env\lib\site-packages (from pandas>=0.14->pymeasure) (2.8.2)
Requirement already satisfied: pytz>=2017.3 in c:\programdata\miniconda3\envs\cd_env\lib\site-packages (from pandas>=0.14->pymeasure) (2023.3.post1)
Requirement already satisfied: six>=1.5 in c:\programdata\miniconda3\envs\cd_env\lib\site-packages (from python-dateutil>=2.7.3->pandas>=0.14->pymeasure) (1.16.0)
Installing collected packages: pymeasure
Successfully installed pymeasure-0.13.0

(CD_ENV) C:\Users\cphatak>python
Python 3.7.2 (default, Feb 21 2019, 17:35:59) [MSC v.1915 64 bit (AMD64)] :: Anaconda, Inc. on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import pymeasure
Traceback (most recent call last):
[... cut ...] \pymeasure\__init__.py", line 34, in <module>
    import setuptools_scm
ModuleNotFoundError: No module named 'setuptools_scm'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\ProgramData\Miniconda3\envs\CD_ENV\lib\site-packages\pymeasure\__init__.py", line 39, in <module>
    from importlib.metadata import version, PackageNotFoundError
ModuleNotFoundError: No module named 'importlib.metadata'

bilderbuchi, 2023-10-05:
Note that contrary to what you wrote, this seems to have installed pymeasure 0.13:

> Using cached PyMeasure-0.13.0-py3-none-any.whl (679 kB)

> No module named 'importlib.metadata'

This is because you are on Python3.7 IIRC, and can be avoided by installing a pymeasure version that still supports this -- 0.12.
Can you do `pip install pymeasure=0.12`, instead?

> No module named 'setuptools_scm'

This means that `setuptools_scm` is not installed. I'm not sure if this is because the pip package misses some dependency information, your pip is too old to deal with the [build requirement ](https://github.com/pymeasure/pymeasure/blob/master/pyproject.toml#L2), or something else is going on.
It's easy to workaround if this reappears by just installing it manually.

Side remark: It's weird that you use conda for the environment, but then use _pip_/pypi to install your package -- afaik from previous information you shouldn't mix those if possible. Why don't you use conda to manage your conda environments?

bilderbuchi, 2023-10-05:
That you get 0.13 is a problem on our side apparently, I've filed #973.

bilderbuchi, 2023-10-05:
I have fixed the problem and started the process to remove the broken 0.13.0 version from pypi. Until that happens, you need to do `pip install pymeasure==0.12` as advised above (or use `conda`/`mamba` and keep more sanity :-P)
Feel free to reopen discussion if you still experience problems.

closed 2023-10-05 by bilderbuchi