pymeasure/pymeasure #973: Ensure correct python version limitations

bilderbuchi, 2023-10-05:
It seems that at least pip installations don't pick up Python version limitations correctly. On Python=3.7, `pip install pymeasure` picks pymeasure 0.13, but this does not support 3.7 anymore and importing it fails with
```
    from importlib.metadata import version, PackageNotFoundError
ModuleNotFoundError: No module named 'importlib.metadata'
```
We should asap:
* [ ] Find out if this affects pip _and_ conda installations
* [ ] Determine what is missing in pyproject.toml or setup.cfg
* [ ] Properly constrain minimum Python versions
* [ ] Test this in pip and conda environments (should be fair to create conda envs only with python, and use pip for the rest)
* [ ] Create an updated release. Probably, as this is a build meta change, it's best to work from tag `v0.13` and just bump the build number and leave "0.13" in place. This also avoids that we have a broken version online.

I could use some help with research on what the appropriate fix is.

bilderbuchi, 2023-10-05:
While we are at it, we should also 
* [ ] check the `setuptools_scm` dependency in pip installations.
a user reported in #972 that they got
```
import pymeasure
Traceback (most recent call last):
File "C:\ProgramData\Miniconda3\envs\CD_ENV\lib\site-packages\pymeasure_init_.py", line 34, in
import setuptools_scm
ModuleNotFoundError: No module named 'setuptools_scm
```
but I'm not sure if this is not a side effect of the other problem -- the build requirement is [specified](https://github.com/pymeasure/pymeasure/blob/master/pyproject.toml#L2). 
* [ ] We should check the setuptools_scm if that is the correct place for our usage.

bmoneke, 2023-10-05:
This entry describes how to add the [python_requires option](https://packaging.python.org/en/latest/guides/distributing-packages-using-setuptools/#python-requires) to a pyproject.toml file:
https://stackoverflow.com/questions/72373093/how-to-define-python-requires-in-pyproject-toml-using-setuptools

In a quick glance, I did not find, where we specify the valid python version.

bilderbuchi, 2023-10-05:
> In a quick glance, I did not find, where we specify the valid python version.

only in the "classifiers" section https://github.com/pymeasure/pymeasure/blob/master/setup.cfg#L20-L24, we overlooked that that is not enough.

bmoneke, 2023-10-05:
Yeah, I found the "classifiers" section, but considered it (it seems correctly) as irrelevant for pip.

So the solution seems straightforward to adjust pyproject.toml (or setup.cfg)?

We could test it (in the PR) by restricting the python version such, that not all test environments can succeed (e.g. python >3.8 < 3.10).
If that works, the final PR version can switch to python>=3.8.

bilderbuchi, 2023-10-05:
Darn it, this is more complex than thought -- on pypi I have to "yank" the broken 0.13.0 release (and add a new, later, one), because otherwise it would still get picked up by py37 pip as the latest compatible --  a PEP440 "post-release" (as I intended) would not fix that.

bilderbuchi, 2023-10-05:
I could not reproduce the problem with setuptools_scm, so I'm assuming this was just a weird side-effect. I asked Colin to yank pymeasure 0.13.0 from pypi, as that will otherwise remain available for installation for py37 users. Until that happens, people on py37 and using pip need to install 0.12 explicitly: `pip install pymeasure==0.12.0`. 
I published 0.13.1 with the fixed python version limit to both pypi and conda.

bilderbuchi, 2023-10-06:
0.13.0 has been yanked now.

closed 2023-10-06 by bilderbuchi