pymeasure/pymeasure #974: Actions (CI run?) badge is broken

bmoneke, 2023-10-05:
The CI run badge on the documentation is broken (on readthedocs and in the documetns viewer for the `index.rst` file.

https://github.com/pymeasure/pymeasure/blob/78db2c9175bf0535c8f391a4ba7b1ad2883f40c1/docs/index.rst?plain=1#L19

bilderbuchi, 2023-10-05:
Fixed by [f1898bf](https://github.com/pymeasure/pymeasure/commit/f1898bfe1d6d950870672f8d980a86852545386d) ( I directly pushed that)

bmoneke, 2023-10-06:
It's broken in `index.rst` (original issue cause) as well.

bilderbuchi, 2023-10-06:
oops, I missed that.

CasperSchippers, 2023-10-10:
I also took the liberty to directly fix this on master (only just realised that you might just have pushed it direclty to master in order to get it in before the 0.13.1, but oh well :sweat_smile:)

closed 2023-10-10 by CasperSchippers