pymeasure/pymeasure #976: Emitting data asynchronously generates scrambled csv

erickmartinez, 2023-10-06:
I have a script that retrieves data asynchronously from instruments. I store it in memory and then emit the data. If I save the data using pandas, the data is uncorrupted. However, the output from pymeasure's csv contains line breaks at incorrect locations and some rows get swapped.

Here is a piece of code that reproduces de error:

`
import logging
#log = logging.getLogger(__name__)
#log.addHandler(logging.NullHandler())

import sys
import tempfile
import random
from time import sleep
from pymeasure.log import console_log
from pymeasure.display.Qt import QtWidgets
from pymeasure.display.windows import ManagedWindow
from pymeasure.experiment import Procedure, Results
from pymeasure.experiment import IntegerParameter, FloatParameter, Parameter
from pymeasure.experiment import unique_filename
import numpy as np
import os

class RandomProcedure(Procedure):

    tmax = IntegerParameter('tmax', default=10, units='s')
    dt = FloatParameter('dt', units='s', default=0.005)
    seed = Parameter('Random Seed', default='12345')

    DATA_COLUMNS = [
        "Measurement Time (s)", "Pressure (torr)", "TC1 (°C)", "TC2 (°C)", "Trigger (V)", 
        "Laser output power (W)", "Laser output peak power (W)"
    ]

    def startup(self):
        log.info("Setting the seed of the random number generator")
        random.seed(self.seed)

    def execute(self):
        dt = 0.005
        N = int(self.tmax / self.dt) + 1
        measurement_time = dt * np.arange(0, N)
        pressure = 0.5 * np.exp(-(measurement_time-0.5*self.tmax) ** 2.) + np.random.normal(loc=0., scale=0.5, size=N)
        tc1 = np.random.normal(loc=20., scale=2., size=N)
        tc2 = np.random.normal(loc=20., scale=2., size=N) + 10.
        trigger = np.zeros_like(measurement_time)
        trigger[(0.5 <= measurement_time) & (measurement_time <= 1.0)] = 3.3
        lop = np.random.normal(loc=0., scale=0.5, size=N) + 1000. * trigger
        lpp = np.random.normal(loc=10., scale=1.0, size=N) + 1000. * trigger
        log.info("Starting the loop of %d iterations" % N)
        for i in range(N):
            d = {
                "Measurement Time (s)": measurement_time[i],
                "Trigger (V)": trigger[i],
                "TC1 (°C)": tc1[i],
                "TC2 (°C)": tc2[i],
                "Pressure (torr)": pressure[i],
                "Laser output power (W)": lop[i],
                "Laser output peak power (W)": lpp[i]
            }
            self.emit('results', d)
            log.debug("Emitting results: %s" % d)
            self.emit('progress', 100 * i / N)
            sleep(0.001)
            if self.should_stop():
                log.warning("Caught the stop flag in the procedure")
                break

class MainWindow(ManagedWindow):

    def __init__(self):
        super().__init__(
            procedure_class=RandomProcedure,
            inputs=['tmax', 'dt', 'seed'],
            displays=['tmax', 'dt', 'seed'],
            x_axis='Measurement Time (s)',
            y_axis='Pressure (torr)',
            directory_input=True,
        )
        self.setWindowTitle('GUI Example')

    def queue(self):
        directory = self.directory
        filename = tempfile.mktemp()
        
        filename = unique_filename(directory, prefix='TEST_')
        log_file = os.path.splitext(filename)[0] + '.log'
        formatter = logging.Formatter('%(asctime)s - %(levelname)s - %(message)s')
        fh = logging.FileHandler(log_file)
        fh.setFormatter(formatter)
        fh.setLevel(logging.DEBUG)
        log.addHandler(fh)

        procedure = self.make_procedure()
        results = Results(procedure, filename)
        experiment = self.new_experiment(results)

        self.manager.queue(experiment)

if __name__ == "__main__":
    log = logging.getLogger(__name__)
    log.addHandler(logging.NullHandler())
    
    # create console handler and set level to debug
    has_console_handler = False
    if len(log.handlers) > 0:
        for handler in log.handlers:
            if isinstance(handler, logging.StreamHandler):
                has_console_handler = True

    if not has_console_handler:
        ch = logging.StreamHandler()
        ch.setLevel(logging.DEBUG)
        log.addHandler(ch)
        
    app = QtWidgets.QApplication(sys.argv)
    window = MainWindow()
    window.show()
    sys.exit(app.exec())
`

I am using pymeasure 0.13.1 on python 3.8.16

bmoneke, 2023-10-09:
@msmttchr , @CasperSchippers  do you know, what the reason could be?

erickmartinez, 2023-10-09:
I noticed that my `DATA_COLUMNS` do not follow the same order as the columns in the record I emit. This fixes this issue for the minimal code that I provided. However, it does not seem to resolve my issue with my working code which has real instruments. I somehow blame it on a camera (PySpin) that is acquiring images using `threading.Thread` but I can't reproduce the issue with a simple thread process. 

Maybe this issue should be marked as ''Close as not planned" unless you think differently.

CasperSchippers, 2023-10-09:
Ahh, this is an interesting one; with the example you've provided I cannot seem to reproduce your issue. I think that the order of the columns in the dict you use to emit the results _should_ not matter (and for me it seems it doesn't, but if it does for you that would still be something to look into).

Could you maybe elaborate (or have a simple example) of how the threading is used in this case? Do you  mean that there is a single separate thread that collects the data from the instruments and emits it to the data, or are there multiple?

erickmartinez, 2023-10-09:
Hi @CasperSchippers ,

Looking at the file from the example I provided, it does seem to be ok. However, the data is not showing on the plot. 

When I changed the units from °C to C the graph displayed without issues. This might be something to do with my Windows CP-1252 encoding.

As for the threading method that I am using is a separate thread. I have an experiment that fires a laser using an active high signal, so the camera triggers to the active high line and starts a routing to capture images. I defined a different class to operate the camera and acquire images. This occurs in a loop that takes ~300 1.5 megapixel images (depending on the experiment time and framerate, but usually at 200 fps for 1.5 seconds). Since the images are transmitted on a separate for loop with a short timeout, I could not figure a good way to run that code parallelly with data that is read (1) on the fly from instruments that do not can't save data internally and (2) asynchronously from a data logger. I ended up creating a function on its own thread that saves the images to a local directory. I use threading.Thread for this. A minimal example is this:

```
import logging
import threading
import time
import sys
import tempfile
import random
from time import sleep
from pymeasure.log import console_log
from pymeasure.display.Qt import QtWidgets
from pymeasure.display.windows import ManagedWindow
from pymeasure.experiment import Procedure, Results
from pymeasure.experiment import IntegerParameter, FloatParameter, Parameter
from pymeasure.experiment import unique_filename
import numpy as np
import os
from pymeasure.units import ureg
import cv2

width = 1440
height = 1040

def do_something(path):
    # In reality this code saves >300 images from a camera
    t0 = time.time()
    for i in range(300):
        ti = time.time() - t0
        lg = logging.getLogger(__name__)
        noise = np.random.randint(0, 255, (height, width), dtype=np.uint8)
        cv2.imwrite(os.path.join(path,'noise.png'), noise)
        lg.info(f'Running at time {ti:>5.3f} s ({i:>3d}/300)')
        time.sleep(0.0001)

class RandomProcedure(Procedure):

    tmax = IntegerParameter('tmax', default=10, units='s')
    dt = FloatParameter('dt', units='s', default=0.005)
    seed = Parameter('Random Seed', default='12345')
    
    __directory: str = None

    DATA_COLUMNS = [
        "Measurement Time (s)", "Pressure (torr)", "TC1 (C)", "TC2 (C)", "Trigger (V)", 
        "Laser output power (W)", "Laser output peak power (W)"
    ]
    
    @property
    def directory(self) -> str:
        return self.__directory
        
    @directory.setter
    def directory(self, value: str):
        self.__directory = value

    def startup(self):
        log.info("Setting the seed of the random number generator")
        random.seed(self.seed)

    def execute(self):
        dt = 0.005
        measurement_time = []
        pressure = []
[... 98 lines cut ...]
    log.addHandler(logging.NullHandler())
    
    # create console handler and set level to debug
    has_console_handler = False
    if len(log.handlers) > 0:
        for handler in log.handlers:
            if isinstance(handler, logging.StreamHandler):
                has_console_handler = True

    if not has_console_handler:
        ch = logging.StreamHandler()
        ch.setLevel(logging.DEBUG)
        log.addHandler(ch)
        
    app = QtWidgets.QApplication(sys.argv)
    window = MainWindow()
    window.show()
    sys.exit(app.exec())
```

But I can't reproduce the issue like this. My implementation keeps skipping some data points when I use emit, but the data is not corrupted if I save it using pandas.to_csv. I must say that I have an implementation without the camera which does not have this issue, so that's why I think it has got to do with the separate thread.

https://github.com/erickmartinez/relozwall/blob/main/experiments/laser_test_flir.py

It seems that this is only affecting me though.

CasperSchippers, 2023-10-09:
The fact that it is not plotting is indeed related to the -sign. I'd always urge you to be careful with 'special' characters. On the other hand: the absence of an error message is a mistake on our part (I noticed a number of hideous `except Exception` catch-alls... these need to be removed....)

The corruption, however, is something I do not yet have an explanation for. I'll try looking into this in the coming days, but am not sure if I can reproduce it at all.

Strangely, it seems from your demo code (and the code on your repo) that there should be no interaction between your threaded camera and the data that is written to file (if I understand correctly); so I see no reason why the presence of the thread should have any impact on the CSV file.

Do I understand correctly that the reason to have the "delayed data-writing" (i.e. the fact that you store the data in arrays and write it to file afterwards) has to do with the temperature logger from which you collect all data at once?

erickmartinez, 2023-10-10:
Hi @CasperSchippers,

Thanks for confirming the issue with the sign and for planning to implement a try-catch block to give an error message for this.

You are right, I use a temperature logger from which I download a dataset of 2000 rows containing time and temperature from two thermocouples. Most of my instruments rely on pyserial and I can't get reliable measurements for sampling rates > 10 /s unless I use external data logging which I can query afterward as a whole dataset.

I have a working code that uses the same temperature logger and a TBS 2000 oscilloscope where I use "delayed data-writing" but my issue is only with the code that uses the threaded camera. I am aware of an update for the camera's SDK that might fix some bugs, but I am afraid to get into the weeds right now. 

Thanks.

CasperSchippers, 2023-10-17:
Hi @erickmartinez

I've tried to reproduce your error but, as you sugggest indeed, with no success. I find the behaviour you describe very strange and cannot say that I really understand what is happening.
The strange thing is that there is no link between emitting the data and the thread, so I don't understand how the thread can influence the data-emitting. The only thing I could think of right now is that the images are still being written to file while you start emitting the data? You could try inserting a sleep of a some 10 seconds or so (as a test) between the parallel_process.join() and the start of the for-loop. Or try increasing the sleep in that for-loop.

erickmartinez, 2023-10-19:
Hi @CasperSchippers,

Thanks for the follow-up. I tried both, adding a pause between the `parallel_process.join()` and the start of the for-loop and separately increasing the delay within the for-loop, but neither solution has worked for me. I have not found a way to isolate this issue from my hardware, so we might as well close the issue. Thanks anyway.

Best,

Erick

CasperSchippers, 2023-10-31:
Yeah, I'll mark it as closed/cannot reproduce; if you at any moment find a way to reproduce it (or find the exact problem), I'd be happy to hear it.

closed 2023-10-31 by CasperSchippers

erickmartinez, 2024-07-03:
Hi @CasperSchippers, I just found out _when_ (not why) was I getting this issue. I was emitting the results to a file located in Google Drive. Once I saved the output to a local drive with no syncing, no corruption occurred in the .csv file.

CasperSchippers, 2024-07-05:
Hi, thanks for getting back with this info, that's much appreciated.

That does explain what is happening. Generally, I would not recommend using synced folders to write files to, in any use case where a file is incrementally stored, like the results of a measurement, indeed for the risk of corrupting files. Most cloud storage applications can't handle files that are often and quickly written.