pymeasure/pymeasure #984: Prologix NI-VISA no longer working, gpib library issue

ebaum1211, 2023-10-30:
Hello,

Any help is greatly appreciated! My current setup involves python code in PyCharm IDE(I will paste the code snippet below), a Prologix GPIB-USB adapter connected to my laptop and the Keithley2400 SourceMeter. Connecting to the Keithley2400 via serial library in python works. The keithley can be written commands no problem with this library. I would like to use the pymeasure wrapper and not serial.write. I also want to note that I only started having issues around 10/13/2023, which leads me to believe there may be a bug in the newest NI-VISA version which was released 2 days prior (I have NI-VISA 2023 Q4). I have used the Pymeasure wrapper multiple times over the past year with no issues like this. (I also have an open question about this on NI's forums and the pyvisa github as well, problem still is ongoing)

Things I have noted:

- The Prologix cable is detected by the computer and NI-VISA as ASRL3::INSTR (aka COM3). There should be a GPIB portion involved, but trying even just COM3, NI-VISA can't communicate with via python or via NI-MAX.
- As stated above the python serial library to write prologix commands (ex. ++auto 1) works as expected
- I am using Anaconda3 as my interpreter and have run pyvisa-info and have pasted the results below. You will see the current key error appearing to be a missing gpib library. I haven't found a gpib32.dll. I am not sure which file I would manually upload, how to do that, or what next steps should be taken
- You will see there a is also a PyUSB backend error, which seems a little strange (Update: I fixed that problem now it is just gpib having issues)
- In terms of running the code snippet below, the error is:
vpmeter.use_front_terminals() pyvisa.errors.InvalidSession: Invalid session handle. The resource might be closed.
- There are other odd things that happen too:
VI_ERROR_SYSTEM_ERROR (-1073807360): Unknown system error (miscellaneous error). [This shows up once or twice when listing the resources]
The command to list resources shows the (assuming correct) USB address of the scope and then a few other "ASRLx::INSTR : Not known"  where x is the COM port number. So it at least detects the USB of the Prologix is connected to the laptop

Any help would be greatly appreciated! Thanks!
 

pyvisa-info:

UserWarning: GPIB library not found. Please manually load it using _load_lib(f
ilename). All GPIB functions will raise OSError until the library is manually loaded.
warnings.warn(message)
Machine Details:
Platform ID:  
Processor: Intel64 Family 6 Model 154 Stepping 4, GenuineIntel

Python:
Implementation: CPython
Executable: C:\Users\xx\Anaconda3\python.exe
Version: 3.9.13
Compiler: MSC v.1916 64 bit (AMD64)
Bits: 64bit
Build: Aug 25 2022 23:51:50 (#main)
Unicode: UCS4

PyVISA Version: 1.13.0

Backends:
ivi:
Version: 1.13.0 (bundled with PyVISA)
#1: C:\WINDOWS\system32\visa32.dll:
found by: auto
bitness: 64
Vendor: National Instruments
Impl. Version: 24119296
Spec. Version: 7340032
#2: C:\WINDOWS\system32\visa64.dll:
found by: auto
bitness: 64
Vendor: National Instruments
Impl. Version: 24119296
Spec. Version: 7340032
py:
Version: 0.7.0
ASRL INSTR: Available via PySerial (3.5)
TCPIP INSTR: Available
Resource discovery:
- VXI-11: ok
- hislip: ok
TCPIP SOCKET: Available
USB INSTR:
PyUSB does not seem to be properly installed.
Please refer to PyUSB documentation and
install a suitable backend like
libusb 0.1, libusb 1.0, libusbx,
libusb-win32 or OpenUSB.
No backend available
USB RAW:
PyUSB does not seem to be properly installed.
Please refer to PyUSB documentation and
install a suitable backend like
libusb 0.1, libusb 1.0, libusbx,
libusb-win32 or OpenUSB.
No backend available
VICP INSTR:
Please install PyVICP to use this resource type.
GPIB INSTR:
gpib_ctypes is installed but could not locate the gpib library.
Please manually load it using:
gpib_ctypes.gpib.gpib._load_lib(filename)
before importing pyvisa.
GPIB INTFC:
gpib_ctypes is installed but could not locate the gpib library.
Please manually load it using:
gpib_ctypes.gpib.gpib._load_lib(filename)
before importing pyvisa.

 

Code:

```
import pyvisa as visa
from pymeasure.instruments.keithley import Keithley2400
import pymeasure.adapters

adapter = pymeasure.adapters.PrologixAdapter('ASRL3::INSTR')
vpmeter = Keithley2400(adapter.gpib(25))
print(pymeasure.instruments.list_resources())
vpmeter.use_front_terminals() #code fails here
```

bmoneke, 2023-10-30:
Quick idea: Did you try pyvisa-py? It's a fast install and can verify, whether the backend is the culprit or something else.
You can use the pyvisa-py backend with the parameter (to the adapter) `visa_library="@py"`.

ebaum1211, 2023-10-30:
I had tried to set that earlier with no avail, but I had done it in the terminal and may not have done it correctly. Could you send a code snippet of the correct way to do this? I am not sure if it has to be declared before import pymeasure. Will post results here! Thanks so much!

bmoneke, 2023-10-30:
`adapter = pymeasure.adapters.PrologixAdapter('ASRL3::INSTR', visa_library="@py")` should do the trick.

ebaum1211, 2023-10-30:
Thank you very much! I am not in my lab yet, but will post as soon as I get there!

ebaum1211, 2023-10-30:
I made it to the lab and using the code snippet with the adapter line changed to the one you gave I get this output:

`UserWarning: GPIB library not found. Please manually load it using _load_lib(filename). All GPIB functions will raise OSError until the library is manually loaded.
  warnings.warn(message)
0 : USB0::0x0699::0x0408::C012230::INSTR : TEKTRONIX,MDO3024,C012230,CF:91.1CT FV:v1.10
1 : ASRL3::INSTR : Visa IO Error: check connections
VI_ERROR_RSRC_BUSY (-1073807246): The resource is valid, but VISA cannot currently access it.
2 : ASRL7::INSTR : Not known
3 : ASRL9::INSTR : Not known
4 : ASRL10::INSTR : Visa IO Error: check connections
VI_ERROR_SYSTEM_ERROR (-1073807360): Unknown system error (miscellaneous error).
5 : ASRL11::INSTR : Not known
('USB0::0x0699::0x0408::C012230::INSTR', 'ASRL3::INSTR', 'ASRL7::INSTR', 'ASRL9::INSTR', 'ASRL10::INSTR', 'ASRL11::INSTR')

Process finished with exit code 0`

It is an improvement because it does exit with code 0! When I went to try running my full program, I still got the same error as before. This time it failed on It fails on the line with `vpmeter.source_current = 0.000001`, not `vpmeter.use_front_terminals()` like it did before. When I tried it a second time it started working! No idea why but I am excited to have something operational again! I am going to try and run some of my experiments now and will report back if it is a success or if it fails again! While the biggest thing is just getting the test going so we can get data, do you know why the ASRL3:INSTR doesn't have a gpib port number? Why I get asked for a gpib library? And why I get the strange VISA error as it lists the resources? Thanks so so much! Here's hoping for a day of testing!

ebaum1211, 2023-10-31:
It has been running multiple tests, no more issues so I am going to assume that was the fix! However, if you do get time to answer the questions at the end of my last post, I would appreciate it! I want to see if I can fix the warnings or at least understand them a bit better. Seriously thank you so much this has had me stumped for weeks

bmoneke, 2023-10-31:
Good to hear, that it works for you, now, at least kind of.

> why the ASRL3:INSTR doesn't have a gpib port number?

as you are using a prologix adapter, the computer (and VISA backend) do not know, that there is GPIB behind the prologix device. For the computer, there is just a serial connection to the prologix device.

> Why I get asked for a gpib library?

I guess, that pyvisa-py tries to load all connection possibilities (serial, ethernet gpib) but cannot find the gpib version. I found a fitting issue: https://github.com/tivek/gpib_ctypes/issues/2 . However, as you do not use gpib (from the VISA point of view) but serial, that should not be a problem. Your prologix device (and the prologix adapter) do the GPIB work, therefore you do not need gpib support.

> And why I get the strange VISA error as it lists the resources?

I don't know. It seems, that the `list_resources` method of pymeasure is executed.

ebaum1211, 2023-11-07:
Thanks so much! This makes a lot of sense. I think the only question remaining is why NI-VISA stopped working after having been fine for many months. Has NI-VISA been tested with pymeasure (or perhaps pyvisa) lately? Maybe it is an incompatibilty on the NI side? Or maybe some mysterious issue lies within my peronal laptop. Thanks again for the help! P.S. communication with Prologix has been fine ever since you suggested pyvisa-py

bmoneke, 2023-11-07:
I did not notice anything, but I did not change the ni backend, either.
We don't have automatic testing with ni, but I'd say there are quite a few users out there. 

I'm glad, that we could help you.

If the issue is resolved, you may close the issue.

ebaum1211, 2023-11-07:
Will do! If anyone has issues with NI-VISA as a backend or figures out why I might have run into this issue please re-open. Thanks!

closed 2023-11-07 by ebaum1211