pymeasure/pymeasure #985: SR830 pause buffer

FlyingCurryMonster, 2023-11-02:
I'm trying to diagnose some digitization issues on my SR830 and noticed there's a bug related to pauseBuffer().  In the SR830 py file there are several cases where self.pause_buffer() is used, when the actual function call should be  "self.pauseBuffer()".

bmoneke, 2023-11-03:
Yes, indeed. 8 years ago `pauseBuffer` was renamed to `pause_buffer`, but not all occurrences.

Would you mind fixing it? If you need help with it, we're here.

FlyingCurryMonster, 2023-11-03:
For sure.  It seems like the support for measurements with the buffer isn't too well supported at the moment.  That's not too surprising, I didn't need it until I ran into issues where my device noise is lower than the resolution of the SR830.
I'm guessing the workaround for digitization is to use the SR830 buffer and fast sample rate to record ~1000 measurements in the buffer and take the mean of the buffer, rather than record the front reading.  I think by central limit theorem (I think?) that should cure the quantization problem.  

I'll try and fix the pause and the buffer measure methods.

Any input or thoughts are appreciated.

bmoneke, 2023-11-06:
> Any input or thoughts are appreciated

For the naming, `pause_buffer` is the better name, as that is the normal python naming scheme.

Regarding the `buffer_data`, I don't know anything

seb5g, 2023-12-15:
Hi guys, 
I'm actually working on making this driver working again. Indeed the buffer_data was not defined... I should be done with a first update soon

BenediktBurger, 2023-12-15:
Thanks for your comment @seb5g.

FlyingCurryMonster, 2024-06-13:
I've made a pull request here https://github.com/pymeasure/pymeasure/pull/1125

seb5g, 2024-06-13:
I'm a bit lost here, has my PR on the subject been merged?

seb5g, 2024-06-13:
> I'm a bit lost here, has my PR on the subject been merged?

actually yes it is the #999 
@FlyingCurryMonster did you use an earlier version of pymeasure without this patch?

BenediktBurger, 2024-06-13:
Oh, merging your PR I missed that it closes this issue.

@FlyingCurryMonster  could you please check, whether the current master branch or even the current release (#999 is already part of the current released version) solves your issue?

FlyingCurryMonster, 2024-06-13:
Current branch solves my problems!

seb5g, 2024-06-13:
you're welcome ;-)

BenediktBurger, 2024-06-14:
So this issue is resolved in #999

FlyingCurryMonster, 2024-06-21:
I'm finding that this method isnt' working for me as expected.  Here's some code I execute in jupyter cell that produces the error:

`
gpib_addr = 'GPIB2::9::INSTR'
lockin1 = srs.SR830(gpib_addr)

lockin1.sample_frequency = 128
print(lockin1.sample_frequency)

lockin1.reset_buffer()
lockin1.buffer_measure(1280)
`

Here's the error : 
'
0
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
~\AppData\Local\Temp/ipykernel_9436/4029623281.py in <module>
      4 
      5 lockin1.sample_frequency = 128
----> 6 print(lockin1.sample_frequency)
      7 
      8 lockin1.reset_buffer()

~\Documents\GitHub\pymeasure\pymeasure\instruments\common_base.py in __getattribute__(self, name)
    296                 raise AttributeError(
    297                     f"{name} is a reserved variable name and it cannot be read")
--> 298         return super().__getattribute__(name)
    299 
    300     # Channel management

~\Documents\GitHub\pymeasure\pymeasure\instruments\srs\sr830.py in sample_frequency(self)
    444     def sample_frequency(self):
    445         """ Gets the sample frequency in Hz """
--> 446         index = int(self.ask("SRAT?"))
    447         if index == 14:
    448             return None  # Trigger

ValueError: invalid literal for int() with base 10: '\x1e11\n'
'

I believe my fork should be up to date with the current github repo.

BenediktBurger, 2024-06-22:
Ok, I reopen this issue.

BenediktBurger, 2024-06-24:
I'm sorry, but as I do not have the instrument, I'm not able to help with that.
The manual is your best bet (and trial and error).

FlyingCurryMonster, 2024-06-24:
No worries @BenediktBurger I've learned some workarounds for a temporary fix.  Essentially 'FAST0' mode works fine, but 'FAST1' or 'FAST2' is tough to manage.  In the 'FAST1' or 'FAST2' mode what's happening is that data from the buffer is being sent to the computer immediately upon sending the 'STRD' string to start the scan.  So any other query, like getting the buffer_count, produces the value error.

The temporary solution is to just include a argument in the `buffer_measure()` method to specify the fast mode you want to measure in.  FAST1 or FAST2 just doesn't seem supportable yet and needs some thought.  

On a different issue, it looks like the buffer_measure method always seems to return a buffer that is about double the size I requested.  I'm not sure why that is.

BenediktBurger, 2024-06-24:
The `buffer_measure` method returns the mean and std values of both channels or do we talk about different things?

https://github.com/pymeasure/pymeasure/blob/606fae7e8063594394a089ee88d2aa5df8f19570/pymeasure/instruments/srs/sr830.py#L530-L554

FlyingCurryMonster, 2024-06-24:
If you call `pause_buffer` immediately after `buffer_measure`, and check `buffer_count`, I've found that the count is about double than the count parameter passed into `buffer_measure`.  Some code I ran to check is:

`lockin1.reset_buffer()
t0 = time.time()
res_bin = lockin1.buffer_measure(count = 640, delay=0); print(res_bin)
print('meas time', time.time()-t0)
print(lockin1.buffer_count)`

And the output is
(5.774419831183763e-12, 6.255469519147662e-13, -3.386072533473339e-11, 5.891471545429707e-13)
meas time 18.578521728515625
1081

BenediktBurger, 2024-06-24:
A hypothesis: After you finish your `buffer_measure`, the device continues to read values into its internal buffer.
Once you reach the `buffer_count`, it has acquires another 440 measurements, totalling 1081 measurements.

FlyingCurryMonster, 2024-06-24:
Oh I see, seems like the `get_buffer` method takes a while and the buffer populates during the i/o time.  in the 'FAST0' method I don't know what the advantage is in executing `get_buffer` in the while loop since it's executed just before the return.  

I think in `FAST0` mode I would just perform:
`start_buffer(fast=False)`
`wait_for_buffer(count)`
`get_buffer(count)`

The `sleep` in `wait_for_buffer` could allow for other threads in a data acquisition script to execute while the buffer populates.  What do you think?

FlyingCurryMonster, 2024-06-24:
Seems like I'm getting the Fast2 method to work just okay.  As soon as the scan is started, the `read_bytes` method needs to be called.  Kind of like the following:

self.write('FAST2;STRD')
self.read_bytes(4*buffer_size)

where buffer_size is a user input argument.  The count argument on the read should be 4 times the desired buffer size according to the manual (2 bytes encodes ch1 and ch2 each.)  The issue arises when the measurement time exceeds ~4s (ie sample rate 512Hz but desired buffer size of 2048) because a VISA IO timeout error occurs.  I need to somehow pass in an argument in `read_bytes` that lets the timeout error be slightly larger than the expected measurement time.  

The speedup using FAST2 is noticeably larger.  In a scan of 1280 data points at 128Hz sample rate, it takes an extra ~12-15 seconds to send dual channel data to my lab computer.  In FAST2 mode the data is sent as the scan progresses, so the run time of the method is only limited by the sample frequency rate, ie nearly as fast as it can be.  At 512Hz sample rate with a buffer_count of 16383 (maximum allowed) the measurement takes 36s, ie only a 4s lag.

BenediktBurger, 2024-06-24:
Glad to hear, that you understand the instrument.

You can do two things: increase the timeout or you read the data in chunks of 100 bytes, for example. Then you can concatenate all the read chunks for the whole data.

Depending on the buffer size (of your computer), you can even what as long as you whish and read the whole buffer at once at that moment (defeating the purpose of sending while measuring).

BenediktBurger, 2024-06-24:
There is another instrument, which changes the timeout (temporarily) for such a measurement.

You can access the timeout via `inst.adapter.connection.timeout` which is in ms (for the visa Adapter).
Unfortunately it is not harmonized yet regarding the serial adapter (in seconds) or a common "timeout" method (#783).

FlyingCurryMonster, 2024-06-24:
Thanks changing timeout works great, I just worry that in case  the read fails the timeout doesn't get set back to default (I'll try wrapping my read write code in try except.)

On a different note, it looks like in the time between `read_bytes(4*buffer_size)` and `pause_buffer` the SR830 has sent more bytes to the computer.  I can use `read_bytes(-1)` to clear this, but I imagine for slow sampling rates this solution is not optimal as the buffer might not even fill within the time gap of the read and pause. I saw this `flush_read_buffer` method that looked like it would be the perfect thing but can't seem to access it.  Is this a part of the adapter? 

https://pymeasure.readthedocs.io/en/latest/api/adapters.html#pymeasure.adapters.VISAAdapter.flush_read_buffer

BenediktBurger, 2024-06-24:
Iirc, it is part of the visa Adapter.

FlyingCurryMonster, 2024-06-25:
After I try to call `flush_read_buffer` I get the following error :
pyvisa.errors.VisaIOError: VI_ERROR_INV_MASK (-1073807299): Invalid buffer mask specified.
Looking into the `flush` method in the pyvisa library, it looks mask is a necessary parameter.  Not sure what it should be in my case. 
https://pyvisa.readthedocs.io/en/latest/api/visalibrarybase.html#pyvisa.highlevel.VisaLibraryBase.flush.

Clearing the buffer with `read_bytes(-1)` extends the run time considerably, but I imagine there must surely be a way to flush the contents in the read buffer of the communication line.

FlyingCurryMonster, 2024-06-25:
seems like my problem specifically was in the second line of the try block in flush_read_buffer()

`self.connection.flush(pyvisa.constants.BufferOperation.discard_read_buffer)
self.connection.flush(pyvisa.constants.BufferOperation.discard_receive_buffer)`

The second line returns an error when I try it in an empty script.  What worked for me was 
` self.adapter.connection.flush(pyvisa.constants.BufferOperation.discard_read_buffer_no_io)`
The no IO seemed to be important

FlyingCurryMonster, 2024-06-26:
I've figured out how to clear the buffer quickly.  I just set the timeout to 0.5e3 milliseconds and read(-1).  The problem really only occurs for me at 512Hz which is an edge case I'll never visit practically.  Once I try a measurement and complete it successfully I'll start working on a pull request.  

@BenediktBurger considering that the current `buffer_measure` method just does not work in FAST2 mode, and does not obtain the desired buffer size even in the FAST0 mode, should it be totally replaced?

BenediktBurger, 2024-06-27:
@seb5g you reworked the `buffer_measure` method. Would you mind looking at this issue regarding that method?

FlyingCurryMonster, 2024-06-28:
I've made a PR https://github.com/pymeasure/pymeasure/pull/1132