pymeasure/pymeasure #986: Managed Dock Window not allowing dragging of docks

judecompstew, 2023-11-08:
I am trying to use a ManagedDockWindow as part of a GUI for an experimental setup that displays three graphs, similar to the tutorial example using the random number generator.  I see no difference between my code and the equivalent sections of the tutorial code, but in my GUI, I am not able to click on the Dock title bar and manipulate the positioning of three docks. When I click on the blue bars, nothing happens. There is no blue outline or options to move the dock.
The tutorial code works perfectly in the same anaconda environment. The pymeasure version is the 0.13.1. 
I have a current work around of opening the tutorial code, arranging the docks as I want them to appear, saving the dock layout, moving the file into the right directory and renaming it, but this isn't a very good solution. 

I will provide my code if requested, but does use a specific dll file for a Newport delay line I'm using.

Any help with this small issue will be greatly appreciated.

mcdo0486, 2023-11-09:
I'd reccomend just copy-pasting the section from the tutorial into a new file and trying that out:
https://pymeasure.readthedocs.io/en/latest/tutorial/graphical.html#using-the-manageddockwindow

If that doesn't work, yea share the code you are running in a github gist or in this issue.

Also to popout the dock windows you need to double click the blue dock title bar, like "Dock 1" .

Dan

judecompstew, 2023-11-10:
I tried copying the tutorial code and everything worked as expected, but still my code isn't allowing me to drag any docks. When I double click, the windows do pop out in my code, but still no ability to drag them around the window.

I've attached the code I'm using. The original is a Jupyter Notebook, but I can't attach this directly so I've taken the relevant code and made it into a txt file. Hopefully this is easy enough to access and run. 

[GUI code.txt](https://github.com/pymeasure/pymeasure/files/13318093/GUI.code.txt)

It's made for controlling a few bits of equipment, but the basic structure is move a delay line, take some measurements, move a delay line, tale some measurements, etc...

Thanks,
Jude

CasperSchippers, 2023-11-12:
When I run your code (I just needed to comment out some lines due to missing imports), I can actually just drag and drop the docs.

Maybe it is related to the fact that your running it in a jupyter notebook? Have you tried running the code as a normal python file?
(Although, when I run your code from within a single jupyter notebook cell, it also seems to work, although slightly less smooth than as a normal file).

mcdo0486, 2023-11-12:
+1 for running the code as a standalone python file, especially since the dock tutorial code works on your machine.

I also tested your code on a Windows computer and the dock behavior was working, both as standalone file and in a jupyter notebook. I did need to comment out some imports.