pymeasure/pymeasure #997: How to use properties when no specific termination character is set in the protocol

seb5g, 2023-12-15:
I have a very old but working instrument. It communicate over serial but the command list is quite weird and sometime there is a termination character, sometime there isn't. Sometimes there is an answer after writing sometimes not.

Therefore my question is how to handle that? at the moment I always have a timeouterror when setting something because the read immediately returns a timeout error (when no response is expected or when there is no termination character). Is it also possible to change the termination character dynamically within the get/set process? 
If I write down methods to do this I would actually use the timeout error as a termination character but for this to work I have to make sure to read bytes one by one until there is a timeout, then I will return the string.

Thanks for the pointers (if there are...)
Seb

BenediktBurger, 2023-12-15:
The place to write your logic in is the "read" method.
There you can read bytes (one ore more) if you wish. The method is called "read_bytes". Actually, that method can read until timeout. 
There you could set the termination char before doing the read etc.

I have also one instrument, which gets hiccups (zero byte) from time to time and the normal read method doesn't like non ASCII chars. Therefore I read byte for byte and return the correct ones in the "read" method.

seb5g, 2023-12-15:
Of course subclassify the read method...silly me

closed 2023-12-15 by seb5g