pymeasure/pymeasure #1008: Python 3.12 seems to work if NI instruments gets commented out

Sionwage, 2023-12-20:
I would like to report success in using PyMeasure with Python 3.12 when I comment out line 59 in `pymeasure\instruments\__init__.py`

More specifically, the submodule 'NI' within instruments wants to load `DAQmx` which ends up looking for `ctypes.windll.nicaiu` which if not installed raises an AttributeError.

Running the following:

```python
from pymeasure.instruments import Instrument
```

Full error message
```
---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
File ~\.pyenv\pyenv-win\versions\3.12.0\Lib\ctypes\__init__.py:450, in LibraryLoader.__getattr__(self, name)
    449 try:
--> 450     dll = self._dlltype(name)
    451 except OSError:

File ~\.pyenv\pyenv-win\versions\3.12.0\Lib\ctypes\__init__.py:379, in CDLL.__init__(self, name, mode, handle, use_errno, use_last_error, winmode)
    378 if handle is None:
--> 379     self._handle = _dlopen(self._name, mode)
    380 else:

FileNotFoundError: Could not find module 'nicaiu' (or one of its dependencies). Try using the full path with constructor syntax.

During handling of the above exception, another exception occurred:

AttributeError                            Traceback (most recent call last)
Cell In[1], line 6
      3 from io import BytesIO
      5 import numpy as np
----> 6 from pymeasure.instruments import Instrument, RangeException, discreteTruncate
      7 from pyvisa import VisaIOError
      9 from pymeasure.adapters import PrologixAdapter

File ~\AppData\Local\pypoetry\Cache\virtualenvs\py3-12-pymeasure-env-DpKIcySs-py3.12\Lib\site-packages\pymeasure\instruments\__init__.py:59
     57 from . import mksinst
     58 from . import newport
---> 59 from . import ni
     60 from . import novanta
     61 from . import oxfordinstruments

File ~\AppData\Local\pypoetry\Cache\virtualenvs\py3-12-pymeasure-env-DpKIcySs-py3.12\Lib\site-packages\pymeasure\instruments\ni\__init__.py:26
      1 #
      2 # This file is part of the PyMeasure package.
      3 #
   (...)
     22 # THE SOFTWARE.
     23 #
     25 try:
---> 26     from .daqmx import DAQmx
     27 except OSError:
     28     # Error Logging is handled within package
     29     pass

File ~\AppData\Local\pypoetry\Cache\virtualenvs\py3-12-pymeasure-env-DpKIcySs-py3.12\Lib\site-packages\pymeasure\instruments\ni\daqmx.py:38
     36 try:
     37     if platform == "win32":
---> 38         nidaq = ctypes.windll.nicaiu
     39 except OSError as err:
     40     log.info('Failed loading the NI-DAQmx library. '
     41              + 'Check the NI-DAQmx documentation on how to '
     42              + 'install this external dependency. '
     43              + f'OSError: {err}')

File ~\.pyenv\pyenv-win\versions\3.12.0\Lib\ctypes\__init__.py:452, in LibraryLoader.__getattr__(self, name)
    450     dll = self._dlltype(name)
    451 except OSError:
--> 452     raise AttributeError(name)
    453 setattr(self, name, dll)
    454 return dll

AttributeError: nicaiu
```

Would this be a good instance of utilizing dynamic imports with the `importlib` module to prevent searching for the `instrumental` package from https://github.com/mabuchilab/Instrumental unless one is specifically importing `NIQAQ` from `pymeasure.instruments.ni`?

BenediktBurger, 2023-12-30:
Some relations to #662 , where there is a discussion about the NI drivers.

Sionwage, 2024-02-19:
#919 Solves this issue since `daqmx.py` doesn't get imported on initializing the Instrument class.

Sionwage, 2024-02-19:
Solved by 919

BenediktBurger, 2024-02-20:
Now pymeasure works for the other instruments, but the NI instruments won't work with python 3.12, therefore I reopen this issue.

Sionwage, 2024-02-20:
Also, probably need to add an exception in `test_all_instruments.py` for `find_devices_in_module(module)` to except AttributeError for nicaiu to get testing to pass.

BenediktBurger, 2024-05-07:
#1051 adds that error case to `test_all_instruments.py`.
Thanks for finding them, @Sionwage

closed 2026-10-21 by BenediktBurger