pymeasure/pymeasure #1009: example in gui.py missing queue function. ManagedWindow.queue not implemented

TedKus, 2023-12-21:
```python
python .\gui.py
Traceback (most recent call last):
  File "C:\Users\user\AppData\Roaming\Python\Python311\site-packages\pymeasure\display\windows\managed_window.py", line 491, in _queue
    self.queue()
  File "C:\Users\user\AppData\Roaming\Python\Python311\site-packages\pymeasure\display\windows\managed_window.py", line 523, in queue
    raise NotImplementedError(
NotImplementedError: Abstract method ManagedWindow.queue not implemented
```

MainWindow class needs a queue function for the gui button:
Added two imports, tempfile and Results. 
Added MainWindow.queue

This file is substantially similar to console.py.  Perhaps consider merging?
I've put the entire corrected file here:
```python
#
# This file is part of the PyMeasure package.
#
# Copyright (c) 2013-2023 PyMeasure Developers
#
# Permission is hereby granted, free of charge, to any person obtaining a copy
# of this software and associated documentation files (the "Software"), to deal
# in the Software without restriction, including without limitation the rights
# to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
# copies of the Software, and to permit persons to whom the Software is
# furnished to do so, subject to the following conditions:
#
# The above copyright notice and this permission notice shall be included in
# all copies or substantial portions of the Software.
#
# THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
# IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
# FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
# AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
# LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
# OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
# THE SOFTWARE.
#

"""
This example demonstrates how to make a graphical interface, and uses
a random number generator to simulate data so that it does not require
an instrument to use.

Run the program by changing to the directory containing this file and calling:

python gui.py

"""

import sys
import random
import tempfile
from time import sleep

from pymeasure.experiment import (Procedure, IntegerParameter, Parameter,
                                  Results, FloatParameter)
from pymeasure.display.Qt import QtWidgets
from pymeasure.display.windows import ManagedWindow

import logging
log = logging.getLogger('')
log.addHandler(logging.NullHandler())

class TestProcedure(Procedure):

    iterations = IntegerParameter('Loop Iterations', default=100)
    delay = FloatParameter('Delay Time', units='s', default=0.2)
    seed = Parameter('Random Seed', default='12345')

    DATA_COLUMNS = ['Iteration', 'Random Number']

    def startup(self):
        log.info("Setting up random number generator")
        random.seed(self.seed)

    def execute(self):
        log.info("Starting to generate numbers")
        for i in range(self.iterations):
            data = {
                'Iteration': i,
                'Random Number': random.random()
            }
            log.debug("Produced numbers: %s" % data)
            self.emit('results', data)
            self.emit('progress', 100 * i / self.iterations)
            sleep(self.delay)
            if self.should_stop():
                log.warning("Catch stop command in procedure")
                break

    def shutdown(self):
        log.info("Finished")

class MainWindow(ManagedWindow):

    def __init__(self):
        super().__init__(
            procedure_class=TestProcedure,
            inputs=['iterations', 'delay', 'seed'],
            displays=['iterations', 'delay', 'seed'],
            x_axis='Iteration',
            y_axis='Random Number'
        )
        self.setWindowTitle('GUI Example')

    def queue(self):
        filename = tempfile.mktemp()

        procedure = self.make_procedure()
        results = Results(procedure, filename)
        experiment = self.new_experiment(results)

        self.manager.queue(experiment)

if __name__ == "__main__":
    app = QtWidgets.QApplication(sys.argv)
    window = MainWindow()
    window.show()
    sys.exit(app.exec())

```

BenediktBurger, 2023-12-30:
What did you change?
A PR makes it clearer, what line(s) you changed.

Thanks for raising that bug.

CasperSchippers, 2024-01-02:
I actually think something else is the matter: the [`v: latest`](https://pymeasure.readthedocs.io/en/latest/) documentation reflects the latest version on github, not the latest release; I think for the latest release, you need to look at [`v: stable`](https://pymeasure.readthedocs.io/en/stable/). You can switch between the two versions in the bottom left corner.
 
The absence of the `queue` method in the `ManagedWindow` example is by design: as per PR #964, it is not needed anymore to have a boilerplate `queue` method (provided you do not set `enabled_file_input` to `False`).